introlab/rtabmap_ros #967: Error while running RTAB-Map ROS using TUM Datasets

itsthehoneybadger, 2023-05-14:
Hi there, I wanted to run RTAB-Map ROS using TUM datasets, specifically the 'freiburg2_pioneer_slam' sequence. 
I use this launch command to launch:
the RTAB-Map: roslaunch rtabmap_examples rgbdslam_datasets.launch rtabmap_viz:=true
the 'freiburg2_pioneer_slam' sequence': rosbag play --pause --clock rgbd_dataset_freiburg2_pioneer_slam.bag

Before running the rosbag file, I have already run this command: 
python3 tum_rename_world_kinect_frame.py rgbd_dataset_freiburg2_pioneer_slam.bag 

However, I am getting this error while running the RTAB-Map ROS using the TUM dataset:
[image]

What would be the possible cause of this error? Thank you in advance. :D

matlabbe, 2023-05-19:
Those are old rosbags with some invalid frame_id (with leading `/`). TF doesn't allow anymore those frames. You may have to republish the topics with right `frame_id` without leading slash.

itsthehoneybadger, 2023-05-19:
Hi, thank you very much for the reply. I am pretty new to ROS and RTAB-Map, would mind explaining me on how to republish the topics with the right frame_id? Thank you in advance

matlabbe, 2023-05-29:
See http://wiki.ros.org/ROS/Tutorials/WritingPublisherSubscriber%28python%29

Create subscriber on topic your want to change the `frame_id` in the message. Create a publishers to publish the same topic but with its `frame_id` without the leading `/`.