introlab/rtabmap_ros #968: Rtabmap Viz Crash

MrDarknessWolf, 2023-05-15:
Using Ros2 Foxy inside a Jetson Nano with ubuntu 20.4 and a Realsense D435i , for some reason launching examples and mapping works BUT when trying to open normal launch file it kills the process 

[ERROR] [rtabmap_viz-2]: process has died [pid 45223, exit code -6, cmd '/opt/ros/foxy/lib/rtabmap_viz/rtabmap_viz ~/.ros/rtabmap_gui.ini --ros-args -r __ns:=/rtabmap --params-file /tmp/launch_params_b6h3hj2g --params-file /tmp/launch_params_7eirp1_o -r rgb/image:=/camera/rgb/image_rect_color -r depth/image:=/camera/depth_registered/image_raw -r rgb/camera_info:=/camera/rgb/camera_info -r rgbd_image:=rgbd_image_relay -r left/image_rect:=/stereo_camera/left/image_rect_color -r right/image_rect:=/stereo_camera/right/image_rect -r left/camera_info:=/stereo_camera/left/camera_info -r right/camera_info:=/stereo_camera/right/camera_info -r scan:=/scan -r scan_cloud:=/scan_cloud -r odom:=/odom'].

[image]

matlabbe, 2023-05-19:
There are some issues on jetson with native `rtabmap_viz`. I would not launch `rtabmap_viz` and use RVIZ with `rtabmap_rviz_plugins` instead (or even launch visualization on remote computer instead).

MrDarknessWolf, 2023-05-19:
> There are some issues on jetson with native `rtabmap_viz`. I would not launch `rtabmap_viz` and use RVIZ with `rtabmap_rviz_plugins` instead (or even launch `rtabmap_viz` on remote computer instead).

whats interesting is that rtabmap_viz does launch on its own no creashes,and when using the rtabmap_examples for the d435i it launches perfectly everything more so opens BUT the launch file, maybe is the launch file itself but its a bit too complex for me to understand where it must be failing to initialize given the initial parameters

matlabbe, 2023-05-19:
Did you use the "color D435i" or "IR D435i" example launch files? With color, I did have an example with current realsense driver with not valid depth image, so needed to regenerate it in that example.

MrDarknessWolf, 2023-05-19:
> Did you use the "color D435i" or "IR D435i" example launch files? With color, I did have an example with current realsense driver with not valid depth image, so needed to regenerate it in that example.

IR, Color and Stereo open, map without crashing and gives trajectory, 2d mapping and 3d mapping export

matlabbe, 2023-05-19:
So all launch examples are working, can you give the launch / commands that are not working?

MrDarknessWolf, 2023-05-19:
> So all launch examples are working, can you give the launch / commands that are not working?

So the basic 

ros2 launch rtabmap_launch rtabmap.launch.py  doesnt launch at all it just, crashes the Viz, but keeps trying to work out the slam part.

/////////////////////////////////////////////////////////////////////////////////////////////////
basically the same and actually doesnt seem to remap the topics subscribers .
ros2 launch rtabmap_launch rtabmap.launch.py \
 args:="--delete_db_on_start" \
 depth_topic:=/camera/aligned_depth_to_color/image_raw \
 rgb_topic:=/camera/color/image_raw \
 camera_info_topic:=/camera/color/camera_info \
 approx_sync:=false \
 frame_id:=camera_link

/////////////////////////////////////////////////////////////////////////////
the example it self doesnt launch the viz but does open RVIZ2

ros2 launch rtabmap_launch rtabmap.launch.py \
   visual_odometry:=false \
   frame_id:=base_footprint \
   subscribe_scan:=true depth:=false \
   approx_sync:=true \
   odom_topic:=/odom \
   scan_topic:=/scan \
   qos:=2 \
   args:="-d --RGBD/NeighborLinkRefining true --Reg/Strategy 1" \
   use_sim_time:=true \
   rviz:=true

matlabbe, 2023-05-19:
Cannot reproduce the problem on my computer, would have to try on jetson later.

SkyaX, 2023-05-26:
By looking at the message above the error : ```expected {double} got {string}```, might be caused by the ```wait_for_transform``` parameter that is interpreted as a string in ```/rtabmap_viz/src/GuiWrapper.cpp``` ? 

I had this issue and simply replaced the argument declaration (line 117) by the desired value (0.5 for example).
Hope it solves your problem; I am not quite sure why it's happening though...

MrDarknessWolf, 2023-05-30:
> By looking at the message above the error : `expected {double} got {string}`, might be caused by the `wait_for_transform` parameter that is interpreted as a string in [...]

sadly right now i cant test it out but i cant leave it in the notes for future testing , for now its satisfactory the demo launching

sun-rabbit, 2023-10-19:
> 无法在我的计算机上重现该问题，稍后必须在 jetson 上尝试。

hi，i also Using Ros2 Foxy inside a Jetson jeston agx orin with ubuntu 20.4 and a astro pro  , for some reason launching examples 
###########
ros2 launch rtabmap_launch rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
    rgb_topic:=/camera/color/image_raw \
    depth_topic:=/camera/depth/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    frame_id:=base_link \
    approx_sync:=true \
    qos:=2 \
    rviz:=true \
    queue_size:=20 \
    subscribe_odom_info:=false
###############
[rtabmap_viz-3] [WARN] [1697700727.832576678] [rtabmap.rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=20). 

#############
I ensure that I have synchronized the time using NTP.