introlab/rtabmap_ros #969: Ubuntu 22.04 0.21.1-1jammy.20230426.050326 version of ros-humble-rtabmap-ros does not work

ngmor, 2023-05-15:
I haven't been able to use rtabmap-ros with the currently available package for Humble on Ubuntu 22.04.

I recently (5/4/2023) ran a `sudo apt upgrade` which updated the `ros-humble-rtabmap-ros` package from `0.20.22-1jammy.20230303.071330` to `0.21.1-1jammy.20230428.162517`. I used the package through the end of March 2023, so I believe the older version worked.

More details:
```
$ apt-cache policy ros-humble-rtabmap-ros
ros-humble-rtabmap-ros:
  Installed: 0.21.1-1jammy.20230428.162517
  Candidate: 0.21.1-1jammy.20230428.162517
  Version table:
 *** 0.21.1-1jammy.20230428.162517 500
        500 http://packages.ros.org/ros2/ubuntu jammy/main amd64 Packages
        100 /var/lib/dpkg/status
```

Now when I try to run nodes from this package, the following occurs:
```
$ ros2 run rtabmap_ros rtabmap
No executable found
```

When launching from a launch file like this:
```
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():
    return LaunchDescription([Node(package='rtabmap_ros', executable='rtabmap')])
```
I get this error:
```
$ ros2 launch test_package test.launch.py 
[INFO] [launch]: All log files can be found below /home/ngm/.ros/log/2023-05-15-18-50-15-716030-galagongm-11454
[INFO] [launch]: Default logging verbosity is set to INFO
[ERROR] [launch]: Caught exception in launch (see debug for traceback): package 'rtabmap_ros' found at '/opt/ros/humble', but libexec directory '/opt/ros/humble/lib/rtabmap_ros' does not exist
```

Downgrading doesn't seem to work, since the only package available via apt seems to be this broken package.

Tagging @katie-hughes (who also has this issue).

MrDarknessWolf, 2023-05-16:
for some reason the ros2 repository for Rtabmap has no kaunch files inside the ros folder 
[image]

thats why you cant call an executable from it

matlabbe, 2023-05-16:
With the new interface, try 
```
ros2 run rtabmap_slam rtabmap
```

See also http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2

MrDarknessWolf, 2023-05-16:
> With the new interface, try
> 
> ```
> ros2 run rtabmap_slam rtabmap
> ```
> 
> See also http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2

It does launch but it doesnt seem like we cant change its subscriber topics 
[image]

[image]

matlabbe, 2023-05-16:
You are uisng the interface of the launch file:
```
ros2 launch rtabmap_launch rtabmap.launch.py ...
```

ngmor, 2023-05-16:
> With the new interface, try
> 
> ```
> ros2 run rtabmap_slam rtabmap
> ```
> 
> See also http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2

Ah, ok, I did not see this new interface update. It seems to work on my end after following the migration guide.

Thanks for your help!

@MrDarknessWolf I'm going to close this issue, feel free to start a new one if you're still seeing the problems you've described above.

closed 2023-05-16 by ngmor

miku54, 2024-04-24:
Sorry to bother you now, could you tell me how to downgrade?

matlabbe, 2024-04-29:
If you want to use the old interface, uninstall `rtabmap_ros` binaries (`sudo apt remove ros-$ROS_DISTRO-rtabmap-*`) and rebuild [rtabmap_ros version <= 0.20.3](https://github.com/introlab/rtabmap_ros/tree/0.20.23-humble) from source. I didn't test it, but you may still be able to keep `rtabmap` binaries or use latest [rtabmap](https://github.com/introlab/rtabmap) version from source.

miku54, 2024-04-29:
> rebuild [rtabmap_ros version <= 0.20.3](https://github.com/introlab/rtabmap_ros/tree/0.20.23-humble) from source

I tried this solution [rebuild rtabmap_ros version <= 0.20.3 from source]. Unfortunately, the dependency [Rtabconfig] will be missing when compiled alone, and apt-get install cannot download it. This solution takes a lot of time. I Testing has not been completed yet.

matlabbe, 2024-05-06:
The only solution to downgrade is to rebuild rtabmap and rtabmap_ros from source following version above.