introlab/rtabmap_ros #975: Visual odometry stops working unexpectedly

JajamesVid, 2023-05-29:
Hello,  I'm executing stereo_odometry node individually using rtabmap.launch, it works fine and gives me a nice odometry result. My camera doesn't publish the /tf transform so I had to publish it manually using static_transform_publisher.

The issue starts when after a few time of good odometry, the frequency of the /odom topic starts to low it's value, until the odometry gets lost and this warns appears:

[ WARN] [1685374109.176753085]: Could not get transform from base_link to camera_link after 0.200000 seconds (for stamp=1685374109.235505)! Error="Lookup would require extrapolation 0.480000000s into the future.  Requested time 1685374109.235505342 but the latest data is at time 1685374108.755505323, when looking up transform from frame [camera_link] to frame [base_link]. canTransform returned after 0.20014 timeout was 0.2.".

So the question is if the topic /tf must have to be published at a determined frequency rate or must have to be synchronized with other frequencies of the rest of the topics (image_rect, camera_info...)

Thanks in advance

matlabbe, 2023-05-30:
If not, what is your setup? The /odom topic you are referring, is it the output of stereo_odometry? How is base_link to camera_link published? It looks published at least 480 ms in the future.

How fast `static_transform_publisher` rate is? Maybe the published images are using a different clock than your computer generating the static transform.

JajamesVid, 2023-05-30:
Hello matlabbe, thanks for the quick response, no it's not with Euroc dataset and yes I'm referring to the output of stereo_odometry (/rtabmap/odom).

I'm using a stereo camera which publishes image and camera_info topics in 'camera_link' frame and publishing /tf by executing this line in my launch file:
<arg name="optical_rotate" value="0 0 0 -$(arg pi/2) 0 -$(arg pi/2)" />
     <node pkg="tf" type="static_transform_publisher" name="camera_link"
        args="$(arg optical_rotate) base_link camera_link 5" />

I tried to publish it faster but it only slows the moment when the topic starts to low the frequency

How can I check if the clock of the published images is different than the static transform that I'm publishing? and it that case, How can I synchronize them?

matlabbe, 2023-06-04:
What is the driver of the stereo camera? Do they have parameters to tell the driver to use Host timestamp and not the camera timestamp?

If the camera driver publishes images with latency over 5 ms, that static transform is fast enough.

closed 2023-06-06 by JajamesVid