introlab/rtabmap_ros #976: Interpretation of stats_values

sagardhatrak, 2023-06-01:
Hi...

What is interpretation of two stats_values 1. - RtabmapROS/TimeTotal/ms and 2. Timing/Total/ms?? 
Please explain details about these.

matlabbe, 2023-06-04:
- `Timing/Total/ms` is the total time of rtabmap internal processing loop ([Rtabmap::process()](https://github.com/introlab/rtabmap/blob/1b67d6a86a9b0585dec48da2e27de9f8f6042dcf/corelib/src/Rtabmap.cpp#LL1158C6-L1158C6))
- `RtabmapROS/TimeTotal/ms` includes the rtabmap internal processing loop + map management (updating occupancy grid) + ros publishing stuff, see [here](https://github.com/introlab/rtabmap_ros/blob/1bf27fb2d9160d0cfb86c30430862264843edf43/rtabmap_slam/src/CoreWrapper.cpp#L2226) (note that this doesn't include odometry, which is done in another node).

sagardhatrak, 2023-06-05:
Thank you,

1. How rtabmap/detection rate is related with Timing/total/ms and Rtabmap/TimeTotal/ms?
2. For a single value of Timing/total/ms and Rtabmap/TimeTotal/ms, how many frames are assumed to be processed? Is it a single frame? If not, what is the number of frames processed in one bunch of data while calculating both Timings values?

matlabbe, 2023-06-06:
1. If `Timing/Total/ms > 1000` and `Rtabmap/DetectionRate = 1`, the output rate won't be 1 Hz anymore, but less.
2. Single frame

sagardhatrak, 2023-06-07:
1.	Can you please explain detail about first answer? We want to understand, Is there any impact of setting detection rate value on Timing/Total/ms of rtabmap node?
2.	How can output rate be measured in terms of fps?
3.	Can we measure the output rate using rtabmap/MapData? Is it updating and publishing after processing of each frame?
4.	We measured the output rate using RtabmapROS/TimeTotal/ms. we got output rate around 7-8fps with detection rate 30. Why output rate is low? It should be around 30.
5.	RtabmapROS/TimeTotal/ms should be greater than Timing/Total/ms but some time Timing/Total/ms found greater. Why did Timing/Total/ms found to be greater?
6.	Can suggest better way to measure the output rate in terms of fps.

matlabbe, 2023-06-11:
1. `Timing/Total/ms` will increase accordingly to map size. The impact is that with higher rate, you had more data to the map, thus `Timing/Total/ms` will increase faster over time.
2. `rostopic hz /rtabmap/info`
3. Yes, you can use `rostopic hz /rtabmap/mapData` as well.
4. It means rtabmap cannot run in real-time at this rate. I rarely increase rate over 2 hz.
5. Are you comparing results in the same message? As shown here:
https://github.com/introlab/rtabmap_ros/blob/1a87f1598df2a40d01b230808074b45c9a26039e/rtabmap_slam/src/CoreWrapper.cpp#L2248
The ROS total includes internal total processing time. If it is smaller than `Timing/Total/ms`, there is something wrong (bug?). Well, we don't use exactly `Timing/Total/ms` in the cumulative time, but I would assume that the `timeRtabmap` below estimates the same time than `Timing/Total/ms`:
https://github.com/introlab/rtabmap_ros/blob/1a87f1598df2a40d01b230808074b45c9a26039e/rtabmap_slam/src/CoreWrapper.cpp#L2025-L2028

6. I am not sure why you want to estimate the FPS of the mapping node. The best way is `rostopic hz` or you drive it from inverse of `RtabmapROS/TimeTotal/ms`. For odometry, I would suggest to check the outputs of the odometry node, this one is always trying to process frames as fast as possible.

sagardhatrak, 2023-07-03:
Thank you for guidance.

for question 5:
5. we are comparing the results in the same message.

As per your suggestion, we are going with rostopic hz.

closed 2023-07-04 by matlabbe