introlab/rtabmap_ros #977: rtabmap_slam crashes when using Superglue and Superpoint

GVMafeedme, 2023-06-01:
I am testing Superglue with Superpoint, I built rtabmap from source, but when I set the parameters to use Superpoint and Superglue, it crashes instantly. I am using docker with access to gpu (gazebo launches correctly) and has built rtabmap bot for gpu and cpu, here is the Dockerfile for cpu:
```Dockerfile
FROM osrf/ros:noetic-desktop-full
SHELL ["/bin/bash", "-c"]

# ROS INSTALL 
RUN apt-get update && apt-get install apt-utils -y
RUN apt-get install wget git ros-noetic-catkin python3-catkin-tools python-is-python3 python3-pip curl -y

# RTABMAP INSTALL
RUN apt-get install  libopencv-dev python3-opencv ros-noetic-vision-opencv -y

# TUTORIAL PACKAGES INSTALL
RUN apt-get install ros-noetic-grid-map -y

# rebuild rtabmap with torch
RUN apt-get install ros-noetic-rtabmap ros-noetic-rtabmap-ros -y
RUN apt-get remove  ros-noetic-rtabmap ros-noetic-rtabmap-ros -y

RUN apt-get install software-properties-common -y
RUN rm -rf /var/lib/apt/lists/*
RUN add-apt-repository ppa:borglab/gtsam-release-4.0
RUN apt-get update && apt-get install libgtsam-dev libgtsam-unstable-dev -y

WORKDIR /torch
RUN wget https://download.pytorch.org/libtorch/cpu/libtorch-cxx11-abi-shared-with-deps-2.0.1%2Bcpu.zip
RUN apt-get install unzip -y
RUN unzip libtorch-cxx11-abi-shared-with-deps-2.0.1+cpu.zip

WORKDIR /
RUN git clone -b 0.21.0 https://github.com/introlab/rtabmap.git rtabmap
WORKDIR /rtabmap/build
RUN export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/torch/libtorch/lib && \
    CXXFLAGS="-I/usr/include/python3.8/" \
    cmake -D WITH_TORCH=ON \
          -D WITH_PYTHON=ON \
          -D Torch_DIR=/torch/libtorch/share/cmake/Torch \
        .. && make -j8 && make install

WORKDIR /ros_ws
RUN git clone --branch noetic-devel https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
RUN cp -r /torch/libtorch/lib/ /usr/local/lib/libtorch
RUN source devel/setup.bash && \
    export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/torch/libtorch/lib && \
    catkin_make -DRTABMAP_SYNC_MULTI_RGBD=ON -DRTABMAP_SYNC_USER_DATA=ON

ENV LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/torch/libtorch/lib:/usr/local/lib/

WORKDIR /torch/superpoint
RUN wget https://raw.githubusercontent.com/KinglittleQ/SuperPoint_SLAM/master/superpoint.pt

CMD [ "bash" ]
```
The launch group I used is this one:
```xml
<group ns="rtabmap">
    <node pkg="rtabmap_sync" type="rgbd_sync" name="rgbd_sync" output="screen">
      <remap from="rgb/image"         to="/camera/color/image_raw"/>
      <remap from="depth/image"       to="/camera/depth/image_raw"/>
      <remap from="rgb/camera_info"   to="/camera/color/camera_info"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="$(arg rtabmap_args)">
      <param name="database_path"       type="string" value="$(arg database_path)"/>
      <param name="frame_id"            type="string" value="base_link"/>
      <param name="subscribe_rgb"   type="bool" value="false"/>
      <param name="subscribe_depth" type="bool" value="false"/>
      <param name="subscribe_rgbd" type="bool" value="true"/>
      <!-- <param unless="$(eval model=='waffle')" name="subscribe_rgbd" type="bool" value="false"/> -->
      <param name="subscribe_scan"      type="bool"   value="true"/>
[... 110 lines cut ...]
[DEBUG] (2023-06-01 10:07:08.519) OctoMap.cpp:338::OctoMap() fullUpdate_         =true
[DEBUG] (2023-06-01 10:07:08.519) OctoMap.cpp:339::OctoMap() updateError_        =0.010000
[DEBUG] (2023-06-01 10:07:08.519) OctoMap.cpp:340::OctoMap() rangeMax_           =0.000000
[DEBUG] (2023-06-01 10:07:08.519) OctoMap.cpp:341::OctoMap() rayTracing_         =false
[DEBUG] (2023-06-01 10:07:08.519) OctoMap.cpp:342::OctoMap() emptyFloodFillDepth_=0
[DEBUG] (2023-06-01 10:07:08.519) Rtabmap.cpp:317::init() path=/home/germanvega/.ros/rtabmap.db
[ INFO] (2023-06-01 10:07:08.519) Rtabmap.cpp:322::init() Using database "/home/germanvega/.ros/rtabmap.db".
[DEBUG] (2023-06-01 10:07:08.520) Rtabmap.cpp:4552::setWorkingDirectory() Comparing new working directory path "/home/germanvega/.ros" with ""
[DEBUG] (2023-06-01 10:07:08.520) Rtabmap.cpp:289::setupLogFiles() Log disabled!
[DEBUG] (2023-06-01 10:07:08.520) SuperPoint.cc:122::SPDetector() modelPath=/torch/superpoint/superpoint.pt thr=0.010000 nms=1 cuda=0
[ INFO] [1685614028.577616840]: Parameters are not saved! (No configuration file provided...)
rtabmap: Saving database/long-term memory... (located at /home/germanvega/.ros/rtabmap.db)
[ INFO] (2023-06-01 10:07:08.577) Rtabmap.cpp:460::close() databaseSaved=1
[ INFO] (2023-06-01 10:07:08.577) Rtabmap.cpp:6190::clearPath() status=0
[DEBUG] (2023-06-01 10:07:08.578) Rtabmap.cpp:549::parseParameters() 
[DEBUG] (2023-06-01 10:07:08.578) Rtabmap.cpp:289::setupLogFiles() Log disabled!
[DEBUG] (2023-06-01 10:07:08.578) Rtabmap.cpp:678::parseParameters() new detector strategy 2
[DEBUG] (2023-06-01 10:07:08.578) BayesFilter.cpp:114::setPredictionLC() predictionEpsilon = 0.000000
[DEBUG] (2023-06-01 10:07:08.578) BayesFilter.cpp:114::setPredictionLC() predictionEpsilon = 0.000000
rtabmap: Saving database/long-term memory...done! (located at /home/germanvega/.ros/rtabmap.db, 0 MB)
../src/intel/isl/isl.c:2105: FINISHME: ../src/intel/isl/isl.c:isl_surf_supports_ccs: CCS for 3D textures is disabled, but a workaround is available.
[rtabmap/rtabmap-7] process has died [pid 4168, exit code -11, cmd /ros_ws/devel/lib/rtabmap_slam/rtabmap -d --udebug move_base:=/move_base scan:=/scan odom:=/odom rgbd_image:=rgbd_image grid_map:=/map __name:=rtabmap __log:=/home/germanvega/.ros/log/04a71064-0063-11ee-b0d6-0242ac120002/rtabmap-rtabmap-7.log].
log file: /home/germanvega/.ros/log/04a71064-0063-11ee-b0d6-0242ac120002/rtabmap-rtabmap-7*.log
```
I tried with version ```0.21.1``` but it does crash instantly as well. That log file referenced in the end does not exist in my computer.

matlabbe, 2023-06-04:
A while ago I did try it with docker and it worked in ros (using D435i camera live). See https://github.com/introlab/rtabmap/issues/896#issuecomment-1257085425
Never tried it with torch using only CPU. Did you try starting from pytorch CPU docker image instead of the ros one?

GVMafeedme, 2023-06-12:
Thank you! I found the issue: the weights are compatible with a few versions of torch only, so all the weights should be updated, you gave the answer here https://github.com/introlab/rtabmap_ros/issues/957

closed 2023-06-12 by matlabbe

matlabbe, 2024-04-28:
(Adding this note here for reference) There is a superpoint/superglue rtabmap_ros docker image here for convenience: https://github.com/introlab/rtabmap_ros/blob/master/docker/noetic/superpoint/Dockerfile