introlab/rtabmap_ros #978: Detection rate and system performance

sagardhatrak, 2023-06-02:
1.	What does detection rate do in rtabmap? 
2.	Does detection rate select only a single image from set of images if detection rate=1? How does it select single image from set of images if detection rate=1? 
3.	When input frames are given to rtabmap, which parameter determines the time required to create the map and generate optimized poses?  
4.	How can we determine the frame rate of the map and the frame rate of the optimized poses?
5.	How can optimized poses be published as a topic?  
6.	Is there any parameter or value that we can use to monitor system performance while running rtabmap? 
7.	Can you suggest a better approach for evaluating system performance when using rtabmap on an Nvidia orin board?

matlabbe, 2023-06-05:
1. The rate at which a new node is added to graph, and thus at which rate loop closure detection is done (which was the original meaning when rtabmap was just loop closure detection approach). Note that for odometry node, it is independent of this. Odometry rate will be as fast as possible.
2. 1 Hz means 1 image each second is taken. There is no particular selection parameters, it is just by time.
3. It is the main loop of rtabmap, see figure 1 of this [paper](https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf).
4. The rate is the same for both, which is defined by `Rtabmap/DetectionRate` parameter. The thing is that if memory management is disabled (default), as the map grows, it may take longer time than 1 second even if the rate is set to 1 Hz. So the output rate may decrease under 1 Hz. 
5. `/rtabmap/mapGraph` contains the optimized poses
6. If you use rtabmap_viz, open Statistics view. 
7. On any computers, I usually use the Statistics panel from rtabmap_viz to see which process in rtabmap is using the most time.

fbliman, 2023-07-05:
I am trying superpoint on an only cpu pc and though I lowered the detection rate to 0.3 (3s to process the images aprox) and because of that change I am getting this warning:
' 
[ WARN] [1688566339.927395399, 1657118734.147511635]: /rtabmap/rtabmapviz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rtabmapviz subscribed to (exact sync):
   /stereo_odometry \
   /stereo_camera/left/image_rect_color \
   /stereo_camera/right/image_rect \
   /zed_front/zed_node_front/left/camera_info \
   /zed_front/zed_node_front/right/camera_info \
   /rtabmap/odom_info
'
and I do not see the image any more on the rtabmap viz

thanks

matlabbe, 2023-07-10:
Do you use superpoint for `stereo_odometry`? I would use the default features for visual odometry, and use only superpoint on rtbmap node.

If `/setereo_odometry` topic is super slow, like 3 sec updates, and images are published at 30 Hz, you would need to increase the queue_size to at least 90 so that approximate synchronizer can sync the topics. You can compare the framerate of the topics with:
```
/stereo_odometry
/stereo_camera/left/image_rect_color
/stereo_camera/right/image_rect
/zed_front/zed_node_front/left/camera_info
/zed_front/zed_node_front/right/camera_info
/rtabmap/odom_info
```