introlab/rtabmap_ros #982: /usr/bin/ld: warning: libsl_zed.so, needed by /usr/local/lib/librtabmap_core.so.0.21.1, not found (try using -rpath or -rpath-link)

Rubeshthirumani, 2023-06-12:
rtabmap_ros catkin_make issues with zed library

matlabbe, 2023-06-17:
It seems that `libsl_zed.so` is not on the LD_LIBRARY_PATH. If you are using ROS, you may also not build rtabmap library with ZED -> `cmake -DWITH_ZED=OFF`

Rubeshthirumani, 2023-06-23:
I still didn't try that. But solved with this method [https://community.stereolabs.com/t/usr-bin-ld-warning-libsl-zed-so-needed-by-usr-local-lib-librtabmap-core-so-0-21-1-not-found-try-using-rpath-or-rpath-link/2793/4?u=thiru](url)

matlabbe, 2023-06-23:
Humm, maybe RTABMapConfig.cmake should look for ZED dependency, though libraries should be private inside rtabmap. I'll give a try.

matlabbe, 2023-06-25:
retested on intel computer, it should compile/link correctly even if rtabmap library has been built with zed support. At worst, set explicitly the LD_LIBRARY_PATH:
```
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/zed/lib
```

closed 2023-06-25 by matlabbe