introlab/rtabmap_ros #983: There is still no example of two cameras for ROS2? no launch file

AdrianSiGmA, 2023-06-12:
I am trying to do an application with 3 realsense cameras, but I haven't found yet a solution for the remappings in ROS2. Is is possible?

matlabbe, 2023-06-17:
Yes, it should be possible. Is odometry computed externally? 

Here two examples, assuming using Depth-IR mode for realsense (IR emitter disabled). Make sure rtabmap_ros is built with `DRTABMAP_SYNC_MULTI_RGBD`:
```
colcon build --symlink-install --cmake-args -DRTABMAP_SYNC_MULTI_RGBD=ON -DCMAKE_BUILD_TYPE=Release
```

With external odometry:
```py
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():

    return LaunchDescription([

        Node(
            package='rtabmap_sync', executable='rgbd_sync', name='rgbd_sync1', output='screen',
            namespace='/realsense_camera1',
            parameters=[{'approx_sync':False}],
            remappings=[('rgb/image',       'infra1/image_rect_raw'),
                        ('rgb/camera_info', 'infra1/camera_info'),
                        ('depth/image',     'depth/image_rect_raw')]),
        
        Node(
            package='rtabmap_sync', executable='rgbd_sync', name='rgbd_sync2', output='screen',
            namespace='/realsense_camera2',
            parameters=[{'approx_sync':False}],
            remappings=[('rgb/image',       'infra1/image_rect_raw'),
                        ('rgb/camera_info', 'infra1/camera_info'),
                        ('depth/image',     'depth/image_rect_raw')]),

        Node(
            package='rtabmap_slam', executable='rtabmap', output='screen',
            parameters=[{'approx_sync':True, 'subscribe_rgbd':True, 'rgbd_cameras':2}],
            remappings=[('rgbd_image0',     '/realsense_camera1/rgbd_image'),
                        ('rgbd_image1',     '/realsense_camera2/rgbd_image'),
                        ('odom',            '/odom')],
            arguments=['-d']),
    ])
```

With RTAB-Map's visual odometry:
```py
from launch import LaunchDescription
from launch_ros.actions import Node

def generate_launch_description():

    return LaunchDescription([

        Node(
            package='rtabmap_sync', executable='rgbd_sync', name='rgbd_sync1', output='screen',
            namespace='/realsense_camera1',
            parameters=[{'approx_sync':False}],
            remappings=[('rgb/image',       'infra1/image_rect_raw'),
                        ('rgb/camera_info', 'infra1/camera_info'),
                        ('depth/image',     'depth/image_rect_raw')]),
        
        Node(
            package='rtabmap_sync', executable='rgbd_sync', name='rgbd_sync2', output='screen',
            namespace='/realsense_camera2',
            parameters=[{'approx_sync':False}],
            remappings=[('rgb/image',       'infra1/image_rect_raw'),
                        ('rgb/camera_info', 'infra1/camera_info'),
                        ('depth/image',     'depth/image_rect_raw')]),
        
        Node(
            package='rtabmap_odom', executable='rgbd_odometry', output='screen',
            parameters=[{'approx_sync':True, 'subscribe_rgbd':True, 'rgbd_cameras':2}],
            remappings=[('rgbd_image0',     '/realsense_camera1/rgbd_image'),
                        ('rgbd_image1',     '/realsense_camera2/rgbd_image')]),

        Node(
            package='rtabmap_slam', executable='rtabmap', output='screen',
            parameters=[{'approx_sync':True, 'subscribe_rgbd':True, 'rgbd_cameras':1}],
            remappings=[('rgbd_image',     'rgbd_image'),
                        ('odom',           'odom_rgbd_image')],
            arguments=['-d']),
    ])
```

Note that in this example the realsense cameras would be started in different namespaces. You should also provide TF between the cameras and/or base frame of the robot.

AdrianSiGmA, 2023-06-20:
Hello, thank you for the reply! still having some issues
The following comments:

1. Is the rgbd_image topic needed? how is this Depth-IR mode for the realsense?, or how do I disable the IR emitter? I ask this because, in the past (only for one camera), I just neded the color/image_rect_raw, the depth/image_rect_raw, and the color/camera_info topics.
2. Indeed, the odometry is calculated with the RTAB-Map's visual odometry, so I am using the second option you provided.
3. The TF and the namespaces of the cameras are properly set :)

I couldn't make it work, because it gives me this error and I didn't really get what settings it means: 
```
[rgbd_odometry-3] [ERROR] (2023-06-20 13:33:33.082) RegistrationVis.cpp:1528::computeTransformationImpl() Multi-camera 2D-3D PnP registration is only available if rtabmap is built with OpenGV dependency. Use 3D-3D registration approach instead for multi-camera.
[rgbd_odometry-3] [ WARN] (2023-06-20 13:33:33.082) OdometryF2M.cpp:571::computeTransform() Unknown registration error
[rgbd_odometry-3] [INFO] [1687260813.156750301] [rgbd_odometry]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.071391s
```

I built rtabmap with the multi rgbd camera function, but it still gives me this issue. Is there any other thing I am missing? I have tried modifying a lot the launch file. 

Many thanks!

matlabbe, 2023-06-21:
Verify that `/realsense_camera1/rgbd_image` and `/realsense_camera2/rgbd_image` are both publish with `ros2 topic hz`. If so, you would likely need to set `approx_sync` for `rgbd_odometry` to true (in your log you have exact sync for rgbd_odometry).

adricort, 2023-07-25:
Herebz I confirm that this issue has been closed thanks to this pull request: #1002

closed 2023-07-25 by adricort