introlab/rtabmap_ros #984: Mapping an entire area without RTAB-Map crashing

AnoushkaThakur, 2023-06-14:
Hi Mathieu,
I have been trying to map my world of dimension 100*100 with a laptop of 13.5 GB Ram (swap memory 2 gb) of with NVIDIA Graphics with a ryzen graphics card.
Usually during the mapping phase the laptop takes the entire memory and then crashes down. Leaving most of the area unmapped how can i map the entire world without RTAB-Map crashing and logging in the memory. I am also fairly new RTAB-Map
I am including my launch file along with the world i want to map. 
Thanks 

[test_nav.txt](https://github.com/introlab/rtabmap_ros/files/11744896/test_nav.txt)

 
[image]

AnoushkaThakur, 2023-06-19:
I made changes in that parameter, made it to 500
I also changed <param name="Mem/BinDataKept"            type="bool" value="false"/> (with it default value being true)
making this false allowed me to map the entire area without it crashing.
I am sharing the database with you.
But anyways I did notice my laptop's ram taking the entire before using this parameter.

[image]
also attaching a drive link for you to see the db file . 
Have shared those via mail!

AnoushkaThakur, 2023-06-19:
> Do you have a database to share?
>
> This is huge:
>
> <param name="Kp/MaxFeatures"             type="string" value="8000"/>
>
> 500 would be more reasonable.
>
> note also that a symmetric simulated environment with same texture
> everywhere is very challenging for visual loop closure detection.

graph.png
<https://drive.google.com/file/d/1aeunVxG4kU1_8sRqbE-wC3w1pYPUdOnO/view?usp=drive_web>
 rtabmap1.db
<https://drive.google.com/file/d/1XgkUj0SyUznwfCc-kF0jKJnSKSEhR8eS/view?usp=drive_web>
 rtabmap1.pgm
<https://drive.google.com/file/d/1jmVy8acQkUrfD60qqEDZGfimZ-RhIlhU/view?usp=drive_web>
 rtabmap1.yaml
<https://drive.google.com/file/d/1aVbPd-IRY53Zwi8AZGJn252RfLW3BNWF/view?usp=drive_web>
Sharing files here

matlabbe, 2023-06-20:
Did you have also `rtabmap_viz` opened? You may close `rtabmap_viz`and use rviz with only specific topics. Look also with the System Monitor or `top` to see which process is using the most RAM.

AnoushkaThakur, 2023-06-20:
I have done that as well, using the parameter Mem/BinDataKept = "False" has allowed rtabmap not to log more memory but it doesn't provide a 3D map. Here's a screenshot of top to see which process is taking most memory.
When the parameter Mem/BinDataKept is set to true rtabmap takes up the entire ram in just two rounds of the world (back and forth) 
How can i map the entire world with an occupancy grid without rtabmap taking up the entire memory and crashing before proper completion ?
Will it be possible to discuss some issues over a one-on-one? 

[image]

matlabbe, 2023-06-21:
With `Mem/BinDataKept=False`, the occupancy grid would still be created. What is the resolution of the camera? You can also set `Mem/ImagePostDecimation>1` to reduce the resolution of the data kept in memory/database, while being able to show 3D cloud of the map.

AnoushkaThakur, 2023-06-21:
With Mem/BinDataKept =False, I wasn't able to create the 3D occupancy grid since it showed that the point cloud was empty! When kept True i was able to generate an 3D occupancy grid. 
my resolution is kept at 1920 width and 1080 height, should it be less?

matlabbe, 2023-06-22:
1920x1080 is huge, modify your simulated environment to publish 640x480 images. Most default parameters are based on that resolution.

AnoushkaThakur, 2023-06-22:
Thanks, this helped!

matlabbe, 2023-06-22:
Well, it seems you are using odometry topic from gazebo, the `Odom` parameters won't affect anything.