introlab/rtabmap_ros #988: cannot launch node of type [rtabmap_ros/rtabmap]: Cannot locate node of type [rtabmap] in package [rtabmap_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)

marhwy, 2023-06-27:
Can someone help me with this problem?thank,my installation environment is  ubuntu20.04+noetic
ERROR: cannot launch node of type [rtabmap_ros/rtabmap]: Cannot locate node of type [rtabmap] in package [rtabmap_ros]. Make sure file exists in package path and permission is set to executable (chmod +x)
type is rtabmap_ros/rgbd_sync
[ERROR] [1687863704.736196698]: Failed to load nodelet [/rtabmap/rgbd_sync] of type [rtabmap_ros/rgbd_sync] even after refreshing the cache: According to the loaded plugin descriptions the class rtabmap_ros/rgbd_sync with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet apriltag_ros/ContinuousDetector astra_camera/AstraDriverNodelet astra_camera/driver depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/point_cloud_xyzrgb_radial depth_image_proc/register depthimage_to_laserscan/DepthImageToLaserScanNodelet image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image laser_filters/scan_to_cloud_filter_chain libuvc_camera/driver mesh_filter/DepthSelfFiltering nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_legacy/data_odom_sync rtabmap_legacy/data_throttle rtabmap_legacy/obstacles_detection_old rtabmap_legacy/stereo_throttle rtabmap_legacy/undistort_depth rtabmap_odom/icp_odometry rtabmap_odom/rgbd_odometry rtabmap_odom/rgbdicp_odometry rtabmap_odom/stereo_odometry rtabmap_slam/rtabmap rtabmap_sync/rgb_sync rtabmap_sync/rgbd_sync rtabmap_sync/rgbdx_sync rtabmap_sync/rtabmap rtabmap_sync/stereo_sync rtabmap_util/disparity_to_depth rtabmap_util/imu_to_tf rtabmap_util/lidar_deskewing rtabmap_util/obstacles_detection rtabmap_util/point_cloud_aggregator rtabmap_util/point_cloud_assembler rtabmap_util/point_cloud_xyz rtabmap_util/point_cloud_xyzrgb rtabmap_util/pointcloud_to_depthimage rtabmap_util/rgb_sync rtabmap_util/rgbd_relay rtabmap_util/rgbd_split rtabmap_util/stereo_sync stereo_image_proc/disparity stereo_image_proc/point_cloud2
[ERROR] [1687863704.738479349]: The error before refreshing the cache was: According to the loaded plugin descriptions the class rtabmap_ros/rgbd_sync with base class type nodelet::Nodelet does not exist. Declared types are  RobotLocalization/EkfNodelet RobotLocalization/NavSatTransformNodelet RobotLocalization/UkfNodelet SlamGMappingNodelet apriltag_ros/ContinuousDetector astra_camera/AstraDriverNodelet astra_camera/driver depth_image_proc/convert_metric depth_image_proc/crop_foremost depth_image_proc/disparity depth_image_proc/point_cloud_xyz depth_image_proc/point_cloud_xyz_radial depth_image_proc/point_cloud_xyzi depth_image_proc/point_cloud_xyzi_radial depth_image_proc/point_cloud_xyzrgb depth_image_proc/point_cloud_xyzrgb_radial depth_image_proc/register depthimage_to_laserscan/DepthImageToLaserScanNodelet image_proc/crop_decimate image_proc/crop_nonZero image_proc/crop_non_zero image_proc/debayer image_proc/rectify image_proc/resize image_publisher/image_publisher image_rotate/image_rotate image_view/disparity image_view/image laser_filters/scan_to_cloud_filter_chain libuvc_camera/driver mesh_filter/DepthSelfFiltering nodelet_tutorial_math/Plus pcl/BAGReader pcl/BoundaryEstimation pcl/ConvexHull2D pcl/CropBox pcl/EuclideanClusterExtraction pcl/ExtractIndices pcl/ExtractPolygonalPrismData pcl/FPFHEstimation pcl/FPFHEstimationOMP pcl/MomentInvariantsEstimation pcl/MovingLeastSquares pcl/NodeletDEMUX pcl/NodeletMUX pcl/NormalEstimation pcl/NormalEstimationOMP pcl/NormalEstimationTBB pcl/PCDReader pcl/PCDWriter pcl/PFHEstimation pcl/PassThrough pcl/PointCloudConcatenateDataSynchronizer pcl/PointCloudConcatenateFieldsSynchronizer pcl/PrincipalCurvaturesEstimation pcl/ProjectInliers pcl/RadiusOutlierRemoval pcl/SACSegmentation pcl/SACSegmentationFromNormals pcl/SHOTEstimation pcl/SHOTEstimationOMP pcl/SegmentDifferences pcl/StatisticalOutlierRemoval pcl/VFHEstimation pcl/VoxelGrid rtabmap_legacy/data_odom_sync rtabmap_legacy/data_throttle rtabmap_legacy/obstacles_detection_old rtabmap_legacy/stereo_throttle rtabmap_legacy/undistort_depth rtabmap_odom/icp_odometry rtabmap_odom/rgbd_odometry rtabmap_odom/rgbdicp_odometry rtabmap_odom/stereo_odometry rtabmap_slam/rtabmap rtabmap_sync/rgb_sync rtabmap_sync/rgbd_sync rtabmap_sync/rgbdx_sync rtabmap_sync/rtabmap rtabmap_sync/stereo_sync rtabmap_util/disparity_to_depth rtabmap_util/imu_to_tf rtabmap_util/lidar_deskewing rtabmap_util/obstacles_detection rtabmap_util/point_cloud_aggregator rtabmap_util/point_cloud_assembler rtabmap_util/point_cloud_xyz rtabmap_util/point_cloud_xyzrgb rtabmap_util/pointcloud_to_depthimage rtabmap_util/rgb_sync rtabmap_util/rgbd_relay rtabmap_util/rgbd_split rtabmap_util/stereo_sync stereo_image_proc/disparity stereo_image_proc/point_cloud2
[rtabmap/rgbd_sync-2] process has died [pid 3136, exit code 255, cmd /opt/ros/noetic/lib/nodelet/nodelet standalone rtabmap_ros/rgbd_sync rgb/image:=/camera/rgb/image_raw depth/image:=/camera/depth/image rgb/camera_info:=/camera/rgb/camera_info rgbd_image:=rgbd_image rtabmap/get_map_data:=get_map_data __name:=rgbd_sync __log:=/home/wheeltec/.ros/log/0026b1a8-14da-11ee-9832-48b02dd8cd8e/rtabmap-rgbd_sync-2.log].
log file: /home/wheeltec/.ros/log/0026b1a8-14da-11ee-9832-48b02dd8cd8e/rtabmap-rgbd_sync-2*.log

matlabbe, 2023-06-27:
You would have to update your launch files. The nodes have moved to different sub packages. See this compatibility table: http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2

marhwy, 2023-06-28:
Thank you very much for your timely answer

closed 2023-06-29 by matlabbe

X1yz, 2024-04-11:
> You would have to update your launch files. The nodes have moved to different sub packages. See this compatibility table: http://wiki.ros.org/rtabmap_ros#rtabmap_ros.2Fnoetic_and_newer.Migration_Guide_New_Interface_Noetic.2FROS2

I have modified the "pkg=rtab_slam" in noetic,but the problem of failure of starting the node is still here

matlabbe, 2024-04-11:
@X1yz do you mean `pkg="rtabmap_slam"`?

X1yz, 2024-04-12:
> @X1yz do you mean `pkg="rtabmap_slam"`?
the problem was not solved with the attempt to amend the launch file merely.There is another problem of environment variable.Firstly ,I deleted the workspace in bash file.Then I use pkg="rtabmap_slam",succeed.

tester23x, 2024-08-29:
Hi, would you explain more about deleting the workspace in bash file supposed to be like? I am still having the problem with the environment maybe

matlabbe, 2024-09-05:
To cleanup your ros workspace (e.g., ~/catkin_ws):
```
rm -rf ~/catkin_ws/devel
rm -rf ~/catkin_ws/build
```