introlab/rtabmap_ros #989: Error when building for ros2 humble on ubuntu 22 - CMake Error "fastcdr"

Matt-Collins-52, 2023-06-27:
I am trying to install on ubuntu 22 with ros2 humble. I have verified that ros has been set up correctly.
I have followed the installation instructions:
```
cd ~/ros2_ws
git clone https://github.com/introlab/rtabmap.git src/rtabmap
git clone --branch ros2 https://github.com/introlab/rtabmap_ros.git src/rtabmap_ros
rosdep update && rosdep install --from-paths src --ignore-src -r -y
export MAKEFLAGS="-j6" # Can be ignored if you have a lot of RAM (>16GB)
colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release
```

I have also done `export MAKEFLAGS="-j1` with the same output.
When running `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release` I get this output to terminal:

```
Starting >>> rtabmap 
Starting >>> rtabmap_msgs
Starting >>> rtabmap_python
--- stderr: rtabmap_python                                                                                        
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< rtabmap_python [1.23s]
--- stderr: rtabmap                                                                           
CMake Warning (dev) at /usr/share/cmake-3.22/Modules/FindPackageHandleStandardArgs.cmake:438 (message):
  The package name passed to `find_package_handle_standard_args` (CSPARSE)
  does not match the name of the calling package (G2O).  This can lead to
  problems in calling code that expects `find_package` result variables
  (e.g., `_FOUND`) to follow a certain pattern.
Call Stack (most recent call first):
  cmake_modules/FindG2O.cmake:9 (find_package_handle_standard_args)
  CMakeLists.txt:484 (FIND_PACKAGE)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< rtabmap [3.14s]
Finished <<< rtabmap_msgs [5.61s]                      
Starting >>> rtabmap_conversions
--- stderr: rtabmap_conversions                          
CMake Error at /home/matt/ros2_humble/install/builtin_interfaces/share/builtin_interfaces/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package):
  By not providing "Findfastcdr.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "fastcdr", but
  CMake did not find one.

  Could not find a package configuration file provided by "fastcdr" with any
  of the following names:

    fastcdrConfig.cmake
    fastcdr-config.cmake

  Add the installation prefix of "fastcdr" to CMAKE_PREFIX_PATH or set
  "fastcdr_DIR" to a directory containing one of the above files.  If
  "fastcdr" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  /home/matt/ros2_humble/install/builtin_interfaces/share/builtin_interfaces/cmake/builtin_interfacesConfig.cmake:41 (include)
  /home/matt/ros2_humble/install/sensor_msgs/share/sensor_msgs/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /home/matt/ros2_humble/install/sensor_msgs/share/sensor_msgs/cmake/sensor_msgsConfig.cmake:41 (include)
  /opt/ros/humble/share/cv_bridge/cmake/ament_cmake_export_dependencies-extras.cmake:21 (find_package)
  /opt/ros/humble/share/cv_bridge/cmake/cv_bridgeConfig.cmake:41 (include)
  CMakeLists.txt:9 (find_package)

---
Failed   <<< rtabmap_conversions [0.41s, exited with code 1]

Summary: 3 packages finished [6.22s]
  1 package failed: rtabmap_conversions
  3 packages had stderr output: rtabmap rtabmap_conversions rtabmap_python
  10 packages not processed
```

matlabbe, 2023-06-27:
Install `ros-humble-fastcdr`, not sure why it failed as rtabmap doesn't explicitly depends on that package.
```
sudo apt install ros-humble-fastcdr
```

Matt-Collins-52, 2023-06-27:
When running 
`sudo apt install ros-humble-fastcdr`
I get:
`Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
ros-humble-fastcdr is already the newest version (1.0.24-2jammy.20230425.230328).
ros-humble-fastcdr set to manually installed.
The following packages were automatically installed and are no longer required:
  libactionlib1d libarmadillo9 libb64-0d libboost-filesystem1.71.0
  libboost-iostreams1.71.0 libboost-program-options1.71.0 libboost-regex1.71.0
  libboost-thread1.71.0 libbullet2.88 libcfitsio8 libclass-loader1d libcpp-common0d
  libdap25 libdap27 libdapclient6v5 libdart6 libdart6-collision-bullet
  libdart6-external-odelcpsolver libdart6-utils libepsilon1 libfcl0.5 libgazebo11
  libgdal26 libhdf5-103 libicu66 libignition-common3 libignition-fuel-tools4
  libignition-msgs5 libignition-transport8 libimage-transport0d libinteractive-markers2d
  libjson-c4 libjsoncpp1 liblaser-geometry0d liblog4cxx12 libmessage-filters1d
  libnetcdf15 libodbc1 libogre1.12.10 libpocofoundation80 libpoppler97 libproj15
  libprotobuf17 libqhull7 libqwt-qt5-6 libresource-retriever0d librosconsole-bridge0d
  librosconsole3d libroscpp-serialization0d libroscpp4d libroslib0d librospack0d
  librostime1d librviz6d libsdformat9 libsimbody3.6 libspnav0 libssl1.1 libstb0 libtar0
  libtf1d libtf2-2d libtf2-ros1d libtinyxml2-6a liburdf1d libwebp6 libxmlrpcpp3d libzip5
  odbcinst odbcinst1debian2 sdformat9-sdf
Use 'sudo apt autoremove' to remove them.
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
`

And then I get the same exact error when running `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release`

matlabbe, 2023-06-27:
Looks like more a ros2 issue. Make sure to source setup.bash before colcon:
```
source /opt/ros/humble/setup.bash
```

Matt-Collins-52, 2023-06-28:
I ended up reinstalling ros humble, ran `source /opt/ros/humble/setup.bash` and then followed the installation instructions again. When running `colcon build --symlink-install --cmake-args -DCMAKE_BUILD_TYPE=Release` I now get this error:
```
Starting >>> rtabmap 
Starting >>> rtabmap_msgs
Starting >>> rtabmap_python
Finished <<< rtabmap [0.76s]                                                                                           
--- stderr: rtabmap_python                                                                 
/usr/lib/python3/dist-packages/setuptools/command/easy_install.py:158: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/setuptools/command/install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.16.0-unknown is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 1.1build1 is an invalid version and will not be supported in a future release
  warnings.warn(
/usr/lib/python3/dist-packages/pkg_resources/__init__.py:116: PkgResourcesDeprecationWarning: 0.1.43ubuntu1 is an invalid version and will not be supported in a future release
  warnings.warn(
---
Finished <<< rtabmap_python [1.40s]
Finished <<< rtabmap_msgs [6.71s]                      
Starting >>> rtabmap_conversions
Finished <<< rtabmap_conversions [0.29s]                      
Starting >>> rtabmap_sync
Starting >>> rtabmap_util
Starting >>> rtabmap_rviz_plugins
Finished <<< rtabmap_rviz_plugins [0.31s]                                                                                                   
Finished <<< rtabmap_sync [0.46s]                                                             
Starting >>> rtabmap_viz
Finished <<< rtabmap_util [0.55s]                                                          
Starting >>> rtabmap_odom
Starting >>> rtabmap_slam
Finished <<< rtabmap_viz [0.45s]                                                                                                
Finished <<< rtabmap_odom [0.40s]
--- stderr: rtabmap_slam                              
In file included from /home/matt/Documents/ros/rtabmap_ws/src/rtabmap_ros/rtabmap_slam/src/CoreWrapper.cpp:28:
/home/matt/Documents/ros/rtabmap_ws/src/rtabmap_ros/rtabmap_slam/include/rtabmap_slam/CoreWrapper.h:70:10: fatal error: rtabmap_msgs/msg/landmark_detection.hpp: No such file or directory
   70 | #include "rtabmap_msgs/msg/landmark_detection.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
gmake[2]: *** [CMakeFiles/rtabmap_slam_plugins.dir/build.make:76: CMakeFiles/rtabmap_slam_plugins.dir/src/CoreWrapper.cpp.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:139: CMakeFiles/rtabmap_slam_plugins.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
---
Failed   <<< rtabmap_slam [0.99s, exited with code 2]
                                  
Summary: 9 packages finished [8.84s]
  1 package failed: rtabmap_slam
  2 packages had stderr output: rtabmap_python rtabmap_slam
  4 packages not processed
```

I also have rtabmap_slam installed and everything was just downloaded today. I can also use other ros packages just fine.

matlabbe, 2023-06-29:
This header should be generated by ros2:
https://github.com/introlab/rtabmap_ros/blob/49316bff009e84d13dbcc284c00f32be46e686bb/rtabmap_msgs/CMakeLists.txt#L28

What is generated in your ros2 workspace install directory:
```
~/ros2_ws/install/rtabmap_msgs/include/rtabmap_msgs/msg$ ls
camera_model.h         goal.hpp                landmark_detections.hpp  odom_info.hpp   rgbd_images.hpp
camera_model.hpp       gps.h                   link.h                   path.h          rosidl_generator_c__visibility_control.h
camera_models.h        gps.hpp                 link.hpp                 path.hpp        rosidl_typesupport_fastrtps_cpp__visibility_control.h
camera_models.hpp      info.h                  map_data.h               point2f.h       rosidl_typesupport_fastrtps_c__visibility_control.h
detail                 info.hpp                map_data.hpp             point2f.hpp     rosidl_typesupport_introspection_c__visibility_control.h
env_sensor.h           key_point.h             map_graph.h              point3f.h       scan_descriptor.h
env_sensor.hpp         key_point.hpp           map_graph.hpp            point3f.hpp     scan_descriptor.hpp
global_descriptor.h    landmark_detection.h    node_data.h              rgbd_image.h    user_data.h
global_descriptor.hpp  landmark_detection.hpp  node_data.hpp            rgbd_image.hpp  user_data.hpp
goal.h                 landmark_detections.h   odom_info.h              rgbd_images.h
```

Matt-Collins-52, 2023-07-05:
Within my ~/ros2_ws/install I do not see rtabmap folders, but `ros2 pkg list | grep rtabmap` returns this:

rtabmap_conversions
rtabmap_demos
rtabmap_examples
rtabmap_launch
rtabmap_msgs
rtabmap_odom
rtabmap_python
rtabmap_ros
rtabmap_rviz_plugins
rtabmap_slam
rtabmap_sync
rtabmap_util
rtabmap_viz

anath93, 2023-08-05:
@matlabbe [https://github.com/introlab/rtabmap_ros/issues/970](url) as I had commented in that post, I still have the Landmark detection messages if I build library from source as @Matt-Collins-52 is talking about. I checked the install folder for permissions and it seems write, tried clearing workspace and building individual packages like rtabmap_msgs before slam package but no luck so going a step ahead I saw in CLI for "ros2 show interface | grep rtabmap" I do not see the files being registered also in the system ? Any inputs as I know you mentioned that your CI passed the test.

[image]

matlabbe, 2023-08-06:
Make sure you uninstall any rtabmap_msgs binaries:
```
sudo apt remove ros-$ROS_DISTRO-rtabmap-msgs -y
```
I have the same error if I don't uninstall the binaries one.

Example with docker's rtabmap_ros:humble-latest:
```
docker run -it --rm introlab3it/rtabmap_ros:humble-latest /bin/bash
ros2 interface list | grep rtabmap_msgs
    rtabmap_msgs/msg/CameraModel
    rtabmap_msgs/msg/CameraModels
    rtabmap_msgs/msg/EnvSensor
    rtabmap_msgs/msg/GPS
    rtabmap_msgs/msg/GlobalDescriptor
    rtabmap_msgs/msg/Goal
    rtabmap_msgs/msg/Info
    rtabmap_msgs/msg/KeyPoint
    rtabmap_msgs/msg/LandmarkDetection
    rtabmap_msgs/msg/LandmarkDetections
    rtabmap_msgs/msg/Link
    rtabmap_msgs/msg/MapData
    rtabmap_msgs/msg/MapGraph
    rtabmap_msgs/msg/NodeData
    rtabmap_msgs/msg/OdomInfo
    rtabmap_msgs/msg/Path
    rtabmap_msgs/msg/Point2f
    rtabmap_msgs/msg/Point3f
    rtabmap_msgs/msg/RGBDImage
    rtabmap_msgs/msg/RGBDImages
    rtabmap_msgs/msg/ScanDescriptor
    rtabmap_msgs/msg/UserData
    rtabmap_msgs/srv/AddLink
    rtabmap_msgs/srv/CleanupLocalGrids
    rtabmap_msgs/srv/DetectMoreLoopClosures
    rtabmap_msgs/srv/GetMap
    rtabmap_msgs/srv/GetMap2
    rtabmap_msgs/srv/GetNodeData
    rtabmap_msgs/srv/GetNodesInRadius
    rtabmap_msgs/srv/GetPlan
    rtabmap_msgs/srv/GlobalBundleAdjustment
    rtabmap_msgs/srv/ListLabels
    rtabmap_msgs/srv/LoadDatabase
    rtabmap_msgs/srv/PublishMap
    rtabmap_msgs/srv/RemoveLabel
    rtabmap_msgs/srv/ResetPose
    rtabmap_msgs/srv/SetGoal
    rtabmap_msgs/srv/SetLabel
```

anath93, 2023-08-07:
@matlabbe thank you, build got successful.