introlab/rtabmap_ros #990: Understanding of localization mode

sagardhatrak, 2023-07-03:
Hi..,
1. How does localization happen in the rtabmap with lidar scan?  and It does not detect loop closure. Can you provide detail theoretical understanding? 
2. Loop closure automatically localizes the robot in the generated map with RGBD data. How does it localizes?  which algorithm or logic does it uses? Can you provide theoretical details?

matlabbe, 2023-07-04:
See https://introlab.3it.usherbrooke.ca/mediawiki-introlab/images/7/7a/Labbe18JFR_preprint.pdf (section 3.4)

The only difference between localization mode and SLAM mode is that in localization mode, the map won't be updated after a loop closure or proximity detection.

sagardhatrak, 2023-07-12:
Hi thanks for guidance.

closed 2023-07-14 by matlabbe