introlab/rtabmap_ros #991: Unreliable ground detection

ashBabu, 2023-07-03:
I have an intel realsense d435i camera and my OS is Ubuntu 20.04. I am launching the following 
* `roslaunch realsense2_camera rs_camera.launch filters:=pointcloud`
* `roslaunch obstacle_detection rtabmap_obstacles.launch`

where the contents of the `rtabmap_obstacles.launch` is as follows
```
<launch>

     <node pkg="nodelet" type="nodelet" name="points_xyz" args="standalone rtabmap_util/point_cloud_xyz">-->
       <remap from="depth/image"         to="/camera/depth/image_rect_raw"/>
       <remap from="depth/camera_info"   to="/camera/depth/camera_info"/>
       <remap from="cloud"               to="/camera/depth/color/points"/>
       <param name="decimation"  type="double" value="2"/>
       <param name="voxel_size"  type="double" value="0.05"/>
       <param name="approx_sync" type="bool" value="false"/>
    </node>

    <node pkg="nodelet" type="nodelet" name="obstacles_detection" args="standalone rtabmap_util/obstacles_detection stereo_nodelet">
      <remap from="cloud" to="/camera/depth/color/points"/>

      <param name="frame_id"             type="string" value="camera_link"/>
      <param name="wait_for_transform"   type="bool"   value="true"/>
      <param name="Grid/MinClusterSize"     type="int"    value="20"/>
      <param name="Grid/ClusterRadius"       type="double" value="0.5"/>
      <param name="Grid/MaxObstacleHeight" type="double" value="3.0"/>
    </node>

    <arg name="rvizconfig" default="$(find mk_path_tracking)/rviz/obstacle_detection.rviz" />
    <node name="rviz" pkg="rviz" type="rviz" args="-d $(arg rvizconfig)" required="true"/>

</launch>
```

When I visualize the ground detected in RViz (grey color), a lot of points seems ok while several others are well above the ground as seen in the attached video.

https://github.com/introlab/rtabmap_ros/assets/10324110/27a2ab08-4501-4176-af97-04f111aab055

How can I improve this behaviour? A rosbag is here.

https://www.dropbox.com/s/f0lv5t1pimihm2z/a.bag?dl=0

matlabbe, 2023-07-04:
Try:
```xml
    <node pkg="nodelet" type="nodelet" name="obstacles_detection" args="standalone rtabmap_util/obstacles_detection stereo_nodelet">
      <remap from="cloud" to="/camera/depth/color/points"/>

      <param name="frame_id"             type="string" value="camera_link"/>
      <param name="wait_for_transform"   type="bool"   value="true"/>
      <param name="Grid/MaxGroundHeight"   type="double"   value="0.01"/>
    </node>
```
[image]

ashBabu, 2023-07-04:
Thank you very much @matlabbe for the quick response. It works significantly better now.
Closing this.

closed 2023-07-04 by ashBabu