introlab/rtabmap_ros #996: Localization Issue when Mapping Similar Multiple Floors.

brayanpa, 2023-07-12:
I am trying to map multiple floors that are very similar to each other. For this purpose, I am creating a session for each floor, and It works fine as long as it doesn't see features of another floor. I have been reading and came across the GridGlobal/AltitudeDelta parameter, which I believe could help me with this issue. I have set this parameter to 2.0 and verified that the odometry changes in the Z-axis when transitioning between floors (around 4.0 meters for each floor). However, even with this setting, the robot can still localize itself on the wrong floor.

For example, when the robot is on the fourth floor with a Z-axis odometry value of 12 meters and the GridGlobal/AltitudeDelta parameter set to 2.0, it may erroneously localize itself on a different floor.

I would greatly appreciate your assistance since I could not find much documentation regarding this specific case or the mentioned parameter. Therefore, I am unsure if I am configuring it correctly. Thank you very much.

matlabbe, 2023-07-14:
From the doc:

> Param: GridGlobal/AltitudeDelta = "0"                      [Assemble only nodes that have the same altitude of +-delta meters of the current pose (0=disabled). This is used to generate 2D occupancy grid based on the current altitude (e.g., multi-floor building).]

That parameter is used only to assemble the 2D occupancy grid depending on altitude. It won't affect how loop closure is done. Ideally, the wrong loop closures between different floors would be detected by `RGBD/OptimizeMaxError` and then rejected. You may try decreasing `RGBD/OptimizeMaxError` value. However, the odometry covariance should be carefully tuned to make it work properly. The idea behind `RGBD/OptimizeMaxError` is that after loop closure detection, if the graph is deformed more than X time the covariance error in the links, the loop closure will be rejected. So if covariance is over estimating the real error, the loop closure may still be accepted.

matlabbe, 2023-07-31:
That is a difficult one, the current code doesn't really support an altitude range for loop closures. The hack approach using GPS altitude would not work as altitude [is actually ignored](https://github.com/introlab/rtabmap/blob/e2f037189a6963296dc3752ee51437d0ea5676dc/corelib/src/Memory.cpp#L5892). 

To feed the correct Z in the optimized graph (with `Optimize/PriorsIgnored=false`), maybe `global_pose` topic could be used instead by setting the floor level in z. Setting high covariance to all other parameters (in particular setting >=9999 for rotation parameters, optimizer will just use x-y-z in the graph as priors). If covariance of z is very small, normally during optimization on a loop closure between 2 floors, it won't deform the graph between the floors, creating a large loop closure transform that would be rejected.

closed 2024-05-27 by brayanpa