introlab/rtabmap_ros #1001: Two RGBD cameras using ROS2

takahiko-hasegawa-hdjp, 2023-07-24:
I have two RGB-D cameras: Front & Rear camera. I want to use both cameras for my localization. 
Is there a way to do that? if so do you have any example using ROS launch file?

Thanks in advance

adricort, 2023-07-25:
Hi @takahiko-hasegawa-hdjp you can take a look into my examples with the  D405x2 and D405x3 cameras. I am doing simultaneous mapping using multi session cameras, but I am only localizing so far with the front one. I think there is a way to use two, even three, by configuring the odometry node with rgbd_cameras:2, but I think there are more configs you have to setup.
At least, the mapping, is done synchronized.

takahiko-hasegawa-hdjp, 2023-07-25:
@adricort Thanks so much! I will take a look!

matlabbe, 2023-07-30:
That [example](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/rtabmap_D405x2.launch.py) uses 2 cameras for localization, 1 for odometry. You could make `rgbd_odometry` working with both cameras by also setting `rgbd_cameras` to 2 (with correct remaps). I suggest to build rtabmap with [OpenGV](https://laurentkneip.github.io/opengv/) to get better odometry/localization when using multi-camera setup.

takahiko-hasegawa-hdjp, 2023-12-06:
Well, so I have my own wheel enocders that publishes odom-base_link so I dont need visual odometry. I want to use both two camera for pure localization. 
can I do that?

matlabbe, 2023-12-10:
Yes, you can set `rgbd_cameras` paramater to 2 for `rtabmap` node and provide the input topics (possibly outputs of two `rgbd_sync` nodes)