introlab/rtabmap_ros #1003: GPS no been saved on DB file.

brayanpa, 2023-07-24:
Hi Mathieu,
I'm setting these parameters trying to save GPS in the database to use it during location. But the field in the database is always null. Am I missing something?

'Optimizer/PriorsIgnored': 'False'
'Rtabmap/LoopGPS': 'True'

I'm publishing the GPS data to the "/gps/fix" topic, and I can see that rtabmap is subscribing to that topic.

Thank you.

matlabbe, 2023-07-26:
Is `/gps/fix` correctly stamped? You don't have to set any parameters to get it saved in the database, just make rtabmap subscribing to it, but timestamp should be set correctly. 
https://github.com/introlab/rtabmap_ros/blob/c1d1c01d5d732784ceb6b225ebf6eebdb700ceb0/rtabmap_slam/src/CoreWrapper.cpp#L1911-L1914

brayanpa, 2023-07-28:
Thank you very much Mathieu, after correcting the timestamp the GPS is being saved correctly.

closed 2023-07-28 by brayanpa