introlab/rtabmap_ros #1005: /map topic only showing the last session.

brayanpa, 2023-07-25:
Hello Mathieu, I hope you're doing well. 

I am currently facing an issue while mapping a large environment using multi-session. I found a problem while making a new session. The map from the first session disappears from the topic /map and doesn't show up even after exploring that area further. I have looked at the topic /cloud_map _(**green color in Fig1**)_ and there the merged sessions are shown, while in /map only the last session is shown _(**white color in Fig1**)_. I am seeking assistance on how to make the merged map appear on /map.

I attempted using the /publish_map service, but it did not yield the desired results. The only progress I've made is by running rtabmap-recovery on the database, after this, the complete map appears in /map as shown in **Fig2**, but it's a process that takes a long time and doesn't solve the problem during the mapping process.

[image] **Fig1.** /map topic showing only the last session.

[image] **Fig2.** Expected behavior topic /map showing the entire map.

Thank you so much.

matlabbe, 2023-07-30:
Upon merging, the whole occupancy grid should be regenerated. I have difficulty to reproduce your issue. How did you trigger a new session, by restarting the rtabmap node or calling `trigger_new_map` service? Can you also show the rtabmap parameters you used in your launch file?

brayanpa, 2023-08-14:
I've been playing around with the parameters a lot and found that it only happens when the **Optimizer/Robust** parameter is enabled.

I am starting a new session by launching rtabmap again.

brayanpa, 2023-08-16:
I will close the issue since everything is working correctly with Optimizer/Robust False and RGBD/OptimizeMaxError 3.

Thank you very much for the help.

closed 2023-08-16 by brayanpa