introlab/rtabmap_ros #1007: Remove free cells beyond walls

derektan95, 2023-07-26:
Hi Rtabmap community, 

I would like to repost [this question from the Rtabmap Forum](http://official-rtab-map-forum.206.s1.nabble.com/Remove-free-cells-beyond-walls-td9580.html) to seek some advice from a wider community.

I am using gazebo simulation with rtabmap, along with a 3D lidar for localization and mapping. I noticed some free cells that is often mapped beyond the walls in my simulation, as seen below. Is there a way to modify certain params such that there's no free cells behind the walls (I understand that this isn't realistic in the real world)?  

[image]

This is an example of no free cells beyond obstacles:
[image]

Warm Regards, 
Derek Tan

matlabbe, 2023-07-30:
Hi,

Is the second figure created with Cartographer?

What are the `Grid/****` parameters that you used for rtabmap? To know in particular if that 2D occupancy grid map is created from OctoMap projection or from the 2D ray tracing approach. If you can share the database, that would be nice.

cheers,
Mathieu

derektan95, 2023-07-31:
Hi @matlabbe, 

The second figure is taken from the internet as an example, but I have tried cartographer in the past to get a clean map like this earlier on.

As for the Grid parameters, I have tried to create both 2D occupancy grid map from octomap projection, and from the 2D ray tracing approach. Both have issues where free cells exist beyond walls. Nevertheless, I will upload the rtabmap database files for both `Grid/3D=True` and `Grid/3D=False` very soon.

Warm Regards, 
Derek Tan

derektan95, 2023-08-01:
Hi @matlabbe, 

Here are the database files you have requested for. This folder contains 2 db files where `Grid/3D=True` and `Grid/3D=False`. Thanks for helping look through this! 

Db Files: https://drive.google.com/drive/folders/1oCluv92DReBZqwBDLn7GPfS57FBwLFaY?usp=sharing

Thanks again, 
Derek Tan

matlabbe, 2023-08-05:
2D (rtabmap ray tracing):
[image]

3D (octomap ray tracing):
[image]

In both cases, the ray tracing seems overshooting a little, maybe because of angle resolution. For rtabmap [ray tracing](https://github.com/introlab/rtabmap/blob/a94a4c9802d412106849ba47817e9766f246705b/corelib/src/util3d_mapping.cpp#L826-L905), this is used [here](https://github.com/introlab/rtabmap/blob/a94a4c9802d412106849ba47817e9766f246705b/corelib/src/util3d_mapping.cpp#L709) for each obstacle.

derektan95, 2023-08-06:
Hi Mathieu, 

Thanks for the feedback! Any suggestions on how I can reduce this overshoot? Is increasing the angle resolution a good idea (i.e. lower magnitude)?

matlabbe, 2023-08-06:
The farther it is, the more accurate the [slope computation](https://github.com/introlab/rtabmap/blob/a94a4c9802d412106849ba47817e9766f246705b/corelib/src/util3d_mapping.cpp#L826-L905) should be. Don't know if using double instead of float in that function could help for ray tracing on far obstacles. Other option would be to stop ray tracing free space 1 cell before the actual obstacle cell.

derektan95, 2023-09-01:
Hi Mathieu, 

I tried stopping ray tracing of free space 1 cell before the actual obstacle, at the proposed code location [here](https://github.com/introlab/rtabmap/blob/a94a4c9802d412106849ba47817e9766f246705b/corelib/src/util3d_mapping.cpp#L709). However, I realized this technique only works on my 2D occupancy grid map only if I set `Grid/3D=false`. This does not work if octomap is activated. May I know how I can implement this feature for octomap raytracing?

matlabbe, 2023-09-03:
The ray tracibg for the 3d case is done by the OctoMap library. You may do a similar hack in their code.