introlab/rtabmap_ros #1008: inconsistent results when running rtabmap in ROS playback mode

alexk1976, 2023-07-30:
Hi,
We run rtabmap in RGBD mode in ROS environment.
Even when we playback same clip on same strong PC - we get different results each execution. 
Difference might be 10cm sometimes and even more preventing us to understand real accuracy/improve results.
No such issue we run TUM sequences.
any ideas to get consistent results?
thanks

matlabbe, 2023-07-30:
If you reprocess the database (`rtabmap-reprocess`), there should not have any difference between the computers, unless rtabmap is not built with same dependencies.

With a rosbag, there could be difference on which frame is used to compute odometry or mapping.

On a real robot, you get same time sync issue than rosbag, and the environment/trajectory may be slightly different between the runs.

alexk1976, 2023-09-07:
@matlabbe 
We rebased code with latest and now we get inconsistent results even when run rtabmap-rgbd_dataset on a TUM sequence a few times.  Even number of poses are different each time.
Any ideas what change in an half-year caused to it?

matlabbe, 2023-09-08:
Which TUM sequence are you using? and what command did you use?

alexk1976, 2023-09-09:
For example:
rtabmap-rgbd_dataset --Reg/Force3DoF false --Mem/UseOdomFeatures true --Rtabmap/DetectionRate 2  --Odom/AlignWithGround false /root/data/TUM/rgbd_dataset_freiburg2_pioneer_slam3

matlabbe, 2023-09-09:
The variability may be caused by RANSAC on image registration. Here three runs:
[image]
[image]
[image]

Where there are missing poses on last one, we can log like this:
```
Iteration 1671/2266: camera=6ms, odom(quality=284/949, kfs=580)=37ms, slam=0ms, rmse=0.074733m
[ WARN] (2023-09-09 13:58:48.286) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=32) between -1 and 1672" (guess=xyz=-0.020133,0.207793,-0.056495 rpy=-0.250968,0.058539,0.987020)
[ WARN] (2023-09-09 13:58:48.287) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.020133,0.207793,-0.056495 rpy=-0.250968,0.058539,0.987020), trying again without a guess.
[ WARN] (2023-09-09 13:58:48.319) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-09-09 13:58:48.319) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 7/20 (matches=113) between -1 and 1672" (guess=xyz=-0.020133,0.207793,-0.056495 rpy=-0.250968,0.058539,0.987020)
[ERROR] (2023-09-09 13:58:48.319) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 1672 is ignored!
Iteration 1672/2266: camera=6ms, odom(quality=7/907, kfs=580)=54ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.365) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=68) between -1 and 1673"
Iteration 1673/2266: camera=6ms, odom(quality=0/912, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.410) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=69) between -1 and 1674"
Iteration 1674/2266: camera=6ms, odom(quality=0/914, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.454) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=72) between -1 and 1675"
Iteration 1675/2266: camera=6ms, odom(quality=0/921, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.499) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=61) between -1 and 1676"
Iteration 1676/2266: camera=6ms, odom(quality=0/901, kfs=580)=37ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.544) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=74) between -1 and 1677"
Iteration 1677/2266: camera=6ms, odom(quality=0/911, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.589) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=80) between -1 and 1678"
Iteration 1678/2266: camera=6ms, odom(quality=0/905, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.634) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=79) between -1 and 1679"
Iteration 1679/2266: camera=6ms, odom(quality=0/919, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.679) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=54) between -1 and 1680"
Iteration 1680/2266: camera=6ms, odom(quality=0/912, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.724) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=87) between -1 and 1681"
Iteration 1681/2266: camera=6ms, odom(quality=0/923, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.769) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=75) between -1 and 1682"
Iteration 1682/2266: camera=6ms, odom(quality=0/926, kfs=580)=37ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.814) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=71) between -1 and 1683"
Iteration 1683/2266: camera=6ms, odom(quality=0/907, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.859) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=69) between -1 and 1684"
Iteration 1684/2266: camera=6ms, odom(quality=0/927, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.904) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=71) between -1 and 1685"
Iteration 1685/2266: camera=6ms, odom(quality=0/932, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.949) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=76) between -1 and 1686"
Iteration 1686/2266: camera=6ms, odom(quality=0/934, kfs=580)=37ms, slam=0ms
[ WARN] (2023-09-09 13:58:48.994) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=68) between -1 and 1687"
[ERROR] (2023-09-09 13:58:48.995) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 1687 is ignored!
Iteration 1687/2266: camera=6ms, odom(quality=0/892, kfs=580)=38ms, slam=0ms
[ WARN] (2023-09-09 13:58:49.039) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=62) between -1 and 1688"
Iteration 1688/2266: camera=6ms, odom(quality=0/930, kfs=580)=37ms, slam=0ms
[ WARN] (2023-09-09 13:58:49.084) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=66) between -1 and 1689"
Iteration 1689/2266: camera=6ms, odom(quality=0/918, kfs=580)=38ms, slam=0ms
[... cut ...]
[ WARN] (2023-09-09 13:58:52.771) OdometryF2M.cpp:562::computeTransform() Registration failed: "Not enough inliers 16/20 (matches=102) between -1 and 1769"
[ERROR] (2023-09-09 13:58:52.771) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 1769 is ignored!
Iteration 1769/2266: camera=7ms, odom(quality=16/921, kfs=580)=39ms, slam=0ms

```

I am surprised that it can recover from lost at some point, generarly on datasets, the robot would stay lost for the rest of the dataset unless it sees again same previously mapped areas.