introlab/rtabmap_ros #1010: Unwanted dependencies

Myzhar, 2023-07-31:
Hi @matlabbe 
As I was planning to try out the ZED ROS 2 Wrapper with RTABmap on ROS 2 Humble, I realized that the package comes with many unwanted dependencies:

```
$ sudo apt install ros-humble-rtabmap-ros 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following additional packages will be installed:
  libabsl-dev libfreenect-dev libfreenect0.5 liblua5.2-dev
  ros-humble-cartographer ros-humble-cartographer-ros
  ros-humble-cartographer-ros-msgs ros-humble-dynamixel-sdk
  ros-humble-hls-lfcd-lds-driver ros-humble-imu-filter-madgwick
  ros-humble-libg2o ros-humble-libnabo ros-humble-libpointmatcher
  ros-humble-librealsense2 ros-humble-realsense2-camera
  ros-humble-realsense2-camera-msgs ros-humble-rtabmap
  ros-humble-rtabmap-conversions ros-humble-rtabmap-demos
  ros-humble-rtabmap-examples ros-humble-rtabmap-launch
  ros-humble-rtabmap-msgs ros-humble-rtabmap-odom ros-humble-rtabmap-python
  ros-humble-rtabmap-rviz-plugins ros-humble-rtabmap-slam
  ros-humble-rtabmap-sync ros-humble-rtabmap-util ros-humble-rtabmap-viz
  ros-humble-turtlebot3 ros-humble-turtlebot3-bringup
  ros-humble-turtlebot3-cartographer ros-humble-turtlebot3-description
  ros-humble-turtlebot3-example ros-humble-turtlebot3-msgs
  ros-humble-turtlebot3-navigation2 ros-humble-turtlebot3-node
  ros-humble-turtlebot3-teleop ros-humble-velodyne ros-humble-velodyne-driver
  ros-humble-velodyne-laserscan ros-humble-velodyne-msgs
  ros-humble-velodyne-pointcloud
The following NEW packages will be installed:
  libabsl-dev libfreenect-dev libfreenect0.5 liblua5.2-dev
  ros-humble-cartographer ros-humble-cartographer-ros
  ros-humble-cartographer-ros-msgs ros-humble-dynamixel-sdk
  ros-humble-hls-lfcd-lds-driver ros-humble-imu-filter-madgwick
  ros-humble-libg2o ros-humble-libnabo ros-humble-libpointmatcher
  ros-humble-librealsense2 ros-humble-realsense2-camera
  ros-humble-realsense2-camera-msgs ros-humble-rtabmap
  ros-humble-rtabmap-conversions ros-humble-rtabmap-demos
  ros-humble-rtabmap-examples ros-humble-rtabmap-launch
  ros-humble-rtabmap-msgs ros-humble-rtabmap-odom ros-humble-rtabmap-python
  ros-humble-rtabmap-ros ros-humble-rtabmap-rviz-plugins
  ros-humble-rtabmap-slam ros-humble-rtabmap-sync ros-humble-rtabmap-util
  ros-humble-rtabmap-viz ros-humble-turtlebot3 ros-humble-turtlebot3-bringup
  ros-humble-turtlebot3-cartographer ros-humble-turtlebot3-description
  ros-humble-turtlebot3-example ros-humble-turtlebot3-msgs
  ros-humble-turtlebot3-navigation2 ros-humble-turtlebot3-node
  ros-humble-turtlebot3-teleop ros-humble-velodyne ros-humble-velodyne-driver
  ros-humble-velodyne-laserscan ros-humble-velodyne-msgs
  ros-humble-velodyne-pointcloud
0 upgraded, 44 newly installed, 0 to remove and 305 not upgraded.
Need to get 31.1 MB of archives.
After this operation, 207 MB of additional disk space will be used.
Do you want to continue? [Y/n]
```
Generally, it's not an issue when dealing with desktop applications. However, it's not advisable to install all the components on embedded systems or Docker containers. If I do not intend to use `freenect`, `realsense`, `cartographer`, `turtlebot3`, `velodyne`, `dynamixel`, etc., I wouldn't want to install them.

matlabbe, 2023-07-31:
rtabmap_ros package is a meta package installing everything. I split the package into sub packages last winter to handle such case. You may want to install more specific rtabmap packages like `ros-humble-rtabmap-slam`, `ros-humble-rtabmap-odom` or `ros-humble-rtabmap-launch`. If you install `ros-humble-rtabmap-examples` or `ros-humble-rtabmap-demos`, it will install a lot of these dependencies. 

So you may try `ros-humble-rtabmap-launch` instead to all nodes without the examples and demos. If there are still too may unwanted dependencies, we may discuss them case by case.

matlabbe, 2023-07-31:
`ros-humble-rtabmap` depends on [freenect](https://github.com/introlab/rtabmap/blob/e2f037189a6963296dc3752ee51437d0ea5676dc/package.xml#L17) and [openni](https://github.com/introlab/rtabmap/blob/e2f037189a6963296dc3752ee51437d0ea5676dc/package.xml#L19). We could probably remove them, or move them as dependencies of `ros-humble-rtabmap-examples` instead.

Myzhar, 2023-07-31:
Thank you @matlabbe :+1:

closed 2023-07-31 by Myzhar