introlab/rtabmap_ros #1013: RTABMAP_FLOAM Integration for ROS2

anath93, 2023-08-02:
Hello, 

I went through the code, and this was my finding. For ROS2 side where the FLOAM was ported over to foxy by Chris, did git clone 

https://github.com/Chris7462/floam/tree/master/src

In RTABMAP I got error for not finding files, as the ROS1 side the original FLOAM had no "_" in naming convention for src/*files. 

[image]

My cmake at FLOAM side 
```
cmake_minimum_required(VERSION 3.5)
project(floam)

set(CMAKE_BUILD_TYPE "Release")
set(CMAKE_CXX_FLAGS_RELEASE "-O3 -Wall -g -pthread")

# Default to C++14
if(NOT CMAKE_CXX_STANDARD)
  set(CMAKE_CXX_STANDARD 14)
endif()

if(CMAKE_COMPILER_IS_GNUCXX OR CMAKE_CXX_COMPILER_ID MATCHES "Clang")
  add_compile_options(-Wall -Wextra -Wpedantic)
endif()

# find dependencies
find_package(ament_cmake REQUIRED)
# uncomment the following section in order to fill in
# further dependencies manually.
# find_package(<dependency> REQUIRED)
find_package(rclcpp REQUIRED)
find_package(sensor_msgs REQUIRED)
find_package(nav_msgs REQUIRED)
find_package(geometry_msgs REQUIRED)
find_package(tf2 REQUIRED)
find_package(tf2_ros REQUIRED)
find_package(tf2_geometry_msgs REQUIRED)
find_package(pcl_conversions REQUIRED)
find_package(PCL REQUIRED)
find_package(Eigen3 REQUIRED)
find_package(Ceres REQUIRED)

include_directories(
  include
  ${PCL_INCLUDE_DIRS}
  ${EIGEN3_INCLUDE_DIR}
  ${CERES_INCLUDE_DIRS}
)

link_directories(
  ${PCL_LIBRARY_DIRS}
  ${CERES_LIBRARY_DIRS}
)

#add_library(floam src/laser_processing_class.cpp src/lidar.cpp src/laser_processing.cpp src/odom_estimation_class.cpp src/odom_estimation.cpp src/lidar_optimization.cpp src/laser_mapping_class.cpp src/laser_mapping_class.cpp)

add_executable(laser_processing_node
  src/laser_processing_node.cpp
  src/laser_processing.cpp
  src/laser_processing_class.cpp
  src/lidar.cpp
)

ament_target_dependencies(laser_processing_node
  rclcpp
  sensor_msgs
  pcl_conversions
  Eigen3
  CERES
)

target_link_libraries(laser_processing_node
  ${PCL_LIBRARIES}
)

add_executable(odom_estimation_node
  src/odom_estimation_node.cpp
  src/odom_estimation.cpp
  src/odom_estimation_class.cpp
  src/lidar_optimization.cpp
  src/lidar.cpp
)

ament_target_dependencies(odom_estimation_node
  rclcpp
  sensor_msgs
  nav_msgs
  geometry_msgs
  tf2
  tf2_ros
  pcl_conversions
  Eigen3
  CERES
)

target_link_libraries(odom_estimation_node
  ${PCL_LIBRARIES}
  ${EIGEN3_LIBRARIES}
  ${CERES_LIBRARIES}
)

add_executable(laser_mapping_node
  src/laser_mapping_node.cpp
  src/laser_mapping.cpp
  src/laser_mapping_class.cpp
  src/lidar.cpp
)

ament_target_dependencies(laser_mapping_node
  rclcpp
  sensor_msgs
  nav_msgs
  pcl_conversions
  tf2
  tf2_ros
  tf2_geometry_msgs
  Eigen3
  CERES
)

target_link_libraries(laser_mapping_node
  ${PCL_LIBRARIES}
)

install(TARGETS laser_processing_node odom_estimation_node laser_mapping_node
  DESTINATION lib/${PROJECT_NAME}
)

install(DIRECTORY launch
  DESTINATION share/${PROJECT_NAME}
)

install(DIRECTORY rviz
  DESTINATION share/${PROJECT_NAME}
)

install(DIRECTORY params
  DESTINATION share/${PROJECT_NAME}
)

if(BUILD_TESTING)
  find_package(ament_lint_auto REQUIRED)
  # the following line skips the linter which checks for copyrights
  # uncomment the line when a copyright and license is not present in all source files
  #set(ament_cmake_copyright_FOUND TRUE)
  # the following line skips cpplint (only works in a git repo)
  # uncomment the line when this package is not in a git repo
  #set(ament_cmake_cpplint_FOUND TRUE)
  ament_lint_auto_find_test_dependencies()
endif()

ament_package()
```

anath93, 2023-08-03:
In your experience, which would be better LeGO_LOAM (I do not have variable terrain, its a flat terrain) or ICP or FLOAM ? I tried to find out more on odometry benchmark testing for RTABMAP but did not find any on the internet except for KITTI dataset benchmarking.

matlabbe, 2023-08-06:
Those approaches appeared relatively at the end of my PhD, didn't have time to really get in newer lidar's odometry approaches comparison (other than original LOAM).

For integration with rtabmap, if it is just the header's names that are changed, you may just have to update them in rtabmap code. Also in my patch on FLOAM for ros1, the 
```
#add_library(floam src/laser_processing_class.cpp src/lidar.cpp src/laser_processing.cpp src/odom_estimation_class.cpp src/odom_estimation.cpp src/lidar_optimization.cpp src/laser_mapping_class.cpp src/laser_mapping_class.cpp)

```
was needed and rtabmap expects such floam library to exist to correctly link the code.

anath93, 2023-08-17:
@thank you for your response, just a side note how will this be ported over to ROS2, 

`<node pkg="nodelet" type="nodelet" name="imu_nodelet_manager" args="manager">
<remap from="imu/data_raw" to="$(arg imu_topic)"/>
<remap from="imu/data" to="$(arg imu_topic)/filtered"/>
</node>

<node pkg="nodelet" type="nodelet" name="imu_filter" args="load imu_filter_madgwick/ImuFilterNodelet imu_nodelet_manager">
<param name="use_mag" value="false"/>
<param name="world_frame" value="enu"/>
<param name="publish_tf" value="false"/>
</node>
<node pkg="nodelet" type="nodelet" name="imu_to_tf" args="load rtabmap_util/imu_to_tf imu_nodelet_manager">
<remap from="imu/data" to="$(arg imu_topic)/filtered"/>
<param name="fixed_frame_id" value="$(arg frame_id)_stabilized"/>
<param name="base_frame_id" value="$(arg frame_id)"/>`

`        Node(
            package='imu_filter_madgwick', executable='imu_filter_madgwick_node', output='screen',
            parameters=[{'use_mag': False, 
                         'world_frame':'enu', 
                         'publish_tf':False}],
            remappings=[
                ('imu/data_raw', '/zed2i/zed_node/imu/data')
                ('imu/data', '/zed2i/zed_node/imu/data/filtered') #published after filtering, orientation
            
            ]),`

I do not see any documentation on imu_to_tf node ? Can you please clarify, I am trying to replicate vlp-16 with D435 lidar deskewing example with ICP odometry but with fusing with wheel encoders and IMU.

matlabbe, 2023-08-20:
It looks like `imu_to_tf` has not been converted yet to ROS2: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_util/src/nodelets/imu_to_tf.cpp

I would try "fuse wheel encoders + IMU with EKF node and back to ICP odometry" first.

Changing `Icp/CCSamplingLimit` will do nothing if you are not using CloudCompare ICP strategy (`Icp/Strategy=2`).

Back to original deskewing example with IMU, you could use your combined IMU+Wheel odometry as fixed frame to do the deskewing instead of using `imu_to_tf`. Your EKF filter is probably giving you a TF frame for odometry. Set `guess_frame_id` to your wheelodom+IMU TF frame so that icp_odometry publishes TF correctly.

On rtabmap node, it is unlikely that all the input topics are exactly sync, so set `approx_sync:=true`.

anath93, 2023-08-20:
@matlabbe  Thank you for the response, I did all the changes as you mentioned. I am just left with this error on icp_odometry which I feel is something to do my LIDAR header stamp is split into secs and nano sec and your wrapper needs one timestamp but can you confirm as I was looking into Msgconversion.cpp file if there is a way around ?

[image]

[image]

[image]

anath93, 2023-08-20:
@matlabbe  is there a format which the ICP wrapper is looking for ? Or Velodyne/Ouster stamp so I can replicate this on the LIDAR sdk ?

matlabbe, 2023-08-20:
Yeah, could could change every references of datatype=7 to 8 in deskewing function, by changing also float to double when getting time field in the point cloud.

The header's timestamp (sec,nanosec) is not the same stamp than the one set in the fields, which are time difference from header's stamp. Along fields `x`, `y` and `z`, there should be a field called `time` or `t`.

anath93, 2023-08-20:
@matlabbe Thank you for the response, I did realize that. The issue is that time stamp in the PCL2 message was not changing when I did echo on the topic and it lead me to next error on ICP node that offsettime < Previous stamp, so currently I have raised the issue to the LIDAR company, hopefully the fix comes out soon.
[image]

I will update this thread next week as soon as I get the driver going again.  Thank you again Mathieu !

anath93, 2023-08-21:
> It looks like `imu_to_tf` has not been converted yet to ROS2: https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_util/src/nodelets/imu_to_tf.cpp
>
> I would try "fuse wheel encoders + IMU with [...]

Hi Mathieu, 

I am stuck on one part, EKF will give relation from odom to base of the robot and hence I cannot change the frame_id coming from that node so leads me to Rtabmap have multiple odom tf's to compute for tf unless I am missing something ?
I have also changed the datatypes in Deskew function so the error went away. 
```
ros__parameters:
        frequency: 50.0
        two_d_mode: true
        publish_tf: true
        use_control: false
        
        map_frame: map             
        odom_frame: odom           
        base_link_frame: base_link 
        world_frame: odom
        sensor_timeout: 0.1

        #x     , y     , z,
        #roll  , pitch , yaw,
        #vx    , vy    , vz,
        #vroll , vpitch, vyaw,
        #ax    , ay    , az
        odom0: /diffbot/odom
        odom0_differential: false
        odom0_config: [false, false, false,
                       false, false, false,
                       true, true, false,
                       false, false, true,
                       false, false, false]

        imu0: /zed2i/zed_node/imu/data
        imu0_remove_gravitational_acceleration: true
        #imu0_queue_size: 2
        imu0_differential: false
        im0_config: [false, false, false,
                    true,  true,  true,
                    false, false, false,
                    true,  true,  true,
                    false,  false,  false]
```
`            Node(
          package='rtabmap_util', executable='lidar_deskewing', output='screen',
          parameters=[{
              'fixed_frame_id':'odom',
              'slerp':False,
              'wait_for_transform':0.01,
            }],
          remappings=[
                ('input_cloud', '/lidar_points3/deskewed')
            ]),      
        
        Node(
            package='rtabmap_odom', executable='icp_odometry', output='screen',
            parameters=[{
              'frame_id':'base_link',
              'odom_frame_id':'icp_odom',
              'guess_frame_id':'odom',
              'publish_tf':True,
              'approx_sync':True,
              'wait_for_transform':0.2,
              'queue_size':1,
              'publish_null_when_lost':True,
              'expected_update_rate':10.0,
              'deskewing':deskewing,
              'use_sim_time':use_sim_time,
              'deskewing_slerp':False,
            }],
            remappings=[
              ('scan_cloud', '/lidar_points3'),
              ('odom','/odom_icp'),
              ('odom_info','/rtabmap/odom_info')
              #('imu','/zed2i/zed_node/imu/data/filtered')
            ],
           
        Node(
            package='rtabmap_sync', executable='rgb_sync', output='screen',
            remappings=[
              ('rgb/image', '/zed2i/zed_node/rgb/image_rect_color'),
              ('rgb/camera_info','/zed2i/zed_node/rgb/camera_info')
            ],),
        Node(
            package='rtabmap_slam', executable='rtabmap', output='screen',
            parameters=[{
              'frame_id':'base_link',
              'map_frame_id':'map',
              'publish_tf':True,
              'subscribe_depth':False,
              'subscribe_scan ':False,
              'subscribe_scan_cloud':True,
              'subscribe_stereo':False,
              'subscribe_rgbd':True,
              'approx_sync':True,
              'wait_for_transform':0.2,
              'odom_frame_id':'odom',
              'rgbd_cameras':1,
              'use_sim_time':use_sim_time,

            }`

[image]

anath93, 2023-08-22:
Hi Mathieu,

As you said above, If I just do ICP odom, I see computation in the logger, without guess_frame_id and publishing ekf node odom. But once the ekf and deskewing comes into factor, things go haywire. 
map -> odom -> odom_combined -> base_link -> sensors, wheel, camera ? Isn't this correct ?

anath93, 2023-08-24:
Just a strong belief that rtabmap_odom is not publishing child to parent transform, I looking into cpp file to investigate further
[image]

anath93, 2023-08-27:
Hi Mathieu,

I tried just to launch lidar deskewing and icp_odomery and I get these 2 errors,  

[image]

[image]

[image]

Just ICP works fine as mentioned above, any inputs please ?

Thanks

anath93, 2023-08-27:
Hi Mathieu, 

Is there any way you can verify this on ROS2 side or Any input on this will be appreciated ?

matlabbe, 2023-08-29:
For that [post](https://github.com/introlab/rtabmap_ros/issues/1013#issuecomment-1688625007):
 * You set `'guess_frame_id':''`, so if icp_odometry gets lost, it won't recover unless you enable auto reset. If `guess_frame_id` is set to `odom_combined`, you will get indeed `odom -> odom_combined` published by icp_odometry.
 * It would be similar to use wheel odom+imu than ORB_slam for icp guess, as long as lidar sees enough complex geometry.
 * `Reg/Strategy=1` will use vision for initial guess for icp for global loop closure detection. For proximity detection, it will be done locally only with lidar. I never use `Reg/Strategy=2`.

For that [post](https://github.com/introlab/rtabmap_ros/issues/1013#issuecomment-1694528087):
 * The timestamp channel seems not in same format that we are expecting in deskewing. The `timestamp` should be the offset in seconds from the stamp in the header. In the error msg, the resulting stamps are negative. I would try to print the resulting offset read from the PointCloud data [here](https://github.com/introlab/rtabmap_ros/blob/28db57b29a7268c1f9cdf07b55519f089b8c705a/rtabmap_conversions/src/MsgConversion.cpp#L2810).

closed 2023-09-13 by anath93