introlab/rtabmap_ros #1016: OPENCV compatibility

fbliman, 2023-08-07:
Hi!
I am quite new in robotics and rtabmap-ros has been my most used SLAM software. Up to now i have used it only in the cPU version, but recently I got a GPU laptop and I am trying to install it with all cUDA capabilities with no luck.

HARDWARE: Lenovo P5 AMD RYZEN 5 with NVIDIA Geforce 3050Ti
OS: UBUNTU 20.04
Ros noetic

I know that ros noetic is built with opencv 4.2 and I was planning to build from source opencv 4.2 with CUDA so that I could use SuperPoint+SuperGlue, but I couldn't make it work so I ended up installing opencv 4.5.2 with cuda 11.4 so now I am running with opencv 4.2/4.5 conflict when installing rtabmap--ros

First I tryed ti install opencv 4.2 with either cuda 10 or 11 but failed in both cases ( I couldnt make my gpu work with cuda 10 and I couldnt make opencv 4.2 work with cuda 11)

I would really appreciate some guidance in which would be the correct setup to make rtabmap_ros + ros noetic working in this hardware.

Thanks!!

Federico Bliman

fbliman, 2023-08-09:
I fix this with this with the update weights [here](https://github.com/introlab/rtabmap/tree/master/archive/2022-IlluminationInvariant#how-reproduce-results-shown-in-the-paper)
 

 and now I can get the nodes up, but I am getting a terminate called after throwing an instance of 

> 'c10::CUDAOutOfMemoryError'
  what():  CUDA out of memory. Tried to allocate 226.00 MiB (GPU 0; 3.81 GiB total capacity; 18.41 MiB already allocated; 219.44 MiB free; 30.00 MiB reserved in total by PyTorch) If reserved memory is >> allocated memory try setting max_split_size_mb to avoid fragmentation.  See documentation for Memory Management and PYTORCH_CUDA_ALLOC_CONF

any clues/?

matlabbe, 2023-08-13:
The last error is "out of GPU memory", you may need more memory. 

For special builds like you did, the long way is to uninstall `ros-$ROS_DISTRO-cv-bridge` completely, rebuild/install opencv with CUDA support, then rebuild `cv_bridge` from source (and any packages from source you need that have been uninstalled when removing cv_bridge).

For CUDA/pytorch compatibility with rtabmap_ros, got an example here: https://github.com/introlab/rtabmap_ros/blob/master/docker/noetic/superpoint/Dockerfile