introlab/rtabmap_ros #1019: indeterminate system when mapping multi session using GTSAM.

brayanpa, 2023-08-10:
Hello, I am running multiple sessions in a very large environment. The problem is that when navigating the area there are certain points where the robot jumps out of the map and I see this error in the logs:

_rtabmap-1] Indeterminant linear system detected while working near variable
[rtabmap-1] 3140 (Symbol: 3140).
[rtabmap-1]
[rtabmap-1] Thrown when a linear system is ill-posed. The most common cause for this
[rtabmap-1] error is having underconstrained variables. Mathematically, the system is
[rtabmap-1] underdetermined. See the GTSAM Doxygen documentation at
[rtabmap-1] http://borg.cc.gatech.edu/ on gtsam::IndeterminantLinearSystemException for
[rtabmap-1] more information.
[rtabmap-1] Graph has 22 edges and 17 vertices_

I would try using TORO but the tag detection is being very helpful.

matlabbe, 2023-08-10:
Which rtabmap version are you using? I thoughts I've put enough guards in the code to reject gtsam opt when it diverges.

Note that g2o can also be used with tag

matlabbe, 2023-08-10:
I'll suggest to use rtabmap's master branch and rtabmap_ros's ros2 branch to get latest fixes. The guards I was talking about were added on July 3 https://github.com/introlab/rtabmap/commit/64f79813cd2c3a7e3f916a7ef2637e93f1f13a5f

GTSAM opt could still diverge, but at the least the robot won't jump. Well, for g2o, it still work most of time for multi-session. I think the issue was coming to bad odometry covariance, though as you see, gtsam can also diverge.

brayanpa, 2023-08-14:
You were right, in those branches the problem no longer occurs and we can see in the logs that those locations are being rejected.

Thank you so much.

closed 2023-08-14 by brayanpa