introlab/rtabmap_ros #1021: Odometry lost!

fbliman, 2023-08-15:
Hi!
First of all Thanks @matlabbe for helping m set up docker and always being active here.

I am using rtabmap_ros:superpoint docker and everything is working.

I am working in an underground tunnel proyect with and rgbd camera and imu, and working over the standard rtabmap.launch file.

I would appreciate some help with the setup.

My main problem now is that when Visual odometry gets lost the position gets fixed and the map doesn't continue building.
Then I tryed with Odom/ResetCountdown and it generate a lot of small maps but then I cannot make a global optimization to "join"them, even if not perfect using the imu + visual data, is this posible? what settings should I use? and how should I proceed after the bag finishes to post process the data (it is in the docker, not sure how to save it after rtabmap closes)

Thanks!!
[image]

fbliman, 2023-08-16:
docker run --name rtabmap_ros -it --rm --privileged \
  -e DISPLAY=$DISPLAY \
  -e QT_X11_NO_MITSHM=1 \
  -e NVIDIA_VISIBLE_DEVICES=all \
  -e NVIDIA_DRIVER_CAPABILITIES=all \
  -v $XAUTH:$XAUTH \
  -v /tmp/.X11-unix:/tmp/.X11-unix \
  --runtime=nvidia \
  -e ROS_HOME=/tmp/.ros \
  --network host \
  -v ~/.ros:/tmp/.ros \
  -v ~/Facultad/BagFiles:/workspace/Bagfiles \
  rtabmap_ros:superpoint roslaunch rtabmap_launch rtabmap.launch \
  rtabmap_viz:=true \
  database_path:=/tmp/.ros/rtabmap.db \
  rtabmap_args:="--delete_db_on_start --SuperPoint/ModelPath /workspace/superpoint_v1.pt \
  --PyMatcher/Path /workspace/SuperGluePretrainedNetwork/rtabmap_superglue.py \
  --Kp/DetectorStrategy 10 --Vis/FeatureType 10 --Vis/CorNNType 1 --Reg/RepeatOnce true \
  --Odom/ResetCountdown  2\
  --Rtabmap/DetectionRate 2" \
  odom_args:="--Vis/CorNNType 1 --Reg/RepeatOnce true --Vis/CorGuessWinSize 40 --Vis/FeatureType 10" \
  depth_topic:=/camera/aligned_depth_to_color/image_raw \
  rgb_topic:=/camera/color/image_raw \
  camera_info_topic:=/camera/color/camera_info \
  approx_sync:=false \
  wait_imu_to_init:=true \
  imu_topic:=/rtabmap/imu

fbliman, 2023-08-18:
I get a some warnings like this 

<details><summary>Details</summary>
<p>
[ WARN] (2023-08-18 13:01:54.152) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 15/20 (matches=190) between -1 and 159" (guess=xyz=-0.004506,0.040616,0.017191 rpy=-0.000399,0.017786,0.042718)
[ WARN] (2023-08-18 13:01:54.152) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=-0.004506,0.040616,0.017191 rpy=-0.000399,0.017786,0.042718), trying again without a guess.
[ WARN] (2023-08-18 13:01:54.184) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-08-18 13:01:54.184) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=77) between -1 and 159" (guess=xyz=-0.004506,0.040616,0.017191 rpy=-0.000399,0.017786,0.042718)
[ INFO] [1692363714.185472189]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.070344s
[ WARN] (2023-08-18 13:01:54.224) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 12/20 (matches=191) between -1 and 160" (guess=xyz=0.000000,0.000000,0.000000 rpy=-0.001035,0.040429,0.069009)
[ WARN] (2023-08-18 13:01:54.224) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=0.000000,0.000000,0.000000 rpy=-0.001035,0.040429,0.069009), trying again without a guess.
[ WARN] (2023-08-18 13:01:54.256) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-08-18 13:01:54.256) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=66) between -1 and 160" (guess=xyz=0.000000,0.000000,0.000000 rpy=-0.001035,0.040429,0.069009)
[ INFO] [1692363714.257349523]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.070094s
[ INFO] [1692363714.297805997]: Odom: quality=26, std dev=0.011653m|0.020556rad, update time=0.039068s
[ INFO] [1692363714.354486967]: Odom: quality=51, std dev=0.009213m|0.022810rad, update time=0.042063s
[ INFO] [1692363714.426302351]: Odom: quality=27, std dev=0.011822m|0.020040rad, update time=0.045240s
[ INFO] [1692363714.490558560]: Odom: quality=28, std dev=0.006144m|0.016542rad, update time=0.040736s
[ WARN] (2023-08-18 13:01:54.549) OdometryF2M.cpp:566::computeTransform() Registration failed: "Too low inliers after bundle adjustment: 18<20" (guess=xyz=0.011295,0.008805,0.000308 rpy=-0.001262,-0.021044,0.003121)
[ WARN] (2023-08-18 13:01:54.549) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=0.011295,0.008805,0.000308 rpy=-0.001262,-0.021044,0.003121), trying again without a guess.
[ WARN] (2023-08-18 13:01:54.581) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-08-18 13:01:54.581) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=88) between -1 and 165" (guess=xyz=0.011295,0.008805,0.000308 rpy=-0.001262,-0.021044,0.003121)
[ INFO] [1692363714.583073574]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.070497s
[ INFO] [1692363714.634498865]: Odom: quality=22, std dev=0.013143m|0.023616rad, update time=0.045369s
[ INFO] [1692363714.693702772]: Odom: quality=30, std dev=0.007247m|0.021909rad, update time=0.042858s

</p>
</details> 

from which I can recover

and then I lose it

<details><summary>Details</summary>
<p>

 WARN] (2023-08-18 13:02:11.885) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 17/20 (matches=209) between -1 and 425" (guess=xyz=0.020212,-0.001554,0.014354 rpy=0.032769,-0.019100,0.025111)
[ WARN] (2023-08-18 13:02:11.885) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=0.020212,-0.001554,0.014354 rpy=0.032769,-0.019100,0.025111), trying again without a guess.
[ INFO] [1692363731.889274199]: rtabmap (29): Rate=0.50s, Limit=0.000s, Conversion=0.0018s, RTAB-Map=0.1558s, Maps update=0.0000s pub=0.0323s (local map=29, WM=29)
[ WARN] (2023-08-18 13:02:11.918) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-08-18 13:02:11.918) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=80) between -1 and 425" (guess=xyz=0.020212,-0.001554,0.014354 rpy=0.032769,-0.019100,0.025111)
[ INFO] [1692363731.920583245]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.070635s
[ WARN] (2023-08-18 13:02:11.961) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 11/20 (matches=186) between -1 and 426" (guess=xyz=0.000000,0.000000,0.000000 rpy=0.049868,-0.021901,0.050777)
[ WARN] (2023-08-18 13:02:11.961) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=0.000000,0.000000,0.000000 rpy=0.049868,-0.021901,0.050777), trying again without a guess.
[ WARN] (2023-08-18 13:02:11.994) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-08-18 13:02:11.994) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=69) between -1 and 426" (guess=xyz=0.000000,0.000000,0.000000 rpy=0.049868,-0.021901,0.050777)
[ INFO] [1692363731.995799825]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.073624s
[... cut ...]
[ WARN] (2023-08-18 13:02:12.975) OdometryF2M.cpp:314::computeTransform() Failed to find a transformation with the provided guess (xyz=0.000000,0.000000,0.000000 rpy=0.075144,0.026478,0.191546), trying again without a guess.
[ WARN] (2023-08-18 13:02:13.008) OdometryF2M.cpp:556::computeTransform() Trial with no guess still fail.
[ WARN] (2023-08-18 13:02:13.008) OdometryF2M.cpp:566::computeTransform() Registration failed: "Not enough inliers 0/20 (matches=63) between -1 and 440" (guess=xyz=0.000000,0.000000,0.000000 rpy=0.075144,0.026478,0.191546)
[ INFO] [1692363733.009056349]: Odom: quality=0, std dev=0.000000m|0.000000rad, update time=0.067728s

</p>
</details>

matlabbe, 2023-08-18:
With `-v ~/.ros:/tmp/.ros `, the default rtabmap.db would be created on your host computer under `~/.ros/rtabmap.db`. 

Did you try default `Vis/FeatureType`? and if you wanted to use superpoint, you should set it to 11, not 10.

What is the resolution of the images?

fbliman, 2023-08-21:
Thanks, i am using -v ~/.ros:/tmp/.ros  and actually what it is working is that if I run in SLAM mode 5...6 times each round it advances a little bit more based on previous maps. (this with default feature types)

height: 720
width: 1280

what I do not understand is why when it looses VO why doesnt it continue with IMU data till it can recover VO

matlabbe, 2023-08-27:
The VO approach implemented in rtabmap is not a tightly VIO approach. It won't continue estimating only with IMU. You may try reducing by 2 your image resolution for odometry, this can be done by convience with parameter `--Odom/ImageDecimation 2`

fbliman, 2023-08-28:
So It only uses the IMU for a first guess for the VO, but if it can not get a good estimate using VO then it will discard that link?

matlabbe, 2023-08-29:
Yes, the imu change in orientation between last frame and current timestamp is used as guess rotation for visual feature matching. If features cannot be tracked, or there is a lack of inliers, VO will fail and discard that frame. The odometry will then be [lost](https://github.com/introlab/rtabmap/wiki/Kinect-mapping#lostodometry).