introlab/rtabmap_ros #1023: Adding Planner Algorithms

furkancetiny, 2023-08-18:
Hello

Can we integrate planner algorithms such as DWA planner and TEB planner into the 3D map created with this package? Are there any tutorials on this?

matlabbe, 2023-08-19:
You can use them with the generated 2D occupancy grid though. See http://wiki.ros.org/rtabmap_ros/Tutorials/MappingAndNavigationOnTurtlebot for an example using DWA. For TEB, you would need to modify move_base parameters.

furkancetiny, 2023-08-22:
Thank you very much. I made changes to the move_base parameters and was able to output. I'm also doing this on a real robot using the d435i realsense camera. I have an existing map. However, I could not make pose estimates with real Lidar data on the map. I updated the 2D pose estimate topic name to "/rtabmap/initialpose" in the RVIZ environment. But I didn't get any output. What can I do about this issue?

matlabbe, 2023-08-27:
> I could not make pose estimates with real Lidar data on the map

Are you trying to localize on a 2D occupancy grid with a lidar? (e.g., using AMCL?)

furkancetiny, 2023-08-28:
Yes. I am using neo_localization. I have an existing map I'm placing on it.

matlabbe, 2023-08-29:
If you are using `neo_localization` for localization, you don't need `rtabmap` node at the same time.

closed 2023-09-05 by furkancetiny