introlab/rtabmap_ros #1024: Tracking fail

robotdevel, 2023-08-22:
I have a problem when aligndepth image does not have any depth data,

[ INFO] [1692743054.843064981, 1692589725.078166641]: rtabmap (1): Rate=0.50s, Limit=0.000s, Conversion=0.0079s, RTAB-Map=0.0476s, Maps update=0.0045s pub=0.0029s (local map=1, WM=1)
[ INFO] [1692743054.911950702, 1692589725.148594304]: Publishing maps... poses=2
[ INFO] [1692743054.912307689, 1692589725.148594304]: rtabmap (2): Rate=0.50s, Limit=0.000s, Conversion=0.0023s, RTAB-Map=0.0384s, Maps update=0.0014s pub=0.0004s (local map=2, WM=2)
[ WARN] (2023-08-23 07:24:15.482) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
OpenCV Error: Assertion failed (src[i].dims <= 2 && src[i].rows == src[0].rows && src[i].type() == src[0].type()) in hconcat, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 2865
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:2865: error: (-215) src[i].dims <= 2 && src[i].rows == src[0].rows && src[i].type() == src[0].type() in function hconcat

I know the reason but..

do you have any solution for this such as that rtabmap has pause mode when depth data is not comming?

i would be happy if rtamap had a pause mode such as rtabmap stop to estimate camera pose and than if depth data is comming, relocalization and tracking again..

i think rtabmap have a re-localization mode so i thought that rtabmap has a solution.

matlabbe, 2023-08-27:
Somewhere in the code doesn't handle when depth contains only null values. In your log, I don't see which node is crashing, is it rtabmap? or rviz's mapcloud plugin?

I have difficulty to reproduce the issue on my side. I can generate the warning by hiding the depth sensor, but not the opencv error:
```
[ WARN] (2023-08-27 14:07:41.022) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
```
Which rtabmap version are you using?

kdhRGT, 2023-08-29:
```
 * /rosdistro: melodic
 * /rosversion: 1.14.13
 * /rtabmap/rtabmap/GTSAM/Optimizer: 0
 * /rtabmap/rtabmap/Grid/3D: true
 * /rtabmap/rtabmap/Grid/CellSize: 0.025
 * /rtabmap/rtabmap/Grid/RangeMax: 22
 * /rtabmap/rtabmap/Grid/Sensor: 2
 * /rtabmap/rtabmap/Kp/DetectorStrategy: 2
 * /rtabmap/rtabmap/Kp/MaxFeatures: 400
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/NotLinkedNodesKept: false
 * /rtabmap/rtabmap/Optimizer/Epsilon: 0.000001
 * /rtabmap/rtabmap/Optimizer/Iterations: 300
 * /rtabmap/rtabmap/Optimizer/Strategy: 1
 * /rtabmap/rtabmap/Reg/Force3DoF: true
 * /rtabmap/rtabmap/Reg/Strategy: 1
 * /rtabmap/rtabmap/Rtabmap/DetectionRate: 2
 * /rtabmap/rtabmap/SURF/HessianThreshold: 100
 * /rtabmap/rtabmap/Vis/MinInliers: 30
 * /rtabmap/rtabmap/database_path: rtabmap.db
 * /rtabmap/rtabmap/depth_distTheshold: 2.0
 * /rtabmap/rtabmap/depth_voxelTheshold: 0.025
 * /rtabmap/rtabmap/frame_id: ekf_base_...
 * /rtabmap/rtabmap/laser/dist/xmin: 0.0
 * /rtabmap/rtabmap/laser/radfilter/MinNeighbors: 4
 * /rtabmap/rtabmap/laser/radfilter/Radius: 0.2
 * /rtabmap/rtabmap/laser_distTheshold: 22.0
 * /rtabmap/rtabmap/mapgentopic: mapGenflag
 * /rtabmap/rtabmap/odom_frame_id: ekf_odom
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_scan: True

NODES
  /
    ekf_base_link_to_scan (tf/static_transform_publisher)
    ekf_base_to_realsense (tf/static_transform_publisher)
    ekf_camera_to_colorframe (tf2_ros/static_transform_publisher)
    ekf_colorframe_to_depth_opticalframe (tf2_ros/static_transform_publisher)
    ekf_colorframe_to_opticalframe (tf2_ros/static_transform_publisher)
    ekf_foot_to_base (tf/static_transform_publisher)
    tf_pub (robot_pose_ekf/tf_pub)
  /rtabmap/
    rtabmap (rtabmap_ros/rtabmap)

auto-starting new master
process[master]: started with pid [2808]
ROS_MASTER_URI=http://localhost:11311

setting /run_id to 068751f0-4663-11ee-9b28-a8a1596e9eda
process[rosout-1]: started with pid [2830]
started core service [/rosout]
process[ekf_foot_to_base-2]: started with pid [2833]
process[ekf_base_to_realsense-3]: started with pid [2834]
process[ekf_base_link_to_scan-4]: started with pid [2835]
process[ekf_camera_to_colorframe-5]: started with pid [2836]
process[ekf_colorframe_to_opticalframe-6]: started with pid [2837]
process[ekf_colorframe_to_depth_opticalframe-7]: started with pid [2851]
process[tf_pub-8]: started with pid [2864]
process[rtabmap/rtabmap-9]: started with pid [2878]
[ INFO] [1693310162.557310490]: ----> TF publisher Started.
[ INFO] [1693310163.127490980]: ----> Core Node Started.
[ INFO] [1693310163.148614756]: Initializing nodelet with 12 worker threads.
[ INFO] [1693310163.416901610]: ----> MapsManager Node Started.
[ INFO] [1693310163.420385349]: Mapsmanager depth_distTheshold = 2.000000 
[ INFO] [1693310163.420414886]: Mapsmanager depth_voxelTheshold = 0.025000 
[ INFO] [1693310163.420432492]: Mapsmanager laser_distTheshold = 22.000000 
[ INFO] [1693310163.420450143]: /rtabmap/rtabmap(maps): map_filter_radius          = 0.000000
[ INFO] [1693310163.420465241]: /rtabmap/rtabmap(maps): map_filter_angle           = 30.000000
[ INFO] [1693310163.420486029]: /rtabmap/rtabmap(maps): map_cleanup                = true
[... 45 lines cut ...]
[ INFO] [1693310164.023101371]: /rtabmap/rtabmap: subscribe_rgb = true
[ INFO] [1693310164.023128709]: /rtabmap/rtabmap: subscribe_stereo = false
[ INFO] [1693310164.023159098]: /rtabmap/rtabmap: subscribe_rgbd = false (rgbd_cameras=1)
[ INFO] [1693310164.023183449]: /rtabmap/rtabmap: subscribe_odom_info = false
[ INFO] [1693310164.023233109]: /rtabmap/rtabmap: subscribe_user_data = false
[ INFO] [1693310164.023271627]: /rtabmap/rtabmap: subscribe_scan = true
[ INFO] [1693310164.023304575]: /rtabmap/rtabmap: subscribe_scan_cloud = false
[ INFO] [1693310164.023335523]: /rtabmap/rtabmap: subscribe_scan_descriptor = false
[ INFO] [1693310164.023361601]: /rtabmap/rtabmap: queue_size    = 10
[ INFO] [1693310164.023393621]: /rtabmap/rtabmap: approx_sync   = true
[ INFO] [1693310164.023453408]: Setup depth callback
[ INFO] [1693310164.062568690]: 
/rtabmap/rtabmap subscribed to (approx sync):
   /camera1/color/image_raw \
   /camera1/aligned_depth_to_color/image_raw \
   /camera1/color/camera_info \
   /scan
[ INFO] [1693310164.191580034]: rtabmap 0.20.19 started...
```

i think my rtabmap version is "0.20.19"

if that is version problem... how can i update it?

I am using opencv 3.2 version also.

robotdevel, 2023-08-30:
I think I found where have some problem in my case.

```
				if(!groundCells.empty() && !scanGroundCells.empty())
					cv::hconcat(groundCells, scanGroundCells, groundCells);
				else if(!scanGroundCells.empty())
					groundCells = scanGroundCells;

				if(!obstacleCells.empty() && !scanObstacleCells.empty())
					cv::hconcat(obstacleCells, scanObstacleCells, obstacleCells);
				else if(!scanObstacleCells.empty())
					obstacleCells = scanObstacleCells;

				if(!emptyCells.empty() && !scanEmptyCells.empty())
					cv::hconcat(emptyCells, scanEmptyCells, emptyCells);
				else if(!scanEmptyCells.empty())
					emptyCells = scanEmptyCells;
```

here code have a crush with opencv 3.2 from this error.

```
[ WARN] (2023-08-30 00:38:44.655) util3d.cpp:605::cloudFromDepthRGB() Cloud with only NaN values created!
OpenCV Error: Assertion failed (src[i].dims <= 2 && src[i].rows == src[0].rows && src[i].type() == src[0].type()) in hconcat, file /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp, line 2865
terminate called after throwing an instance of 'cv::Exception'
  what():  /build/opencv-L2vuMj/opencv-3.2.0+dfsg/modules/core/src/matrix.cpp:2865: error: (-215) src[i].dims <= 2 && src[i].rows == src[0].rows && src[i].type() == src[0].type() in function hconcat
```

can you suggest me to solve this issue?

matlabbe, 2023-08-31:
I missed the detail that you set `"Grid/Sensor"="2"`. Can you add `--udebug` in rtabmap's `args` to show those logs?
https://github.com/introlab/rtabmap/blob/6a5b844bacfb3e4dc56a9d6783887d959ddbca14/corelib/src/OccupancyGrid.cpp#L477-L479

robotdevel, 2023-08-31:
@matlabbe 

I solved the problem!

```
if(!obstacleCells.empty() && !scanObstacleCells.empty() && obstacleCells.cols != scanObstacleCells.cols)
   cv::hconcat(obstacleCells, scanObstacleCells, obstacleCells);
else if(!scanObstacleCells.empty())
   obstacleCells = scanObstacleCells;
```

This is how I solve the problem.

But i dont know why if obstacleCells.cols and scanObstacleCells.cols are same than program is dead

can you give me adivse? thank you so much

matlabbe, 2023-09-04:
I was able to reproduce the issue with `Grid/Sensor=2` and sending a depth image with all zeros. It should be fixed in commit above.

closed 2023-09-04 by matlabbe