introlab/rtabmap_ros #1027: [rtabmap/stereo_odometry-3] process has died

mahammadirfan, 2023-08-29:
Hi, 
I am using ubuntu noetic, whenever i start the rtabmap the odometry node crashes. Help me please
[image]

mahammadirfan, 2023-08-29:
my 
[image]
gdb result

mahammadirfan, 2023-08-30:
hi 
my latest installation gives this warning

[ 97%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/icp_odometry
[ 98%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/rgbd_odometry
[100%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/stereo_odometry
[100%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/rgbdicp_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_icp_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_rgbd_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_stereo_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_rgbdicp_odometry

mahammadirfan, 2023-08-30:
Hi,
I have installed fresh rtabmap & rtabmap_ros latest for noetic....installed successfully but gives the warning at the end.....i am posting the installation & when i am launching rtabmap_ros ...it crashes like same with odometry error........

[ 95%] Built target rtabmap_odom
Scanning dependencies of target rtabmap_odom_plugins
[ 95%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_odom_plugins.dir/src/nodelets/rgbd_odometry.cpp.o
[ 96%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_odom_plugins.dir/src/nodelets/stereo_odometry.cpp.o
[ 96%] Linking CXX shared library /home/drone/rtabmap_ws/devel/lib/librtabmap_legacy_plugins.so
[ 96%] Built target rtabmap_legacy_plugins
[ 96%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_odom_plugins.dir/src/nodelets/rgbdicp_odometry.cpp.o
[ 96%] Linking CXX shared library /home/drone/rtabmap_ws/devel/lib/librtabmap_slam_plugins.so
[ 96%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_odom_plugins.dir/src/nodelets/icp_odometry.cpp.o
[ 96%] Built target rtabmap_slam_plugins
[ 97%] Linking CXX shared library /home/drone/rtabmap_ws/devel/lib/librtabmap_odom_plugins.so
[ 97%] Built target rtabmap_odom_plugins
Scanning dependencies of target rtabmap_rgbd_odometry
Scanning dependencies of target rtabmap_rgbdicp_odometry
Scanning dependencies of target rtabmap_icp_odometry
Scanning dependencies of target rtabmap_stereo_odometry
[ 97%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_rgbd_odometry.dir/src/RGBDOdometryNode.cpp.o
[ 97%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_rgbdicp_odometry.dir/src/RGBDICPOdometryNode.cpp.o
[ 97%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_icp_odometry.dir/src/ICPOdometryNode.cpp.o
[ 97%] Building CXX object rtabmap_ros/rtabmap_odom/CMakeFiles/rtabmap_stereo_odometry.dir/src/StereoOdometryNode.cpp.o
[ 97%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/icp_odometry
[ 98%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/rgbd_odometry
[100%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/stereo_odometry
[100%] Linking CXX executable /home/drone/rtabmap_ws/devel/lib/rtabmap_odom/rgbdicp_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_icp_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_rgbd_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
/usr/bin/ld: warning: libopencv_videoio.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_videoio.so.4.2
/usr/bin/ld: warning: libopencv_calib3d.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_calib3d.so.4.2
/usr/bin/ld: warning: libopencv_dnn.so.4.5, needed by /usr/lib/aarch64-linux-gnu/libopencv_video.so.4.5.4, may conflict with libopencv_dnn.so.4.2
[100%] Built target rtabmap_stereo_odometry
/usr/bin/ld: warning: libopencv_stitching.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_stitching.so.4.2
/usr/bin/ld: warning: libopencv_photo.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_photo.so.4.2
/usr/bin/ld: warning: libopencv_video.so.4.5, needed by /usr/local/lib/librtabmap_core.so.0.21.1, may conflict with libopencv_video.so.4.2
[... cut ...] _transforms odom:=odom imu:=/imu/data __name:=rtabmap __log:=/home/drone/.ros/log/f9d49ee8-4731-11ee-8dd1-48b02d3a8516/rtabmap-rtabmap-4.log].
log file: /home/drone/.ros/log/f9d49ee8-4731-11ee-8dd1-48b02d3a8516/rtabmap-rtabmap-4*.log
^C[points_xyzrgb-6] killing on exit
[rviz-5] killing on exit
[camera_base_link1-2] killing on exit
^C^C^C^C^C^C^C^C^C^C^C^C^C^C[rosout-1] killing on exit
^C^C[master] killing on exit
^Cshutting down processing monitor...
... shutting down processing monitor complete

matlabbe, 2023-08-31:
It looks like you have two opencv versions installed. On runtime, the wrong library is taken in compared to one it has been built with. From the install [instructions](https://github.com/introlab/rtabmap_ros#build-from-source), if you don't want to use opencv 4.2:

> On Noetic, build from source with xfeatures2d module (and nonfree module if needed) the same OpenCV version already installed on the system. You will then avoid breaking cv_bridge with rtabmap_ros. If you want to install a more recent OpenCV version, I recommend to uninstall libopencv* libraries (with all ros packages depending on it) and rebuild all those ros packages in your catkin workspace (to make sure cv_bridge is linked on the OpenCV version you just compiled).