introlab/rtabmap_ros #1029: 3D Obstacle Height

furkancetiny, 2023-09-05:
Hello,

I'm trying to make a 3D map of the environment using the d435i realsense_camera and rplidar I have. And I can extract the map just fine. I am trying to navigate on this map. A question that comes to my mind is what should I do against high obstacles such as a table? My robot can only see the feet of the table, causing it to crash into the table.

The relevant code in the .launch file is as follows:

<group ns="rtabmap">
    <node pkg="rtabmap_sync" type="rgbd_sync" name="rgbd_sync" output="screen">
      <remap from="rgb/image"         to="/camera/color/image_raw"/>
      <remap from="depth/image"       to="/camera/depth/image_rect_raw"/>
      <remap from="rgb/camera_info"   to="/camera/color/camera_info"/>
    </node>

    <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="$(arg rtabmap_args)">
      <param name="database_path"       type="string" value="$(arg database_path)"/>
      <param name="frame_id"            type="string" value="base_footprint"/>
      <param name="subscribe_rgb"   type="bool" value="false"/>
      <param name="subscribe_depth" type="bool" value="false"/>
      <param name="subscribe_rgbd" type="bool" value="true"/>
      <param name="subscribe_scan"      type="bool"   value="true"/>
      <param name="approx_sync"         type="bool"   value="true"/>

      
      <param name="use_action_for_goal" type="bool" value="true"/>
      <remap from="move_base"            to="/move_base"/>

      
      <remap from="scan"            to="/scan"/>
      <remap from="odom"            to="/base_controller/odom"/>
      <remap from="rgbd_image"       to="rgbd_image"/>

      
      <remap from="grid_map" to="/map"/>

      
      <param name="Reg/Strategy"                 type="string" value="1"/>
      <param name="Reg/Force3DoF"                type="string" value="true"/>
      <param name="GridGlobal/MinSize"           type="string" value="20"/>
      <param name="RGBD/NeighborLinkRefining"    type="string" value="true"/>

      
      <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
      <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
    </node>

    
    <node if="$(arg rtabmap_viz)" pkg="rtabmap_viz" type="rtabmap_viz" name="rtabmap_viz" args="-d $(find rtabmap_demos)/launch/config/rgbd_gui.ini" output="screen">
      <param name="subscribe_scan"   type="bool"   value="true"/>
      <param name="subscribe_odom"   type="bool"   value="true"/>
      <param name="frame_id"         type="string" value="base_footprint"/>
      <param name="approx_sync"      type="bool"   value="true"/>

      <remap from="odom"            to="/base_controller/odom"/>
      <remap from="scan"            to="/scan"/>
    </node>
  </group>

matlabbe, 2023-09-09:
Can the camera sees the top of the table? You may try creating the occupancy grid from the camera: `Grid/Sensor=1`

sun-rabbit, 2023-10-13:
> 你好，
> 
> 我正在尝试使用我拥有的 d435i realsense_camera 和 rplidar 制作环境的 3D 地图。我可以很好地提取地图。我正在尝试在这张地图上导航。我想到的一个问题是，面对桌子等高障碍物，我该怎么办？我的机器人只能看到桌子的脚，导致它撞到桌子上。
> 
> .launch文件中相关代码如下：
> 
> ```
> <node name="rtabmap" pkg="rtabmap_slam" type="rtabmap" output="screen" args="$(arg rtabmap_args)">
>   <param name="database_path"       type="string" value="$(arg database_path)"/>
>   <param name="frame_id"            type="string" value="base_footprint"/>
>   <param name="subscribe_rgb"   type="bool" value="false"/>
>   <param name="subscribe_depth" type="bool" value="false"/>
>   <param name="subscribe_rgbd" type="bool" value="true"/>
>   <param name="subscribe_scan"      type="bool"   value="true"/>
>   <param name="approx_sync"         type="bool"   value="true"/>
> 
>   
>   <param name="use_action_for_goal" type="bool" value="true"/>
>   <remap from="move_base"            to="/move_base"/>
> 
>   
>   <remap from="scan"            to="/scan"/>
>   <remap from="odom"            to="/base_controller/odom"/>
>   <remap from="rgbd_image"       to="rgbd_image"/>
> 
>   
>   <remap from="grid_map" to="/map"/>
> 
>   
>   <param name="Reg/Strategy"                 type="string" value="1"/>
>   <param name="Reg/Force3DoF"                type="string" value="true"/>
>   <param name="GridGlobal/MinSize"           type="string" value="20"/>
>   <param name="RGBD/NeighborLinkRefining"    type="string" value="true"/>
> 
>   
>   <param     if="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="false"/>
>   <param unless="$(arg localization)" name="Mem/IncrementalMemory" type="string" value="true"/>
> </node>
> 
> 
> <node if="$(arg rtabmap_viz)" pkg="rtabmap_viz" type="rtabmap_viz" name="rtabmap_viz" args="-d $(find rtabmap_demos)/launch/config/rgbd_gui.ini" output="screen">
>   <param name="subscribe_scan"   type="bool"   value="true"/>
>   <param name="subscribe_odom"   type="bool"   value="true"/>
>   <param name="frame_id"         type="string" value="base_footprint"/>
>   <param name="approx_sync"      type="bool"   value="true"/>
> 
>   <remap from="odom"            to="/base_controller/odom"/>
>   <remap from="scan"            to="/scan"/>
> </node>
> ```

hello，May I ask what model of your robot is? I also want to perform 3D graphics on Xuri Pi, but I do not have relevant information. Have you written a tutorial for this experiment

smile6023326, 2024-04-16:
Hello, I have used this program recently, but cannot connect to my d435i to make the robot move. Can you share the complete code? Thanks
my results:
[image]