introlab/rtabmap_ros #1032: the loop closure detection has poor performance and sometimes incorrectly adjusts the previously accurate model.

tandongxu0604, 2023-09-10:
Hello, I need your help. I'm trying to model the wall of a flat pool using the rtabmap algorithm. The sensors include a stereo camera and high-precision fiber optic inertial navigation. While controlling the robot's movement, I've been facing some persistent issues.
 The first issue is that when the robot moves directly towards the wall, the modeled wall appears curved. However, if the robot is tilted at a certain angle, the modeled wall appears straight, but the point cloud has many gaps.
 The second issue is that the modeled wall by the robot always has a lot of overlapping and multiple depths.
 The third issue is that when the robot encounters loop closures, the loop closure detection has poor performance and sometimes incorrectly adjusts the previously accurate model.
 I have tested the data multiple times and I feel like it's not a problem with the stereo_odometry node, but rather with the rtabmap node. There are so many parameters in the rtabmap algorithm, and even though I've tried many times, the results are still not satisfactory. Can you give me some advice? What parameters are useful for me to change?
![Uploading 2023-09-11 00-06-56屏
[image]
幕截图.png…]()

matlabbe, 2023-09-10:
> The first issue is that when the robot moves directly towards the wall, the modeled wall appears curved. However, if the robot is tilted at a certain angle, the modeled wall appears straight, but the point cloud has many gaps.

Do you have a database to share?

> The second issue is that the modeled wall by the robot always has a lot of overlapping and multiple depths.

This is expected, as each node has its own cloud. The clouds can be combined when exporting afterwards though.

> The third issue is that when the robot encounters loop closures, the loop closure detection has poor performance and sometimes incorrectly adjusts the previously accurate model.

If you have poor odometry or under/over estimated covariance, or loop closure transform is just not accurate, the accumulated errors may be poorly redistributed along the graph. Are you using external odometry?

> I have tested the data multiple times and I feel like it's not a problem with the stereo_odometry node, but rather with the rtabmap node. There are so many parameters in the rtabmap algorithm, and [...]

If you are using stereo_odometry, most of the issues you observed, except loop closures, maybe related to poor visual odometry. Is your stereo camera correctly calibrated, what kind of environment are you scanning?

tandongxu0604, 2023-09-11:
Hello, I'm glad to Get your help.

Firstly, I tested in a large water tank using an underwater robot with a stereo camera and high-precision fiber optic inertial navigation. I want to model one side of the water tank, which has a very flat wall with less geometric features but rich texture.

Regarding your first question, I have two databases as follows:
1.https://drive.google.com/file/d/1ijkrKNyyj1qNonTAyP2nB4KsH-Hbos0u/view?usp=drive_link
2.https://drive.google.com/file/d/1DPPMLUqF2SekLgLhKchvAoPrTHoucUeE/view?usp=drive_link

    The first one involves the underwater robot moving laterally at an angle to the wall of the water tank.
    The second one involves the underwater robot moving laterally directly facing the wall of the water tank.

In the first dataset, the rtabmap algorithm models a fairly straight wall, but there are many point cloud gaps. Additionally, during loop closure, the algorithm incorrectly adjusts the previously accurate model, leading to a curved wall (imageId = 1785).

In the second dataset, there are rich points cloud in the model, but even without loop closure, the rtabmap algorithm still produces a curved wall.

Regarding your second question, are you mean that  using rtabmap-databaseViewer to merge overlapping point clouds?

For your third question, in both datasets, I used stereo-IMU odometry, and the covariance matrix of the odometry is provided by the rtabmap-stereo_odometry node. If the poor performance is due to loop closure, which parameters should I try adjusting? I've already tried many parameters, but the problem remains unresolved.

Thank you! 😊

matlabbe, 2023-09-12:
In the database, it seems you were using a RGB-D input for rtabmap node. I cannot see the stereo images you used for stereo_odometry. The covariance looks wrong, if odometry is coming from stereo_odometry, I would have expected it to vary over time, it seems fixed in the database.

The depth image looks also wrong, we cannot see any details in geometry:
[image]

Also here the rule looks flat (is it a tape? or should it appear distant to wall?):

[image]
Color is depth change (the rule is invisible):
[image]

For loop closures, there are two issues:
1. I cannot explain why there is a gap between this two images, because the matches look right. A wrong depth/calibration could explain this though.
[image]
2. Here is a wrong loop closure, but because the covariance is fixed, it is difficult to reject it after optimization:
[image]

Overall, the odometry is pretty bad, loop closure optimization won't really help to fix it. If it is a calibration issue, then you may be able to get better odometry by better calibration of your sensor, otherwise if calibration is perfect/depth is also perfect, then you may look for another odometry approach.
[image]