introlab/rtabmap_ros #1036: Using my own bag data

wangjinhoon, 2023-09-22:
Hello, i want to run my D435i rosbag file with the following command:

roslaunch rtabmap_ros rtabmap.launch \
    rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3" \
    depth_topic:=/camera/aligned_depth_to_color/image_raw \
    rgb_topic:=/camera/color/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    approx_sync:=false 
    
so i run below command:
   
roslaunch realsense2_camera rs_camera.launch     align_depth:=true\  unite_imu_method:="linear_interpolation"     enable_gyro:=true \     enable_accel:=true

and 

rosbag record -o my_rosbag.bag /camera/aligned_depth_to_color/image_raw /camera/color/image_raw /camera/color/camera_info

finally i run below command:
roslaunch rtabmap_launch rtabmap.launch     rtabmap_args:="--delete_db_on_start --Optimizer/GravitySigma 0.3"  \   depth_topic:=/camera/aligned_depth_to_color/image_raw  \   rgb_topic:=/camera/color/image_raw  \  camera_info_topic:=/camera/color/camera_info     approx_sync:=false  \   wait_imu_to_init:=false

and 

rosbag play my_rosbag_2023-09-22-13-38-59.bag

and, i show below error:
[ WARN] [1695357639.145484881]: Could not get transform from camera_link to camera_color_optical_frame after 0.200000 seconds (for stamp=1695357558.041034)! Error="canTransform: target_frame camera_link does not exist. canTransform: source_frame camera_color_optical_frame does not exist.. canTransform returned after 0.20112 timeout was 0.2.".

[ WARN] [1695357638.927236462]: /rtabmap/rgbd_odometry: Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. Parameter "approx_sync" is false, which means that input topics should have all the exact timestamp for the callback to be called.
/rtabmap/rgbd_odometry subscribed to (exact sync):
   /camera/color/image_raw \
   /camera/aligned_depth_to_color/image_raw \
   /camera/color/camera_info

how can i do?? 
Is there a guideline I can follow if I've made a mistake?
please help

nsa05605, 2023-09-22:
hello

I have same issue, but I solved it. lol lol

wangjinhoon, 2023-09-22:
Recording /tf_static along with it works well. Thank you.

closed 2023-09-22 by wangjinhoon

matlabbe, 2023-09-22:
You may always record `/tf` and `/tf_static` by default when recording a bag.