introlab/rtabmap_ros #1037: guidance for rendering 3D maps from /rtabmap/mapData topic in a web browser

360wcui, 2023-09-24:
Is there any example of rendering 3D maps from `/rtabmap/mapData` in a browser?

I am able to extract the messages using `react-ros` and `roslibjs`. Any advice on visualizing them in JavaScript would be greatly appreciated.

matlabbe, 2023-09-26:
You would have to reconstruct the point clouds from the depth image and calibration. In rtabmapviz and rviz's mapcloud plugin, we create a point cloud for each node in the map. On loop closures, we only need to update the point cloud's pose.

360wcui, 2023-09-26:
Is /voxel_cloud the consolidated point clouds from the each node?   If not,  what is the difference?   Thank you.

closed 2023-09-29 by 360wcui

MRSHIMMER, 2024-04-16:
@360wcui      Hi, I want to visualize the 3D reconstructed model in real time in a web application. Can I directly use the data in ‘/rtabmap/mapData’? But it's not clear to me what the data in this topic means.

matlabbe, 2024-04-17:
You may use `/rtabmap/cloud_map` directly (PointCloud2). To avoid streaming the whole cloud at each update, then you would need to subscribe to `mapData` topic. It is more complicated to handle thought, but it has everything you need (rgb image, depth image, calibration, optimized poses of all nodes in the graph). 

For the `voxel_cloud` that @360wcui was refering, it is the cloud created from latest camera data (at same framerate than camera topics) mainly used in the examples for visualization purpose or for obstacle detection, it is not an ouput of rtabmap node.

MRSHIMMER, 2024-04-17:
> You may use `/rtabmap/cloud_map` directly (PointCloud2). To avoid streaming the whole cloud at each update, then you would need to subscribe to `mapData` topic. It is more complicated to handle [...]

Thank you very much for your answer, and I also want to thank you for your contribution to this project. To avoid unnecessary expenses, I will try using data from the 'mapGraph' topic. I am currently using my phone and it is not convenient to access information. My initial idea is to refer to rviz's mapcloud plugin. Thank you again.