introlab/rtabmap_ros #1038: Pybind11 error when building rtabmap_ros:superpoint Docker image

mjpc13, 2023-09-25:
I am trying to build the Docker image **rtabmap_ros:superpoint** by doing:
```bash
git clone git@github.com:introlab/rtabmap_ros.git && cd rtabmap_ros
docker build -t rtabmap_ros:superpoint -f docker/noetic/superpoint/Dockerfile .
```

But I am having this error when the RTABMap compilation starts:
```
CMake Warning at /opt/conda/lib/python3.8/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:22 (message):
  static library kineto_LIBRARY-NOTFOUND not found.
Call Stack (most recent call first):
  /opt/conda/lib/python3.8/site-packages/torch/share/cmake/Torch/TorchConfig.cmake:127 (append_torchlib_if_found)
  CMakeLists.txt:385 (FIND_PACKAGE)

-- Found Torch: /opt/conda/lib/python3.8/site-packages/torch/include;/opt/conda/lib/python3.8/site-packages/torch/include/torch/csrc/api/include
-- Found Python3: /opt/conda/bin/python3.8 (found version "3.8.13") found components: Interpreter Development NumPy Development.Module Development.Embed 
-- Found Python3
CMake Error at CMakeLists.txt:395 (FIND_PACKAGE):
  By not providing "Findpybind11.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "pybind11",
  but CMake did not find one.

  Could not find a package configuration file provided by "pybind11" with any
  of the following names:

    pybind11Config.cmake
    pybind11-config.cmake

  Add the installation prefix of "pybind11" to CMAKE_PREFIX_PATH or set
  "pybind11_DIR" to a directory containing one of the above files.  If
  "pybind11" provides a separate development package or SDK, be sure it has
  been installed.

-- Configuring incomplete, errors occurred!
```

I also tried to add the package python3-pybind11 to the apt install command, but it gave me a different error:
```
CMake Error at /opt/ros/noetic/lib/rtabmap-0.21/RTABMap_coreTargets.cmake:55 (set_target_properties):
  The link interface of target "rtabmap::core" contains:

    pybind11::embed

  but the target was not found.  Possible reasons include:

    * There is a typo in the target name.
    * A find_package call is missing for an IMPORTED target.
    * An ALIAS target is missing.

Call Stack (most recent call first):
  /opt/ros/noetic/lib/rtabmap-0.21/RTABMapConfig.cmake:66 (include)
  CMakeLists.txt:10 (find_package)

CMake Generate step failed.  Build files cannot be regenerated correctly
```

Any idea how to fix this?

matlabbe, 2023-09-25:
It is a new dependency when building with python, it should be `pybind11-dev`

matlabbe, 2023-10-01:
The commit above should fix the pybind11 error

closed 2023-10-01 by matlabbe