introlab/rtabmap_ros #1039: While build the rtabmap and rtabmap_ros package in Xavier CPU with a galactic version . these error will come every time.  Any suggestions ??

sunilbodla, 2023-09-27:
/bin/ld: ../../bin/librtabmap_core.so.0.21.2: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/ConsoleApp/CMakeFiles/consoleApp.dir/build.make:216: bin/rtabmap-console] Error 1
make[1]: *** [CMakeFiles/Makefile2:1193: tools/ConsoleApp/CMakeFiles/consoleApp.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
/bin/ld: ../../bin/librtabmap_core.so.0.21.2: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/Camera/CMakeFiles/camera.dir/build.make:216: bin/rtabmap-camera] Error 1
make[1]: *** [CMakeFiles/Makefile2:1361: tools/Camera/CMakeFiles/camera.dir/all] Error 2
/bin/ld: ../../bin/librtabmap_core.so.0.21.2: undefined reference to `octomath::Pose6D::Pose6D(octomath::Pose6D const&)'
collect2: error: ld returned 1 exit status
make[2]: *** [tools/ExtractObject/CMakeFiles/extractObject.dir/build.make:216: bin/rtabmap-extractObject] Error 1
make[1]: *** [CMakeFiles/Makefile2:1305: tools/ExtractObject/CMakeFiles/extractObject.dir/all] Error 2
make: *** [Makefile:152: all] Error 2

anath93, 2023-09-27:
Looks like you have issue with octomap library missing, can you check with sudo apt install ros-galactic-octomap, then try to clean workspace and re-build ?

matlabbe, 2024-01-13:
Maybe related to https://github.com/introlab/rtabmap_ros/issues/737#issuecomment-1407157456