introlab/rtabmap_ros #1040: rtabmap

Zhang-JiLi, 2023-09-28:
Hello author, I am running rtabmap on euroc dataset ，but there were some issues.
[image]

matlabbe, 2023-10-01:
Is it rtabmap_ros binary version? Are you using this launch file https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_examples/launch/euroc_datasets.launch?

Zhang-JiLi, 2023-10-09:
I use this launch file https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_examples/launch/euroc_datasets.launch. Through GDB parsing, the following content is obtained:

Zhang-JiLi, 2023-10-09:
[image]
[image]
[image]
[image]
[image]

closed 2023-10-09 by Zhang-JiLi

matlabbe, 2023-10-15:
`malloc(): Memory corruption` from `cv::Mat::create` may be related to rtabmap built with a different opencv version than the one used on runtime. Were there warnings in cmake logs about one opencv hidden by another one?