introlab/rtabmap_ros #1043: Advice on Updating Point Cloud's Pose

360wcui, 2023-10-02:
I am porting rtabmap rviz plugins to Foxglove.  I notice that the point cloud's pose is not updated as well as one in RViz (pix attached).  

Here is the code snippet I have to update point cloud's pose using s.getPose().   Any advice on how to update the poses would be greatly appreciated. 
//            odomCloud = rtabmap::util3d::transformPointCloud(cloud, s.getPose());
//            pcl::toROSMsg(*odomCloud, *odomCloudMsg);
//            odomCloudMsg->header.frame_id = "map";
//            amavCloudPub.publish(odomCloudMsg);

[
[image]
[image]

matlabbe, 2023-10-03:
For the rviz plugin, each time we receive a new MapData topic, we get the current poses, 
https://github.com/introlab/rtabmap_ros/blob/68066b7e9a32d87766440a3f0033a0763f146350/rtabmap_rviz_plugins/src/MapCloudDisplay.cpp#L276-L282

then update all point clouds added so far with the current optimized poses (actually we just adjust the anchor of each point cloud, we don't transform individual points):
https://github.com/introlab/rtabmap_ros/blob/68066b7e9a32d87766440a3f0033a0763f146350/rtabmap_rviz_plugins/src/MapCloudDisplay.cpp#L731-L762

360wcui, 2023-10-03:
Thank you.  What would be the easiest way for me to publish all the updated point clouds for visualization?  Do you recommend doing so?  My rationale is to have C++ do all the heavy lifting and let JavaScript handle only the visualization.

matlabbe, 2023-10-03:
The MapData way would limit the bandwidth usage, as only latest node data has to be published. 

If you want to get the full map cloud after every frame, you can subscribe to /rtabmap/map_cloud topic, which is a PointCloud2 msg. 

Note that when the occupancy grid is created by 2d scans, the map cloud will be 2d.

360wcui, 2023-10-03:
I don't see any messages in /rtabmap/map_cloud.   Do I need to enable anything?   'rtabmap_ros' is 'tag: 0.21.1-noetic',   The bag is your 'demo_mapping.bag'.   Thank you.

closed 2023-10-06 by 360wcui

MRSHIMMER, 2024-04-16:
> The MapData way would limit the bandwidth usage, as only latest node data has to be published.
>
> If you want to get the full map cloud after every frame, you can subscribe to /rtabmap/map_cloud [...]

> /rtabmap/map_cloud

Does the "/rtabmap/map_cloud" topic mentioned in your answer actually refer to the "/rtabmap/cloud_map" topic?

matlabbe, 2024-04-17:
Yeah sorry, I corrected it.