introlab/rtabmap_ros #1046: [BUG] Incorrect subscription of RGBD topics when rgbd_cameras=0

rayferric, 2023-10-10:
See the implementation of [setupCallbacks](https://github.com/introlab/rtabmap_ros/blob/68066b7e9a32d87766440a3f0033a0763f146350/rtabmap_sync/src/CommonDataSubscriber.cpp#L356). If rtabmap_sync is compiled without RTABMAP_SYNC_MULTI_RGBD as it is done on ROS buildfarm and the node is run with `rgbd_cameras` parameter set to zero, the node will subscribe to both `rgbd_images` and `rgbd_image` callbacks at the same time.
This is how it looks in logs:
```
[rtabmap-9] [INFO] [1696926405.797020800] [rtabmap.rtabmap]: rtabmap: Database version = "0.21.1".
[rtabmap-9] [INFO] [1696926405.797077275] [rtabmap.rtabmap]: rtabmap: SLAM mode (Mem/IncrementalMemory=true)
[rtabmap-9] [INFO] [1696926405.823313099] [rtabmap.rtabmap]: Setup callbacks
[rtabmap-9] [INFO] [1696926405.823573975] [rtabmap.rtabmap]: rtabmap: subscribe_depth = false
[rtabmap-9] [INFO] [1696926405.823600875] [rtabmap.rtabmap]: rtabmap: subscribe_rgb = false
[rtabmap-9] [INFO] [1696926405.823609892] [rtabmap.rtabmap]: rtabmap: subscribe_stereo = false
[rtabmap-9] [INFO] [1696926405.823617556] [rtabmap.rtabmap]: rtabmap: subscribe_rgbd = true (rgbd_cameras=0)
[rtabmap-9] [INFO] [1696926405.823625661] [rtabmap.rtabmap]: rtabmap: subscribe_odom_info = false
[rtabmap-9] [INFO] [1696926405.823632974] [rtabmap.rtabmap]: rtabmap: subscribe_user_data = false
[rtabmap-9] [INFO] [1696926405.823640128] [rtabmap.rtabmap]: rtabmap: subscribe_scan = false
[rtabmap-9] [INFO] [1696926405.823646540] [rtabmap.rtabmap]: rtabmap: subscribe_scan_cloud = false
[rtabmap-9] [INFO] [1696926405.823653994] [rtabmap.rtabmap]: rtabmap: subscribe_scan_descriptor = false
[rtabmap-9] [INFO] [1696926405.823662019] [rtabmap.rtabmap]: rtabmap: queue_size      = 50
[rtabmap-9] [INFO] [1696926405.823666878] [rtabmap.rtabmap]: rtabmap: qos_image       = 0
[rtabmap-9] [INFO] [1696926405.823671186] [rtabmap.rtabmap]: rtabmap: qos_camera_info = 0
[rtabmap-9] [INFO] [1696926405.823675393] [rtabmap.rtabmap]: rtabmap: qos_scan        = 0
[rtabmap-9] [INFO] [1696926405.823679591] [rtabmap.rtabmap]: rtabmap: qos_odom        = 0
[rtabmap-9] [INFO] [1696926405.823683999] [rtabmap.rtabmap]: rtabmap: qos_user_data   = 0
[rtabmap-9] [INFO] [1696926405.823688037] [rtabmap.rtabmap]: rtabmap: approx_sync     = true
[rtabmap-9] [INFO] [1696926405.823702634] [rtabmap.rtabmap]: Setup rgbdX callback
[rtabmap-9] [INFO] [1696926405.825769882] [rtabmap.rtabmap]: Setup rgbd callback
[rtabmap-9] [INFO] [1696926405.826745989] [rtabmap.rtabmap]: 
[rtabmap-9] rtabmap subscribed to:
[rtabmap-9]    /d455_front/rgbd_sync/rgbd_image
```
It happens due to a missing `else` before `if(rgbdCameras>1)`.

matlabbe, 2023-10-15:
The commits above should fix this issue on ros1 and ros2

closed 2023-10-15 by matlabbe

shivamtrip, 2024-10-03:
`[rgbd_sync-2] [INFO] [1727969194.305150319] [camera_2.rgbd_sync_2]: rgbd_sync_2: approx_sync = false
[rgbd_sync-2] [INFO] [1727969194.305254483] [camera_2.rgbd_sync_2]: rgbd_sync_2: topic_queue_size  = 1
[rgbd_sync-2] [INFO] [1727969194.305263040] [camera_2.rgbd_sync_2]: rgbd_sync_2: sync_queue_size  = 10
[rgbd_sync-2] [INFO] [1727969194.305267471] [camera_2.rgbd_sync_2]: rgbd_sync_2: qos             = 1
[rgbd_sync-2] [INFO] [1727969194.305271939] [camera_2.rgbd_sync_2]: rgbd_sync_2: qos_camera_info = 1
[rgbd_sync-2] [INFO] [1727969194.305276540] [camera_2.rgbd_sync_2]: rgbd_sync_2: depth_scale = 1.000000
[rgbd_sync-2] [INFO] [1727969194.305289956] [camera_2.rgbd_sync_2]: rgbd_sync_2: decimation = 1
[rgbd_sync-2] [INFO] [1727969194.305294497] [camera_2.rgbd_sync_2]: rgbd_sync_2: compressed_rate = 0.000000
[rgbd_sync-1] [INFO] [1727969194.306040160] [camera_1.rgbd_sync_1]: rgbd_sync_1: approx_sync = false
[rgbd_sync-1] [INFO] [1727969194.306146512] [camera_1.rgbd_sync_1]: rgbd_sync_1: topic_queue_size  = 1
[rgbd_sync-1] [INFO] [1727969194.306155174] [camera_1.rgbd_sync_1]: rgbd_sync_1: sync_queue_size  = 10
[rgbd_sync-1] [INFO] [1727969194.306159587] [camera_1.rgbd_sync_1]: rgbd_sync_1: qos             = 1
[rgbd_sync-1] [INFO] [1727969194.306163956] [camera_1.rgbd_sync_1]: rgbd_sync_1: qos_camera_info = 1
[rgbd_sync-1] [INFO] [1727969194.306169687] [camera_1.rgbd_sync_1]: rgbd_sync_1: depth_scale = 1.000000
[rgbd_sync-1] [INFO] [1727969194.306182662] [camera_1.rgbd_sync_1]: rgbd_sync_1: decimation = 1
[rgbd_sync-1] [INFO] [1727969194.306187067] [camera_1.rgbd_sync_1]: rgbd_sync_1: compressed_rate = 0.000000
[rgbd_sync-2] [INFO] [1727969194.317703674] [camera_2.rgbd_sync_2]: 
[rgbd_sync-2] rgbd_sync_2 subscribed to (exact sync):
[rgbd_sync-2]    /camera_image_2,
[rgbd_sync-2]    /stereo_depths_2,
[rgbd_sync-2]    /camera_info_2
[rgbd_sync-1] [INFO] [1727969194.319000134] [camera_1.rgbd_sync_1]: 
[rgbd_sync-1] rgbd_sync_1 subscribed to (exact sync):
[rgbd_sync-1]    /camera_image_1,
[rgbd_sync-1]    /stereo_depths_1,
[rgbd_sync-1]    /camera_info_1
[rgbd_odometry-3] [INFO] [1727969194.320283351] [rtabmap.rgbd_odometry]: Odometry: frame_id               = right_stereo_camera_1
[rgbd_odometry-3] [INFO] [1727969194.320394350] [rtabmap.rgbd_odometry]: Odometry: odom_frame_id          = odom
[rgbd_odometry-3] [INFO] [1727969194.320402207] [rtabmap.rgbd_odometry]: Odometry: publish_tf             = true
[rgbd_odometry-3] [INFO] [1727969194.320406643] [rtabmap.rgbd_odometry]: Odometry: wait_for_transform     = 0.100000
[rgbd_odometry-3] [INFO] [1727969194.320419369] [rtabmap.rgbd_odometry]: Odometry: log_to_rosout_level    = 4
[rgbd_odometry-3] [INFO] [1727969194.320436979] [rtabmap.rgbd_odometry]: Odometry: initial_pose           = xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000
[rgbd_odometry-3] [INFO] [1727969194.320441897] [rtabmap.rgbd_odometry]: Odometry: ground_truth_frame_id  = 
[rgbd_odometry-3] [INFO] [1727969194.320446220] [rtabmap.rgbd_odometry]: Odometry: ground_truth_base_frame_id = right_stereo_camera_1
[rgbd_odometry-3] [INFO] [1727969194.320450600] [rtabmap.rgbd_odometry]: Odometry: config_path            = 
[rgbd_odometry-3] [INFO] [1727969194.320454719] [rtabmap.rgbd_odometry]: Odometry: publish_null_when_lost = true
[rgbd_odometry-3] [INFO] [1727969194.320458911] [rtabmap.rgbd_odometry]: Odometry: publish_compressed_sensor_data = false
[rgbd_odometry-3] [INFO] [1727969194.320464502] [rtabmap.rgbd_odometry]: Odometry: guess_frame_id         = 
[rgbd_odometry-3] [INFO] [1727969194.320468676] [rtabmap.rgbd_odometry]: Odometry: guess_min_translation  = 0.000000
[rgbd_odometry-3] [INFO] [1727969194.320473323] [rtabmap.rgbd_odometry]: Odometry: guess_min_rotation     = 0.000000
[rgbd_odometry-3] [INFO] [1727969194.320478593] [rtabmap.rgbd_odometry]: Odometry: guess_min_time         = 0.000000
[rgbd_odometry-3] [INFO] [1727969194.320483277] [rtabmap.rgbd_odometry]: Odometry: expected_update_rate   = 0.000000 Hz
[rgbd_odometry-3] [INFO] [1727969194.320487951] [rtabmap.rgbd_odometry]: Odometry: max_update_rate        = 0.000000 Hz
[rgbd_odometry-3] [INFO] [1727969194.320492521] [rtabmap.rgbd_odometry]: Odometry: min_update_rate        = 0.000000 Hz
[rgbd_odometry-3] [INFO] [1727969194.320497087] [rtabmap.rgbd_odometry]: Odometry: wait_imu_to_init       = false
[rgbd_odometry-3] [INFO] [1727969194.320501377] [rtabmap.rgbd_odometry]: Odometry: sensor_data_compression_format = .jpg
[rgbd_odometry-3] [INFO] [1727969194.320505625] [rtabmap.rgbd_odometry]: Odometry: sensor_data_parallel_compression = true
[rgbd_odometry-3] [INFO] [1727969194.320512566] [rtabmap.rgbd_odometry]: Odometry: stereoParams_=0 visParams_=1 icpParams_=0
[rgbd_odometry-3] [INFO] [1727969194.340824355] [rtabmap.rgbd_odometry]: RGBDOdometry: approx_sync    = false
[rgbd_odometry-3] [INFO] [1727969194.340859913] [rtabmap.rgbd_odometry]: RGBDOdometry: topic_queue_size = 1
[rgbd_odometry-3] [INFO] [1727969194.340865935] [rtabmap.rgbd_odometry]: RGBDOdometry: sync_queue_size  = 5
[rgbd_odometry-3] [INFO] [1727969194.340870388] [rtabmap.rgbd_odometry]: RGBDOdometry: qos             = 1
[rgbd_odometry-3] [INFO] [1727969194.340874739] [rtabmap.rgbd_odometry]: RGBDOdometry: qos_camera_info = 1
[rgbd_odometry-3] [INFO] [1727969194.340879264] [rtabmap.rgbd_odometry]: RGBDOdometry: subscribe_rgbd = true
**[rgbd_odometry-3] [INFO] [1727969194.340883754] [rtabmap.rgbd_odometry]: RGBDOdometry: rgbd_cameras   = 1**
[... cut ...]
[rtabmap_viz-4] [INFO] [1727969200.471537760] [rtabmap.rtabmap_viz]: rtabmap_viz: approx_sync     = false
[rtabmap_viz-4] [INFO] [1727969200.471542767] [rtabmap.rtabmap_viz]: Setup rgbdX callback
[rtabmap_viz-4] [INFO] [1727969200.472195260] [rtabmap.rtabmap_viz]: 
[rtabmap_viz-4] rtabmap_viz subscribed to (exact sync):
[rtabmap_viz-4]    /rtabmap/odom \
**[rtabmap_viz-4]    /rtabmap/rgbd_images**
[rtabmap_viz-4] [INFO] [1727969200.473272351] [rtabmap.rtabmap_viz]: rtabmap_viz started.
[rtabmap_viz-4] [WARN] [1727969201.464961271] [rtabmap.rtabmap_viz]: Service "rtabmap/resume" not available.
`