introlab/rtabmap_ros #1047: Visualize created map

nicolasukd, 2023-10-10:
Where can i visualize the map thats created? 
I know that its located in `~/.ros/rtabmap.db` by default, but i dont know how to open it.

matlabbe, 2023-10-12:
You can open it with:
```
$ rtabmap ~/.ros/rtabmap.db
```
Or 
```
$ rtabmap-databaseViewer ~/.ros/rtabmap.db
```

matlabbe, 2023-10-15:
Can you do `which` to see which rtabmap is started? You may explicitly call `~/rtabmap/build/bin/rtabmap` to make sure it is the one you just built with latest master version.

nicolasukd, 2023-10-17:
[image]

How do i explicitly call ~/rtabmap/build/bin/rtabmap?

matlabbe, 2023-10-17:
That database may have been created on another computer using a more recent version of rtabmap. If you did:
```
cd rtabmap
git pull origin master
cd build
make
sudo make install
```
I'll suggest to remove ros binaries:
```
sudo apt remove ros-melodic-rtabmap
```
so you will make sure you will be using 0.21 installed from source instead.

To explicitly call `~/rtabmap/build/bin/rtabmap`, just do `~/rtabmap/build/bin/rtabmap` in terminal.

nicolasukd, 2023-11-06:
It worked. Im wondering now if i can visualize this map in windows.

matlabbe, 2023-11-11:
You should be able to visualize on windows using the 0.21.0 binaries (see assets section): https://github.com/introlab/rtabmap/releases/tag/0.21.0

closed 2023-11-11 by matlabbe