introlab/rtabmap_ros #1048: missimg：rtabmap_ros don't have launch

sun-rabbit, 2023-10-12:
when i successfully run colcon build --symlink-install，
ros2 launch rtabmap_ros turtlebot3_scan.launch.py
but rtabmap_ros don't have launch
ubuntu20.04，ros2 foxy

sun-rabbit, 2023-10-12:
[image]
and 
# OR with rtabmap.launch.py
ros2 launch rtabmap_launch rtabmap.launch.py \
   visual_odometry:=false \
   frame_id:=base_footprint \
   subscribe_scan:=true depth:=false \
   approx_sync:=true \
   odom_topic:=/odom \
   scan_topic:=/scan \
   qos:=2 \
   args:="-d --RGBD/NeighborLinkRefining true --Reg/Strategy 1" \
   use_sim_time:=true \
   rviz:=true

#启动导航
ros2 launch nav2_bringup navigation_launch.py use_sim_time:=True
ros2 launch nav2_bringup rviz_launch.py
No 3D map appears， give me some guidance，please, thank you very much

matlabbe, 2023-10-15:
`ros2 launch rtabmap_demos turtlebot3_scan.launch.py`

sun-rabbit, 2023-10-16:
> `ros2 launch rtabmap_demos turtlebot3_scan.launch.py`

but I'm having issues getting an image to appear when running the following command:

ros2 launch rtabmap_ros realsense_d435i_color.launch.py

My setup is:
ubuntu20.04，ros2 foxy
旭日x3派 and i have a jeston nano

matlabbe, 2023-10-16:
This one would be:
```
ros2 launch rtabmap_examples realsense_d435i_color.launch.py
```

sun-rabbit, 2023-10-19:
> 这一个是：
> 
> ```
> ros2 launch rtabmap_examples realsense_d435i_color.launch.py
> ```
Hello, I'm very sorry for the delayed response. My D435i camera is not successfully producing point cloud data. I have switched to an Astra Pro camera, but when I run 'ros2 launch rtabmap_examples realsense_d400.launch.py' and modify 'realsense_d400.launch.py' to:

```python
remappings=[
  #('rgb/image', '/camera/color/image_raw'),
  ('rgb/image', '/camera/color/image_raw'),
  #('rgb/camera_info', '/camera/color/camera_info'),
  ('rgb/camera_info', '/camera/color/camera_info'),
  #('depth/image', '/camera/aligned_depth_to_color/image_raw'),
  ('depth/image', '/camera/depth/image_raw')],
```

I still haven't been successful. I would greatly appreciate any guidance you can provide.

matlabbe, 2023-10-19:
I would stick on D435i example as I cannot have an Astra to make sure it is working. Normally, if you follow the instructions on top of this file https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_color.launch.py it should do something:

```
# Requirements:
#   A realsense D435i
#   Install realsense2 ros2 package (ros-$ROS_DISTRO-realsense2-camera)
# Example:
#   $ ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=1 enable_sync:=true
#
#   $ ros2 launch rtabmap_examples realsense_d435i_color.launch.py
```

sun-rabbit, 2023-10-19:
> 我会继续以 D435i 为例，因为我无法使用 Astra 来确保它正常工作。[通常，如果您按照此文件https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_color.launch.py](https://github.com/introlab/rtabmap_ros/blob/ros2/rtabmap_examples/launch/realsense_d435i_color.launch.py)​​ 顶部的说明进行操作，它应该执行以下操作：
> 
> ```
> # Requirements:
> #   A realsense D435i
> #   Install realsense2 ros2 package (ros-$ROS_DISTRO-realsense2-camera)
> # Example:
> #   $ ros2 launch realsense2_camera rs_launch.py enable_gyro:=true enable_accel:=true unite_imu_method:=1 enable_sync:=true
> #
> #   $ ros2 launch rtabmap_examples realsense_d435i_color.launch.py
> ```

Hello, just now I saw an example:

In ROS, to start the Orbbec Astra camera, you use the following command:

Copy code
roslaunch astra_launch astra.launch
To initiate RTAB-Map ROS, you can use this command in ROS:

go
Copy code
roslaunch rtabmap_ros rtabmap.launch rtabmap_args:="--delete_db_on_start" rviz:=true rtabmapviz:=false
If the window disappears quickly, try running 'rtabmap' first. After closing 'rtabmap,' execute the previous command again.

How can I execute these commands in ROS 2? Thank you.

sun-rabbit, 2023-10-22:
> I would stick on D435i example as I cannot have an Astra to make sure it is working. Normally, if you follow the instructions on top of this file [...]
Hello! These two days I bought a D435i. However, when I use `ros2 launch rtabmap_examples realsense_d435i_color.launch.py`, RTAB-Map does not display any images, and the topic rate is as low as around 4 Hz. My D435i's firmware version is 05.13.00.55, and when I don't run it, the Hz is about 19. Can you please help me? I can't figure out what the problem is.

matlabbe, 2023-10-25:
Do you have repeating warnings on terminal telling that rgbd_odometry or rtabmap cannot synchronize (or didn't receive) topics for 5 seconds? You may try on a computer that you know you can get 30 Hz framerate on all streams.

sun-rabbit, 2023-10-26:
> Do you have repeating warnings on terminal telling that rgbd_odometry or rtabmap cannot synchronize (or didn't receive) topics for 5 seconds? You may try on a computer that you know you can get 30 Hz framerate on all streams.

Yes, I have this prompt. What does it indicate?

matlabbe, 2023-10-26:
It indicates what is described in the message. If the topics are indeed published (check with `ros2 topic hz`), but the rate is inconsistent between them, then maybe ros message filters just cannot synchronize them together, you would need to fix the camera driver (or compare on another faster computer). If they are all published at exact same rate, it could ve qos issue. Do `ros2 ropic info --verbose ...` to make sure the publisher and subscriber nodes use the same quality of service, if not, you can manually change it for subscribers with `qos` parameter. However, as it is an example I already tested myself, I think on your side it could be a computer/usb issue causing topics not published all at 30 hz.