introlab/rtabmap_ros #1049: Make error

ManChrys, 2023-10-12:
Hello,

i have this error

`[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/OdometryThread.cpp.o
/home/geo/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp: In constructor ‘rtabmap::OptimizerG2O::OptimizerG2O(const ParametersMap&)’:
/home/geo/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:157:133: error: no matching function for call to ‘g2o::Factory::registerType(const char [17], std::_MakeUniq<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >::__single_object)’
 erType("CACHE_SE3_OFFSET", std::make_unique<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >());
                                                                                                   ^
In file included from /home/geo/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:46:0:
/opt/ros/melodic/include/g2o/core/factory.h:62:12: note: candidate: void g2o::Factory::registerType(const string&, g2o::AbstractHyperGraphElementCreator*)
       void registerType(const std::string& tag, AbstractHyperGraphElementCreator* c);
            ^~~~~~~~~~~~
/opt/ros/melodic/include/g2o/core/factory.h:62:12: note:   no known conversion for argument 2 from ‘std::_MakeUniq<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >::__single_object {aka std::unique_ptr<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset>, std::default_delete<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> > >}’ to ‘g2o::AbstractHyperGraphElementCreator*’
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryF2M.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryMono.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryF2F.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryFovis.cpp.o
[  7%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/odometry/OdometryViso2.cpp.o
corelib/src/CMakeFiles/rtabmap_core.dir/build.make:929: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o' failed
make[2]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/optimizer/OptimizerG2O.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
CMakeFiles/Makefile2:911: recipe for target 'corelib/src/CMakeFiles/rtabmap_core.dir/all' failed
make[1]: *** [corelib/src/CMakeFiles/rtabmap_core.dir/all] Error 2

Makefile:155: recipe for target 'all' failed
make: *** [all] Error 2
`

This error has to do with g2o version ? i tried both these versions and i have the same problem. Should i try an older one ?

[image]

sun-rabbit, 2023-10-12:
What are the versions of ubuntu and ros

ManChrys, 2023-10-12:
ubuntu 18.04 
ros1
melodic

sun-rabbit, 2023-10-12:
> ubuntu 18.04 ros1 melodic

sorry ，my
ubuntu20.04
ros2
foxy

ManChrys, 2023-10-12:
i fix this error by

1)sudo apt remove ros-melodic-rtabmap*
2)git clone https://github.com/RainerKuemmerle/g2o.git
    mkdir build
    cd build
    cmake ../
    make
    sudo make install

closed 2023-10-12 by ManChrys