introlab/rtabmap_ros #1050: catkin_make -j4 error

ManChrys, 2023-10-12:
hello i had this error #1049 . To resolve this error i did sudo apt remove ros-melodic-rtabmap*

and after that.
```
[ 80%] Linking CXX executable /home/geo/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::RuntimeParameters::RuntimeParameters(bool, bool, int, int, sl::REFERENCE_FRAME, bool)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::setFromCameraID(int, sl::BUS_TYPE)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::getCameraInformation(sl::Resolution)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::PositionalTrackingParameters::PositionalTrackingParameters(sl::Transform, bool, bool, sl::String, bool, bool, bool, float, bool, sl::POSITIONAL_TRACKING_MODE)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::~String()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::String(char const*)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::String(sl::String const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InitParameters::InitParameters(sl::RESOLUTION, int, bool, sl::DEPTH_MODE, sl::UNIT, sl::COORDINATE_SYSTEM, bool, int, float, float, bool, int, bool, sl::String, int, CUctx_st*, sl::InputType, sl::String, bool, bool, sl::String, float, bool, float, bool)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::InputType(sl::InputType const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Transform::Transform()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::grab(sl::RuntimeParameters)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::toString(sl::POSITIONAL_TRACKING_STATE const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Transform::Transform(sl::Transform const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::CalibrationParameters::getCameraBaseline()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::getSensorsData(sl::SensorsData&, sl::TIME_REFERENCE)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::setFromSVOFile(sl::String)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::InputType()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::Camera()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Pose::Pose()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::open(sl::InitParameters)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::retrieveImage(sl::Mat&, sl::VIEW, sl::MEM, sl::Resolution)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Pose::~Pose()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Transform::getOrientation() const'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::c_str() const'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::String()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `unsigned char* sl::Mat::getPtr<unsigned char>(sl::MEM) const'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::getPosition(sl::Pose&, sl::REFERENCE_FRAME)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::SensorsData::SensorsData()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Mat::Mat()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::enablePositionalTracking(sl::PositionalTrackingParameters)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Mat::~Mat()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::retrieveMeasure(sl::Mat&, sl::MEASURE, sl::MEM, sl::Resolution)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::~Camera()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Pose::Pose(sl::Transform const&, unsigned long long, int)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::toString(sl::ERROR_CODE const&)'
collect2: error: ld returned 1 exit status
rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_pub.dir/build.make:576: recipe for target '/home/geo/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_pub' failed
make[2]: *** [/home/geo/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_pub] Error 1
CMakeFiles/Makefile2:6712: recipe for target 'rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_pub.dir/all' failed
make[1]: *** [rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_pub.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 80%] Built target odometry2graph
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::RuntimeParameters::RuntimeParameters(bool, bool, int, int, sl::REFERENCE_FRAME, bool)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::setFromCameraID(int, sl::BUS_TYPE)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::getCameraInformation(sl::Resolution)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::PositionalTrackingParameters::PositionalTrackingParameters(sl::Transform, bool, bool, sl::String, bool, bool, bool, float, bool, sl::POSITIONAL_TRACKING_MODE)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::~String()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::String(char const*)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::String(sl::String const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InitParameters::InitParameters(sl::RESOLUTION, int, bool, sl::DEPTH_MODE, sl::UNIT, sl::COORDINATE_SYSTEM, bool, int, float, float, bool, int, bool, sl::String, int, CUctx_st*, sl::InputType, sl::String, bool, bool, sl::String, float, bool, float, bool)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::InputType(sl::InputType const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Transform::Transform()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::grab(sl::RuntimeParameters)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::toString(sl::POSITIONAL_TRACKING_STATE const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Transform::Transform(sl::Transform const&)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::CalibrationParameters::getCameraBaseline()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::getSensorsData(sl::SensorsData&, sl::TIME_REFERENCE)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::setFromSVOFile(sl::String)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::InputType::InputType()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::Camera()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Pose::Pose()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::open(sl::InitParameters)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::retrieveImage(sl::Mat&, sl::VIEW, sl::MEM, sl::Resolution)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Pose::~Pose()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Transform::getOrientation() const'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::c_str() const'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::String::String()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `unsigned char* sl::Mat::getPtr<unsigned char>(sl::MEM) const'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Camera::getPosition(sl::Pose&, sl::REFERENCE_FRAME)'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::SensorsData::SensorsData()'
/usr/local/lib/librtabmap_core.so.0.21.2: undefined reference to `sl::Mat::Mat()'
[... cut ...]
rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/build.make:576: recipe for target '/home/geo/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub' failed
make[2]: *** [/home/geo/catkin_ws/devel/lib/rtabmap_demos/wifi_signal_sub] Error 1
CMakeFiles/Makefile2:6738: recipe for target 'rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/all' failed
make[1]: *** [rtabmap_ros/rtabmap_demos/CMakeFiles/rtabmap_wifi_signal_sub.dir/all] Error 2
[ 82%] Built target interactive_slam
Makefile:145: recipe for target 'all' failed
make: *** [all] Error 2
Invoking "make -j4" failed

```

sun-rabbit, 2023-10-13:
j2  
and
sudo apt remove ros-foxy-rtabmap*

this is  my csdn https://blog.csdn.net/weixin_45007300/article/details/133746311?spm=1001.2014.3001.5502

matlabbe, 2023-10-15:
It seems rtabmap library is built with ZED support, check if the zed library is correctly detected with the installed rtabmap library: 
```
ldd /usr/local/lib/librtabmap_core.so | grep zed
	libsl_zed.so => /usr/local/zed/lib/libsl_zed.so (0x00007f716d324000)
```
As @sun-rabbit , make sure you don't have another rtabmap installed.

ManChrys, 2023-10-16:
hello,

i did `geo@geo-X570-GAMING-X:~/rtabmap/build$ ldd /usr/local/lib/librtabmap_core.so | grep zed`
```
	libsl_zed.so => /usr/local/zed/lib/libsl_zed.so (0x00007f30f6f30000)
	libzed_open_capture.so => /usr/local/lib/libzed_open_capture.so (0x00007f30f6b0f000)
```

i did `cmake -DWITH_ZED=OFF ..`

```
-- Configuring done
-- Generating done
-- Build files have been written to: /home/geo/rtabmap/build
```

but when i `make -j6`

```
[ 28%] Building CXX object corelib/src/CMakeFiles/rtabmap_core.dir/Registration.cpp.o
/home/geo/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp: In constructor ‘rtabmap::OptimizerG2O::OptimizerG2O(const ParametersMap&)’:
/home/geo/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:157:133: error: no matching function for call to ‘g2o::Factory::registerType(const char [17], std::_MakeUniq<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >::__single_object)’
   g2o::Factory::instance()->registerType("CACHE_SE3_OFFSET", std::make_unique<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >());
                                                                                                                                     ^
In file included from /home/geo/rtabmap/corelib/src/optimizer/OptimizerG2O.cpp:46:0:
/opt/ros/melodic/include/g2o/core/factory.h:62:12: note: candidate: void g2o::Factory::registerType(const string&, g2o::AbstractHyperGraphElementCreator*)
       void registerType(const std::string& tag, AbstractHyperGraphElementCreator* c);
            ^~~~~~~~~~~~
/opt/ros/melodic/include/g2o/core/factory.h:62:12: note:   no known conversion for argument 2 from ‘std::_MakeUniq<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> >::__single_object {aka std::unique_ptr<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset>, std::default_delete<g2o::HyperGraphElementCreator<g2o::CacheSE3Offset> > >}’ to ‘g2o::AbstractHyperGraphElementCreator*’
```

i have installed 

```
sudo apt install ros-melodic-rtabmap*
ros-melodic-libg2o
```

Thank you.

matlabbe, 2023-10-16:
I'll need to doublecheck what was exactly the g2o version in melodic, rtabmap would have support it, unless it is related to this [update](https://github.com/introlab/rtabmap/commit/85cc6fe3c742855ad16c8442895e12dbb10b6e8b) for latest g2o version (thus breaking old support). Supporting the different versions of g2o is quite hard, as they don't use any code versioning. The workaround would be to install g2o from source:
```
sudo apt remove ros-melodic-rtabmap* ros-melodic-libg2o

git clone --branch 20230223_git https://github.com/RainerKuemmerle/g2o
cd g2o
mkdir build
cmake ..
make -j4
sudo make install
```