introlab/rtabmap_ros #1051: does rtabmap depend on gazebo classic?

aravi-2001, 2023-10-17:
im trying to get rtabmap work properly on my ROV that works using gazebo garden. I'm having so many issues for the last 3 weeks wiht this integration--my rtabmap wasnt able to be installed from apt-get, i had broken installations so i had to source install it. i'm getting several warnings on my terminal due to this strange way of installations related to the optimizers and what not. im starting to wonder if gazebo garden is a problem causing so many issues because when i tested it on a drone in classic, i had absolutely no issues.

matlabbe, 2023-10-17:
Only `rtabmap_demos` (and `rtabmap_ros` indirectly) is depending on gazebo (on default version of ROS distro). You could install other rtabmap binary packages and only extract the `rtabmap_demos` folder from this repo to put it in your local `ros2_ws` if you want to try the demos with your new gazebo version.

To get only core packages not depending on gazebo:
```
sudo apt install ros-$ROS_DISTRO-rtabmap-launch
```

aravi-2001, 2023-10-17:
Thanks for your reply! The install was successful but i see similiar warnings as before( although the g20 optimizer warnings are gone)

[ WARN] (2023-10-17 18:34:30.269) SensorData.cpp:762::uncompressDataConst() Requested laser scan data, but the sensor data (27682) doesn't have laser scan.
[ WARN] (2023-10-17 18:34:30.269) SensorData.cpp:762::uncompressDataConst() Requested laser scan data, but the sensor data (27841) doesn't have laser scan.

i wonder why i keep getting these warnings? im using the following launch: 
roslaunch rtabmap_launch rtabmap.launch depth_topic:=/slam/depth_image rgb_topic:=/slam/image camera_info_topic:=/slam/camera_info rtabmapviz:=true mapping:=true frame_id:=bluerov2/base_link/rgbd_camera approx_sync:=false

closed 2023-10-25 by aravi-2001

aravi-2001, 2023-10-25:
how can i use map assembler from rtabmap launch? everytime i try it i get a package not found error