introlab/rtabmap_ros #1052: /voxel_cloud looking tilted

aravi-2001, 2023-10-18:
i read in another issue that /voxel_cloud uses the RGBD topics to make a cloud. For me that cloud is coming out to be a little skewed, hence the whole map looks strangely skewed. How can i verify if my input paramters are correct? I'm using RGBD camera from gazebo garden and receiving it in ros1 via the bridge.

matlabbe, 2023-10-18:
If the point cloud is skewed, it looks like calibration parameter are wrong. What looks like the camera info topic?

matlabbe, 2023-10-18:
What is the actual resolution of the rgb image? The camera info says 720x720, but cx/cy are wrong, they should be qround the half of the resolution, not 160 and 120. If there is nothing appearing on loop closure detection panel, the mapping node is not working, check the terminal for warnings.

aravi-2001, 2023-10-18:
yes, i increased the resolution, the original was 256, i changed it to 720 in the sdf 

the terminal warnings look like this:
[ INFO] [1697641304.427944078]: rtabmap (29712): Rate=0.50s, Limit=0.000s, Conversion=0.0021s, RTAB-Map=0.1864s, Maps update=0.0005s pub=0.0023s (local map=61, WM=4721)
[ INFO] [1697641304.428494207]: Odom: quality=322, std dev=0.037289m|0.016851rad, update time=0.074814s
[ INFO] [1697641304.716556198]: Odom: quality=302, std dev=0.071010m|0.014651rad, update time=0.082818s
[ INFO] [1697641304.956128745]: Odom: quality=311, std dev=0.050619m|0.016851rad, update time=0.078620s
[ WARN] (2023-10-18 11:01:45.089) SensorData.cpp:762::uncompressDataConst() Requested laser scan data, but the sensor data (15234) doesn't have laser scan.
[ WARN] (2023-10-18 11:01:45.089) SensorData.cpp:762::uncompressDataConst() Requested laser scan data, but the sensor data (29713) doesn't have laser scan.
[ WARN] (2023-10-18 11:01:45.130) Rtabmap.cpp:2922::process() Rejected loop closure 15234 -> 29713: 
[ INFO] [1697641305.144031193]: rtabmap (29713): Rate=0.50s, Limit=0.000s, Conversion=0.0024s, RTAB-Map=0.1797s, Maps update=0.0014s pub=0.0031s (local map=61, WM=4721)
[ INFO] [1697641305.245070426]: Odom: quality=342, std dev=0.056821m|0.015492rad, update time=0.078233s
[ INFO] [1697641305.418598746]: Odom: quality=330, std dev=0.088469m|0.017423rad, update time=0.073804s

matlabbe, 2023-10-18:
You should adjust the camera calibration parameters if you change the resolution.

Rtabmap looks running. For the warnings about laserscan, I dont know what is your config, subscriptions... if it is from rtabmapviz, you may disable laserscan rendering under Preferences->3d Rendering panel