introlab/rtabmap_ros #1054: Warn: Did not receive data since 5 seconds!

sun-rabbit, 2023-10-19:
hi，everyone i Using Ros2 Foxy inside a Jetson jeston agx orin with ubuntu 20.4 and a astro pro , for some reason launching examples
###########
ros2 launch rtabmap_launch rtabmap.launch.py
rtabmap_args:="--delete_db_on_start"
rgb_topic:=/camera/color/image_raw
depth_topic:=/camera/depth/image_raw
camera_info_topic:=/camera/color/camera_info
frame_id:=base_link
approx_sync:=true
qos:=2
rviz:=true
queue_size:=20
subscribe_odom_info:=false
###############
[rtabmap_viz-3] [WARN] [1697700727.832576678] [rtabmap.rtabmap_viz]: rtabmap_viz: Did not receive data since 5 seconds! Make sure the input topics are published ("$ ros2 topic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=20).

#############
I ensure that I have synchronized the time using NTP.

sun-rabbit, 2023-10-19:
Hello, I successfully configured the camera using the parameters below, 
`ros2 launch rtabmap_launch rtabmap.launch.py \
    rtabmap_args:="--delete_db_on_start" \
    rgb_topic:=/camera/color/image_raw \
    depth_topic:=/camera/depth/image_raw \
    camera_info_topic:=/camera/color/camera_info \
    frame_id:=camera_link \
    use_sim_time:=true \
    approx_sync:=true \
    qos:=2 \
    rviz:=true \
    queue_size:=30`
[image]

but I'm still getting errors in my terminal. When mapping, the camera doesn't seem to update the newly scanned areas in the map
` rgbd_odometry-1] [INFO] [1697708399.950951321] [rtabmap.rgbd_odometry]: Odom: quality=31, std dev=0.011351m|0.021909rad, update time=0.000000s
[rviz2-4] [INFO] [1697708399.992974740] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1697708395.772 for reason 'Unknown'
[rgbd_odometry-1] [WARN] [1697708400.502272086] [rtabmap.rgbd_odometry]: The time difference between rgb and depth frames is high (diff=0.153943s, rgb=1697708399.896419s, depth=1697708399.742476s). You may want to set approx_sync_max_interval lower than 0.02s to reject spurious bad synchronizations or use approx_sync=false if streams have all the exact same timestamp.
[rgbd_odometry-1] [INFO] [1697708400.543996130] [rtabmap.rgbd_odometry]: Odom: quality=30, std dev=0.036308m|0.024346rad, update time=0.000000s
[rviz2-4] [INFO] [1697708400.569555316] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1697708396.472 for reason 'Unknown'
^C[WARNING] [launch]: user interrupted with ctrl-c (SIGINT)
[point_cloud_xyzrgb-5] [INFO] [1697708400.610105938] [rclcpp]: signal_handler(signal_value=2)
[rtabmap-2] [INFO] [1697708400.610142522] [rclcpp]: signal_handler(signal_value=2)
[rtabmap-2] [WARN] [1697708400.612287530] [rtabmap.rtabmap]: We received odometry message, but we cannot get the corresponding TF odom->camera_link at data stamp 1697708375.304139s (odom msg stamp is 1697708375.452965s). Make sure TF of odometry is also published to get more accurate pose estimation. This warning is only printed once.
[rtabmap-2] [ERROR] [1697708400.612323280] [rtabmap.rtabmap]: Could not convert rgb/depth msgs! Aborting rtabmap update...
[rtabmap_viz-3] [INFO] [1697708400.611813528] [rtabmap.rtabmap_viz]: rtabmap_viz stopping spinner...
[rtabmap-2] [ERROR] (2023-10-19 17:39:10.449) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:11.367) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:12.439) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:13.714) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:14.418) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:15.622) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:17.262) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:18.375) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:19.943) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:20.143) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:21.313) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:23.088) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:24.118) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:25.114) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:26.288) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:27.338) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:28.311) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[rtabmap-2] [ERROR] (2023-10-19 17:39:29.660) Rtabmap.cpp:1343::process() RGB-D SLAM mode is enabled, memory is incremental but no odometry is provided. Image 0 is ignored!
[... cut ...] . (wait_for_transform=0.200000)
[rtabmap-2] [ERROR] (2023-10-19 17:40:00.612) MsgConversion.cpp:1941::convertRGBDMsgs() TF of received image 0 at time 1697708375.286428s is not set!
[rtabmap-2] rtabmap: Saving database/long-term memory... (located at /home/sun/.ros/rtabmap.db)
[rtabmap-2] rtabmap: 2D occupancy grid map saved.
[rtabmap-2] rtabmap: Saving database/long-term memory...done! (located at /home/sun/.ros/rtabmap.db, 26 MB)
[INFO] [rtabmap-2]: process has finished cleanly [pid 112554]

`

sun-rabbit, 2023-10-20:
> I never played with the astra on ros2, but the input data you are feeding rtabmap with looks pretty bad:
> 
> ```
> The time difference between rgb and depth frames is high (diff=0.153943s, rgb=1697708399.896419s, depth=1697708399.742476s)
> ```
> 
> What is the delay and frame rate of the input topics?
> 
> ```
> ros2 topic hz /camera/color/image_raw 
> ros2 topic hz /camera/depth/image_raw 
> ros2 topic hz /camera/color/camera_info
> ros2 topic delay /camera/color/image_raw 
> ros2 topic delay /camera/depth/image_raw 
> ros2 topic delay /camera/color/camera_info
> ```
> 
> If you don't have a solid 30 Hz on all topics, please report the issue on the camera driver repo. I cannot help more.
> 
> Not also testing VSLAM in front of a white wall is a bad idea.
Oh my goodness, I've realized that my rates are all around 4 or even lower！！！
`sun@root:~/ros2_ws$ ros2 topic hz /camera/color/image_raw 
average rate: 4.343
	min: 0.007s max: 0.495s std dev: 0.16495s window: 6
average rate: 4.901
	min: 0.004s max: 0.495s std dev: 0.14304s window: 12
average rate: 4.621
	min: 0.004s max: 0.495s std dev: 0.14736s window: 17
average rate: 4.768
	min: 0.004s max: 0.495s std dev: 0.13807s window: 23
average rate: 4.370
	min: 0.004s max: 0.559s std dev: 0.14701s window: 26
average rate: 4.580
	min: 0.004s max: 0.559s std dev: 0.14407s window: 32

sun@root:~/ros2_ws$ ros2 topic hz /camera/depth/image_raw 
average rate: 3.166
	min: 0.072s max: 0.741s std dev: 0.21469s window: 6
average rate: 4.134
	min: 0.017s max: 0.741s std dev: 0.20360s window: 12
average rate: 4.323
	min: 0.017s max: 0.741s std dev: 0.18391s window: 18
average rate: 4.546
	min: 0.017s max: 0.741s std dev: 0.17673s window: 25
average rate: 4.327
	min: 0.017s max: 0.741s std dev: 0.17370s window: 29

sun@root:~/ros2_ws$ ros2 topic hz /camera/color/camera_info
WARNING: topic [/camera/color/camera_info] does not appear to be published yet
average rate: 3.768
	min: 0.082s max: 0.367s std dev: 0.09341s window: 6
average rate: 3.784
	min: 0.030s max: 0.517s std dev: 0.14686s window: 10
average rate: 3.846
	min: 0.030s max: 0.517s std dev: 0.15045s window: 15
average rate: 4.185
	min: 0.000s max: 0.537s std dev: 0.16664s window: 21

sun@root:~/ros2_ws$ ros2 topic delay /camera/depth/image_raw 
average delay: 0.231
	min: 0.223s max: 0.237s std dev: 0.00602s window: 3
average delay: 0.237
	min: 0.216s max: 0.261s std dev: 0.01325s window: 8
average delay: 0.227
	min: 0.199s max: 0.261s std dev: 0.01641s window: 13
average delay: 0.228
	min: 0.199s max: 0.261s std dev: 0.01594s window: 17
average delay: 0.227
	min: 0.199s max: 0.277s std dev: 0.01988s window: 21
average delay: 0.229
	min: 0.199s max: 0.277s std dev: 0.02095s window: 25

sun@root:~/ros2_ws$ ros2 topic delay /camera/color/camera_info
WARNING: topic [/camera/color/camera_info] does not appear to be published yet
average delay: 0.049
	min: 0.039s max: 0.057s std dev: 0.00759s window: 3
average delay: 0.043
	min: 0.027s max: 0.062s std dev: 0.01260s window: 7
average delay: 0.046
	min: 0.027s max: 0.062s std dev: 0.01078s window: 11
average delay: 0.041
	min: 0.024s max: 0.062s std dev: 0.01120s window: 17
average delay: 0.040
	min: 0.024s max: 0.062s std dev: 0.01104s window: 21
average delay: 0.038
	min: 0.024s max: 0.062s std dev: 0.01050s window: 27

`
 I'll leave and seek help in the camera's issue . Thank you very much!

sun-rabbit, 2023-10-20:
> I never played with the astra on ros2, but the input data you are feeding rtabmap with looks pretty bad:
>
> ```
> The time difference between rgb and depth frames is high (diff=0.153943s, [...]

Oh my goodness, I've realized that my rates are all around 4 or even lower!
 ```
sun@root:~/ros2_ws$ ros2 topic hz /camera/color/image_raw 
average rate: 4.343
	min: 0.007s max: 0.495s std dev: 0.16495s window: 6
average rate: 4.901
	min: 0.004s max: 0.495s std dev: 0.14304s window: 12
average rate: 4.621
	min: 0.004s max: 0.495s std dev: 0.14736s window: 17
average rate: 4.768
	min: 0.004s max: 0.495s std dev: 0.13807s window: 23
average rate: 4.370
	min: 0.004s max: 0.559s std dev: 0.14701s window: 26
average rate: 4.580
	min: 0.004s max: 0.559s std dev: 0.14407s window: 32

sun@root:~/ros2_ws$ ros2 topic hz /camera/depth/image_raw 
average rate: 3.166
	min: 0.072s max: 0.741s std dev: 0.21469s window: 6
average rate: 4.134
	min: 0.017s max: 0.741s std dev: 0.20360s window: 12
average rate: 4.323
	min: 0.017s max: 0.741s std dev: 0.18391s window: 18
average rate: 4.546
	min: 0.017s max: 0.741s std dev: 0.17673s window: 25
average rate: 4.327
	min: 0.017s max: 0.741s std dev: 0.17370s window: 29

sun@root:~/ros2_ws$ ros2 topic hz /camera/color/camera_info
WARNING: topic [/camera/color/camera_info] does not appear to be published yet
average rate: 3.768
	min: 0.082s max: 0.367s std dev: 0.09341s window: 6
average rate: 3.784
	min: 0.030s max: 0.517s std dev: 0.14686s window: 10
average rate: 3.846
	min: 0.030s max: 0.517s std dev: 0.15045s window: 15
average rate: 4.185
	min: 0.000s max: 0.537s std dev: 0.16664s window: 21

sun@root:~/ros2_ws$ ros2 topic delay /camera/depth/image_raw 
average delay: 0.231
	min: 0.223s max: 0.237s std dev: 0.00602s window: 3
average delay: 0.237
	min: 0.216s max: 0.261s std dev: 0.01325s window: 8
average delay: 0.227
	min: 0.199s max: 0.261s std dev: 0.01641s window: 13
average delay: 0.228
	min: 0.199s max: 0.261s std dev: 0.01594s window: 17
average delay: 0.227
	min: 0.199s max: 0.277s std dev: 0.01988s window: 21
average delay: 0.229
	min: 0.199s max: 0.277s std dev: 0.02095s window: 25

sun@root:~/ros2_ws$ ros2 topic delay /camera/color/camera_info
WARNING: topic [/camera/color/camera_info] does not appear to be published yet
average delay: 0.049
	min: 0.039s max: 0.057s std dev: 0.00759s window: 3
average delay: 0.043
	min: 0.027s max: 0.062s std dev: 0.01260s window: 7
average delay: 0.046
	min: 0.027s max: 0.062s std dev: 0.01078s window: 11
average delay: 0.041
	min: 0.024s max: 0.062s std dev: 0.01120s window: 17
average delay: 0.040
	min: 0.024s max: 0.062s std dev: 0.01104s window: 21
average delay: 0.038
	min: 0.024s max: 0.062s std dev: 0.01050s window: 27
 ```

 I'll leave and seek help in the camera's issue tracker. Thank you very much!

wakoko79, 2024-05-22:
@matlabbe  I also am experiencing this warning in ROS1 (rtabmap 0.21.4). **And I traced it**.

In my case of using approximate sync, I found out that the synchronizer (`message_filters::Synchronizer`) _callback_ seems to be the issue.  
  

### Here is what happens:
`CommonDataSubscriber` class is the one handling all sensor topic subscription.
It setups the callbacks when messages are received from topics in `CommonDataSubscriber::setupCallbacks()`.
This uses `message_filters::Synchronizer()` with either exact or approximate time policies **to get a set of messages from all relevant topics**.
Also, `setupCallbacks()` will setup callbacks when an appropriate set of sensor messages on all topics is found (BTW this is setup using macros, i took some headache medications XD). In my case, the callback to be called was `CommonDataSubscriber::depthScan3dCallback()`.

The thing is that this callback, through some nested calls, calls `CoreWrapper::process()`, which I **guess** is the main processing body for loop closure. So this will take a considerable amount of time.

So what _I think_ happens is that `Synchronizer` calls `depthScan3dCallback()` since there is a set of sensor messages that arrived.
`Synchronizer` gets a new set of sensor messages, and `depthScan3dCallback()` still **does not** return, so `Synchronizer` can't call it again. 

Essentially `depthScan3dCallback()` blocks the `Synchronizer` because `depthScan3dCallback()` calls `CoreWrapper::process()` which takes a lot of time. 
This will happen again and again, and the time delay gets compounded (mine reaches +20 sec delay).  
\
\
\
As for the `Warn: Did not receive data since 5 seconds!` warning, this is implemented inside `SyncDiagnostic` by `diagnosticTimerCallback()`. It checks if the last time `SyncDiagnostic::tick()` is called is 5 seconds old from the current time.

`SyncDiagnostic::tick()` is called inside `CoreWrapper::process()`, so it is affected by the time delay caused by  `depthScan3dCallback()` blocking the `Synchronizer` (mentioned above).
\
\
Here is a sample output for consecutive calls of `depthScan3dCallback()` with my debug printouts:
Notes: 
(1) Values inside [ ] `(e.g. [1689820113.556744] ) is from ros::Time::now()`
(2) Time with no bracket is the stamp passed to the function (e.g. the last value in: `***** TICK inside CoreWrapper::process() ********* [1689820116.201607] 1689820100.194628 `)
(3) I removed the log throttling for `Warn: Did not receive data since 5 seconds!` inside `SyncDiagnostic::diagnosticTimerCallback()`

```shell

[ WARN] - /rtabmap/rtabmap: [01:28:24.914341] *******
**** SYNC CALLBACK, timenow [1689820113.546681]
image = 1689820100.205681
depth = 1689820100.205681
caminf= 1689820100.205681
ptcld = 1689820100.194628
#######

[ WARN] - /rtabmap/rtabmap: [01:28:24.916490] ***** CALLING CoreWrapper::process() ********* [1689820113.556744] 1689820100.194628
[ WARN] - /rtabmap/rtabmap: [01:28:25.282195] *** SyncDiagnostic::diagnosticTimerCallback [1689820113.923028 - 1689820100.093715 = 13.829312]
/rtabmap/rtabmap: [COMMON_DATA] Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /realsense/color/image_raw \
   /realsense/aligned_depth_to_color/image_raw \
   /realsense/color/camera_info \
   /velodyne_points

[ WARN] - /rtabmap/rtabmap: [01:28:26.273966] *** SyncDiagnostic::diagnosticTimerCallback [1689820114.915351 - 1689820100.093715 = 14.821635]
/rtabmap/rtabmap: [COMMON_DATA] Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /realsense/color/image_raw \
   /realsense/aligned_depth_to_color/image_raw \
   /realsense/color/camera_info \
   /velodyne_points

[ WARN] - /rtabmap/rtabmap: [01:28:27.276261] *** SyncDiagnostic::diagnosticTimerCallback [1689820115.917313 - 1689820100.093715 = 15.823598]
/rtabmap/rtabmap: [COMMON_DATA] Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /realsense/color/image_raw \
   /realsense/aligned_depth_to_color/image_raw \
   /realsense/color/camera_info \
   /velodyne_points

[ WARN] - /rtabmap/rtabmap: [01:28:27.561429] ***** TICK inside CoreWrapper::process() ********* [1689820116.201607] 1689820100.194628
[ WARN] - /rtabmap/rtabmap: [01:28:27.561517] ***** CommonDataSubscriber::tick() ********* [1689820116.201607] 1689820100.194628
[ WARN] - /rtabmap/rtabmap: [01:28:27.561658] ***** SyncDiagnostic::tick() ********* [1689820116.201607] 1689820100.194628 (1689820100.194628)
[... cut ...]
[ INFO] - /rtabmap/rtabmap: [01:28:30.435551] rtabmap (46): Rate=0.10s, Limit=0.000s, Conversion=0.0015s, RTAB-Map=2.4862s, Maps update=0.3787s pub=0.0065s (local map=19, WM=19)
[ WARN] - /rtabmap/rtabmap: [01:28:30.435977] *******
**** SYNC CALLBACK, timenow [1689820119.068099]
image = 1689820100.405816
depth = 1689820100.405816
caminf= 1689820100.405816
ptcld = 1689820100.396348
#######
``` 

So @matlabbe , any solutions for this? Maybe make `depthScan3dCallback()` (or other callbacks for the synchronizer) to just store the data and not call `CoreWrapper::process()` directly?

wakoko79, 2024-05-22:
I also tried to hard code the queue size of the synchronizer policy. I tried `QUEUE_SIZE = 1` but it doesn't work, it is not calling the callback `depthScan3dCallback()` at all.

So I tried `QUEUE_SIZE = 2` (this is in rtabmap_sync/include/rtabmap_sync/CommonDataSubscriberDefines.h line #147):
  
```c++
#define SYNC_DECL4(CLASS, PREFIX, APPROX, QUEUE_SIZE, SUB0, SUB1, SUB2, SUB3) \
		if(APPROX) \
		{ \
			PREFIX##ApproximateSync_ = new message_filters::Synchronizer<PREFIX##ApproximateSyncPolicy>( \
					PREFIX##ApproximateSyncPolicy(2), SUB0, SUB1, SUB2, SUB3); \
			PREFIX##ApproximateSync_->registerCallback(boost::bind(&CLASS::PREFIX##Callback, this, boost::placeholders::_1, boost::placeholders::_2, boost::placeholders::_3, boost::placeholders::_4)); \
		} \
		else \
		{ \
			PREFIX##ExactSync_ = new message_filters::Synchronizer<PREFIX##ExactSyncPolicy>( \
					PREFIX##ExactSyncPolicy(2), SUB0, SUB1, SUB2, SUB3); \
			PREFIX##ExactSync_->registerCallback(boost::bind(&CLASS::PREFIX##Callback, this, boost::placeholders::_1, boost::placeholders::_2, boost::placeholders::_3, boost::placeholders::_4)); \
		} \
		subscribedTopicsMsg_ = uFormat("\n%s subscribed to (%s sync):\n   %s \\\n   %s \\\n   %s \\\n   %s", \
				name_.c_str(), \
				APPROX?"approx":"exact", \
				SUB0.getTopic().c_str(), \
				SUB1.getTopic().c_str(), \
				SUB2.getTopic().c_str(), \
				SUB3.getTopic().c_str());
``` 

When the queue size = 2, it works fine-ish, some are still late:

```shell
[ WARN] - /rtabmap/rtabmap: [06:48:34.843377] *******
**** SYNC CALLBACK, timenow [1689820002.448316]
image = 1689819995.429335
depth = 1689819995.429335
caminf= 1689819995.429335
ptcld = 1689819995.501687
#######

[ WARN] - /rtabmap/rtabmap: [06:48:34.845505] ***** CALLING CoreWrapper::process()  ********* [1689820002.448316] 1689819995.501687
[ WARN] - /rtabmap/rtabmap: [06:48:35.403065] *** SyncDiagnostic::diagnosticTimerCallback [1689820003.011595 - 1689819995.400859 = 7.610736]
/rtabmap/rtabmap: [COMMON_DATA] Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /realsense/color/image_raw \
   /realsense/aligned_depth_to_color/image_raw \
   /realsense/color/camera_info \
   /velodyne_points

[ WARN] - /rtabmap/rtabmap: [06:48:36.412948] *** SyncDiagnostic::diagnosticTimerCallback [1689820004.021142 - 1689819995.400859 = 8.620283]
/rtabmap/rtabmap: [COMMON_DATA] Did not receive data since 5 seconds! Make sure the input topics are published ("$ rostopic hz my_topic") and the timestamps in their header are set. If topics are coming from different computers, make sure the clocks of the computers are synchronized ("ntpdate"). If topics are not published at the same rate, you could increase "queue_size" parameter (current=200).
/rtabmap/rtabmap subscribed to (approx sync):
   /realsense/color/image_raw \
   /realsense/aligned_depth_to_color/image_raw \
   /realsense/color/camera_info \
   /velodyne_points

[ WARN] - /rtabmap/rtabmap: [06:48:36.869964] ***** TICK inside CoreWrapper::process() ********* [1689820004.475136] 1689819995.501687
[ WARN] - /rtabmap/rtabmap: [06:48:36.870035] ***** CommonDataSubscriber::tick() ********* [1689820004.475136] 1689819995.501687
[ WARN] - /rtabmap/rtabmap: [06:48:36.870119] ***** SyncDiagnostic::tick() ********* [1689820004.475136] 1689819995.501687 (1689819995.501687)
[ INFO] - /rtabmap/rtabmap: [06:48:36.870156] rtabmap (4): Rate=0.10s, Limit=0.000s, Conversion=0.0021s, RTAB-Map=1.7466s, Maps update=0.2711s pub=0.0066s (local map=1, WM=1)

[ WARN] - /rtabmap/rtabmap: [06:48:37.023094] *******
**** SYNC CALLBACK, timenow [1689820004.626130]
image = 1689820004.369122
depth = 1689820004.369122
caminf= 1689820004.369122
ptcld = 1689820004.377366
#######
``` 

Note the time on `**** SYNC CALLBACK, timenow [1689820004.626130]` and stamp time for ticks `ptcld = 1689820004.377366`. They are near now. The first message on top still lags about 7 seconds though.

matlabbe, 2024-05-26:
You should not have to hard-code the queue size, there is a parameter called `queue_size` that can be used for that.

What is the frame rate of all topics? If `ptcld` is slower, `queue_size` should be large enough so that `message_filters` can buffer enough images to synchronize.

Another comment, when synchronizing with a lidar, I recommend to synchronize the image topics together with [rgbd_sync](http://wiki.ros.org/rtabmap_sync#rtabmap_sync.2Frgbd_sync) node before rtabmap node, then set `subscribe_rgbd:=true` for rtabmap node and remap to correct `rgbd_image` topic generated by `rgbd_sync`.

wakoko79, 2024-05-27:
> What is the frame rate of all topics? If `ptcld` is slower, `queue_size` should be large enough so that `message_filters` can buffer enough images to synchronize.  

Here:
```
                   topic                       rate   min_delta   max_delta   std_dev    window
===============================================================================================
/realsense/color/image_raw                    30.0    0.01006     0.06035     0.0080                   topic                       rate   min_delta   max_delta   std_dev    window
===============================================================================================
/realsense/color/image_raw                    30.0    0.01006     0.06035     0.00801    330   
/realsense/aligned_depth_to_color/image_raw   29.98   0.01006     0.07039     0.007512   330   
/realsense/color/camera_info                  29.98   0.01005     0.06034     0.00752    331   
/velodyne_points                              9.913   0.09051     0.111       0.002297   331 1    330   
/realsense/aligned_depth_to_color/image_raw   29.98   0.01006     0.07039     0.007512   330   
/realsense/color/camera_info                  29.98   0.01005     0.06034     0.00752    331   
/velodyne_points                              9.913   0.09051     0.111       0.002297   331 
```  
<br />
<br />
<br />

> You should not have to hard-code the queue size, there is a parameter called `queue_size` that can be used for that.

@matlabbe The `queue_size` parameter affects the individual input topic size **as well as the synchronizer's queue size**.  
The thing is, if you want message_filters' Synchronizer to call your callback with the latest set of data, you want the input topics to be high enough, but Synchronizer's OWN queue size (from initializing message_filters::sync_policies::AppoximateTimeSyncPolicy or ExactTimeSyncPolicy) **should be low** ([ref](https://answers.ros.org/question/259944/ros-message_filter-exacttime-queue/)).  
  
So if you want the synchronizer's to call your callback with the latest synchronized set, the queue size should be 1.

**BUT** there is an issue with giving it `queue_size=1`, its devs know it:
https://github.com/ros/ros_comm/blob/845f74602c7464e08ef5ac6fd9e26c97d0fe42c9/utilities/message_filters/include/message_filters/sync_policies/approximate_time.h#L125

So for the synchronizer `queue_size=1` does not work (mostly, the callback isn't called). `queue_size=2` does work better than having a bigger `queue_size`, like '`queue_size=100`, but it still gets delayed for some time, then recovers back to current time.

<br />

I made a simple test node to verify this behaviour. The callback registered to the synchronizer just sleeps for 1 second to simulate large processing time. Then we can compare the time the callback gets called, and the sensor data time stamps it  contains:
<details><summary>C++ code</summary>
<p>

```c++
#include "ros/ros.h"
#include <message_filters/subscriber.h>
#include <message_filters/synchronizer.h>
#include <message_filters/sync_policies/approximate_time.h>
#include <sensor_msgs/Image.h>
#include <sensor_msgs/PointCloud2.h>
#include <sensor_msgs/CameraInfo.h>

// using namespace sensor_msgs;
using namespace message_filters;

void callback(  const sensor_msgs::ImageConstPtr& rgbMsg, 
                const sensor_msgs::CameraInfoConstPtr& rgbinfoMsg, 
                const sensor_msgs::ImageConstPtr& depthMsg, 
                const sensor_msgs::PointCloud2ConstPtr& pclMsg)
{
  ROS_WARN("\n*********\nnow   =[%f]\nrgb   = %f\nrgbinf= %f\ndepth = %f\npcl   = %f\n##################\n", 
                                                    ros::Time::now().toSec(), 
                                                    rgbMsg->header.stamp.toSec(),
                                                    rgbinfoMsg->header.stamp.toSec(),
                                                    depthMsg->header.stamp.toSec(),
                                                    pclMsg->header.stamp.toSec());
  ros::Duration(1.0).sleep();
}

int main(int argc, char** argv)
{
  ros::init(argc, argv, "sync_trial_node");

  ros::NodeHandle nh, pnh("~");
  int sync_qsize, sub_qsize;
[... 442 lines cut ...]
[ WARN] - /sync_trial_node: [02:45:24.195161] 
*********
now   =[1689820021.224373]
rgb   = 1689820011.207329
rgbinf= 1689820011.207329
depth = 1689820011.207329
pcl   = 1689820011.235857
##################

``` 

</p>
</details> 

<br />

So it is either change how rtabmap's callbacks work (maybe make it store data to cache instead of processing it) or edit the `message_filters` code to be able to make `queue_size=1` work.

<br />
<br />
<br />

> Another comment, when synchronizing with a lidar, I recommend to synchronize the image topics together with [rgbd_sync](http://wiki.ros.org/rtabmap_sync#rtabmap_sync.2Frgbd_sync) node before rtabmap [...]
  
I'll also try this later. =)

matlabbe, 2024-06-03:
Do you tell me that all this time that [example](http://wiki.ros.org/message_filters#Example_.28C.2B-.2B-.29-1) was wrong?! 

We used to do "sync_qsize=10, sub_qsize=1" by default (like the example) till it was not working anymore on ros noetic: https://github.com/introlab/rtabmap_ros/commit/ed564c3a12293d06ca28e145ff50a974f6159cc9

So now we have `queue_size` set to both topics and synchronizer. Did you also try "sync_qsize=10, sub_qsize=1"? Otherwise, it could make sense to set the sync to 2.

Note that another way to see the delay over time is to do:
```
rostopic delay /rtabmap/mapData
```

I'll try on our robots this week if I see a large difference, in particular for visual odometry if that could decrease the delay we are seeing.

For the rate of your topics, with current code, a `queue_size` of 5 should work unless one of the the topic has also a large delay.

wakoko79, 2024-06-04:
> Do you tell me that all this time that [example](http://wiki.ros.org/message_filters#Example_.28C.2B-.2B-.29-1) was wrong?!  
  
No. What I'm saying is it is not good to make a singular `queue_size` parameter to feed into the queue size of both the individual input topics _and_ to the synchronizer. That example is OK.  I read its code, and I think now that `sync_qsize=2` is kinda bad..
<br />

  
> We used to do "sync_qsize=10, sub_qsize=1" by default (like the example) till it was not working anymore on ros noetic: [ed564c3](https://github.com/introlab/rtabmap_ros/commit/ed564c3a12293d06ca28e145ff50a974f6159cc9)
>
> So now we have `queue_size` set to both topics and synchronizer. Did you also try "sync_qsize=10, sub_qsize=1"? Otherwise, it could make sense to set the sync to 2.  

Honestly, "sync_qsize=10, sub_qsize=1" might be better.  
The problem here is, as stated above, there is only 1 `queue_size` parameter. Combined with the warning:
```
If topics are not published at the same rate, you could increase "queue_size" parameter (current=20).
``` 
This prompts the user to increase the `queue_size`, which in turn, makes the delay worse instead of making it better. 

Here are plots of the time differences of different `sync_qsize` and `sub_qsize` from my example code:
[image]
[image]
[image]

These graphs shows that increasing individual topic queue size (`sub_qsize`) increases the average delay between current time and message set time returned by the synchronizer. 
Also, increasing the synchronizer's queue size (`sync_qsize`) magnifies that delay even more.  
<br />
But this **does not** mean to make "sync_qsize=2, sub_qsize=1".
I've read the code for message_filters and apparently, the `queue_size` for `ApproximateTime` policy should be somehow big enough. If it is too low, the pool of messages its algorithm is using to determine the best set will degrade.  I think (**_I'm not sure_**, I did not read the whole thing >_<) the whole queuing pipeline of `message_filters` runs on 1 thread,

So the way `message_filters::ApproximateTime` is that _each input topic_ has a queue (2 queues actually), with max length as dictated by the `queue_size`.
Whenever any topic overflows, **all the queues for all inputs gets purged and selection for synchronized set gets reset**.  
  
Say you have 2 topics `topicA` (10hz) and `topicB` (1hz), and a queue_size of 2. `topicA`'s queue will fill up in 0.2sec, while `topicB` will still be empty at 0.3sec. So the synchronizer will not output anything, since it will get reset indefinitely.
So there is definitely a minimum value for the **synchronizer's** `queue_size` which depends on the frequency of your input topics.  
<br />
<br />
There is also the issue of the original warning :
`Did not receive data since 5 seconds! Make sure the input topics are published...`

I mentioned this in my previous comment:

> As for the Warn: Did not receive data since 5 seconds! warning, this is implemented inside SyncDiagnostic by diagnosticTimerCallback(). It checks if the last time SyncDiagnostic::tick() is called is 5 seconds old from the current time.
>
>SyncDiagnostic::tick() is eventually called inside CoreWrapper::process(), so it is affected by the time delay caused by depthScan3dCallback() blocking the Synchronizer (mentioned above).  

I still think that the tick should be called directly inside the callback that is called by the synchronizer, just before calling `commonSingleCameraCallback()` (or other similar callbacks), since this tick **_should_** signify the receipt of sensor data. I said "should" here since  this tick directly affects the warning issued by `SyncDiagnostic`, which says `Did not receive data since 5 seconds!`, which _implies_ that the input topics did not send data, which is not true most of the time.
<br />
<br />
I also tried to make `commonSingleCameraCallback()` (the function called by `depthScan3dCallback()`) spawn in another thread using `boost::thread`. But `boost::thread` and `boost::bind` can only accommodate up to a maximum of 9 arguments, `commonSingleCameraCallback()` has 13, so it didn't work. I'm not using `std::thread` because I don't know what will happen if I mix it with boost.  

So I made a helper function to make things work. It did work, but I'm not sure if this is OK or if the current implementation is intended. 

Anyway, it achieved the goal of making the callback called by the synchronizer return fast by not waiting on the processing by rtabmap. It works still even on the current implementation of `queue_size = sync_qsize = sub_qsize` (in my launch file `queue_size=200`).  
  
Here is a plot for the delays, which is run on the actual rtabmap_ros code:
[image]

**Processed** just means that I passed on the data to be processed as normal, **unprocessed** means that the callback was called, but did not process the syncronized messages set since `commonSingleCameraCallback()` is still running on another thread processing the previous data it received.
<br />
The delay is very minimal and stable as we can see.  

Though sometimes, rtabmap gets immediately killed. I'm not very familiar with threading.. Maybe bacause I didn't use any mutex? Or maybe because I built it on Debug? I really don't know why it sometimes kills rtabmap.

matlabbe, 2024-06-10:
We could definitely add a second queue size parameter, maybe called `sync_queue_size` for the synchronizer and keep `queue_size` for the individual subscribers.

matlabbe, 2024-06-12:
With the commit above, we can now set a different queue size for input topics (called `topic_queue_size`) and another one for message_filters synchronizer (called `sync_queue_size`). To keep backward compatibility, `queue_size` has been kept (deprecated), but its value is copied into `sync_queue_size` to have same effect than before.

That change will be ported to ros2 with next master->ros2 merge.

cheers,
Mathieu

wakoko79, 2024-09-17:
@matlabbe BTW, I made a threaded version of the callbacks to keep the time spent in the Synchronizer callback minimal.
This will make rtabmap_ros ignore newer input data coming from the Synchronizer if it is still processing a previous input data. This completely eliminates the issues in this thread, as this effectively makes rtabmap process only the most recent input data.

[threaded_sync](https://github.com/wakoko79/rtabmap_ros/tree/threaded_sync)

I'm not sure though if this behaviour is OK though. 
And its incomplete now since I only implemented it for rtabmap_sync/src/impl/CommonDataSubscriberDepth.cpp right now, cuz it is what i need. 
It is pretty easy to implement to others as you only needed to add "_THREADED" to SYNC_DECLx() calls though.

I can request a pull if you like.=)

matlabbe, 2024-09-17:
I [added](https://github.com/introlab/rtabmap_ros/pull/1206) a thread for the odometry nodes, which seems to have fixed the issue there. I tried to do something similar for rtabmap node (a different implementation than yours but the idea was the same), but I aborted after realizing that making it thread-safe would be a lot of work. In your case, it works as long as you don't call any rtabmap services. For ros1, I am not sure what is the simplest thread-safe solution to do...

I think a ROS2 solution would be to have a [MultiThreadedExecutor](https://docs.ros.org/en/foxy/Concepts/About-Executors.html) for rtabmap node, then set message filter sync callbacks to one [callback group](https://docs.ros.org/en/foxy/How-To-Guides/Using-callback-groups.html), then set all services in another callback group. Then the thread you launch from a sync callback would need to be launched in the same callback group than the one used for the services. That way, we make sure we cannot execute a service callback while a rtabmap update is running.

To integrate the rtabmap update loop with the executor (instead of a disjoint thread), we could use a Timer (with same callback group than the services) that triggers a callback checking (every 100 ms?) if new data is received, then proceeds with the update. The "data" would be protected by a mutex, between the message filter callbacks and the Timer callback. If the timer callback is not done with current update and message filter receives new data, it will just overwrite the buffered one or discard current one till it knows the timer is not using the data buffer anymore. I'll try create a PR with this approach soon and compare with your approach.

AliH17, 2025-04-04:
@matlabbe 
Hey I'm getting this error after running
ros2 launch rtabmap_launch rtabmap.launch.py   rtabmap_args:="--delete_db_on_start"   rgb_topic:=/camera/color/image_raw   depth_topic:=/camera/depth/image_rect_raw   camera_info_topic:=/camera/color/camera_info   frame_id:=camera_link   use_sim_time:=true   approx_sync:=true   qos:=2   rviz:=true   queue_size:=30

[image]

[image]
first I ran 
ros2 launch realsense2_camera rs_launch.py
and these are the topics being published
agxxavier@ubuntu:~$ ros2 topic list
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_depth
/camera/imu
/parameter_events
/rosout
/tf_static

I did this last night and I had the following topics
/camera/color/camera_info
/camera/color/image_raw
/camera/color/metadata
/camera/depth/camera_info
/camera/depth/image_rect_raw
/camera/depth/metadata
/camera/extrinsics/depth_to_color
/camera/extrinsics/depth_to_depth
/camera/imu
/clicked_point
/cloud_ground
/cloud_map
/cloud_obstacles
/goal
/goal_pose
/grid_prob_map
/grid_prob_map_updates
/initialpose
/joint_states
/local_path
/map
/mapPath
/map_updates
/octomap_ground
/parameter_events
/robot_description
/rosout
/tf
/tf_static

matlabbe, 2025-04-06:
@AliH17 Are there warnings like "Warn: Did not receive data since 5 seconds!" Note that to make sure topics are published, do "ros2 topic hz" instead of just "ros2 topic list". Are the frame rates stable and the same for each topic synchronized?

AliH17, 2025-04-07:
> [@AliH17](https://github.com/AliH17) Are there warnings like "Warn: Did not receive data since 5 seconds!" Note that to make sure topics are published, do "ros2 topic hz" instead of just "ros2 topic list". Are the frame rates stable and the same for each topic synchronized?

@matlabbe 
No, rtabmap is receiving the data, the frame rates go out of sync sometimes and mention a time difference. What I managed to fix was by changing the fixed frame to odom, it begins mapping and tf errors are gone. The problem is after mapping about 180- 240 degrees around the room, it just gets stuck and doesn't map anymore. And the logs just mention the odometry, nothing strange.
Also I do get a lot of [rviz2-4] [INFO] [1697708400.569555316] [rviz]: Message Filter dropping message: frame 'camera_color_optical_frame' at time 1697708396.472 for reason 'Unknown' .

matlabbe, 2025-04-08:
"changing the fixed frame to odom" how did you do it? is it `frame_id` or `odom_frame_id` did you change? Do you have another odometry source? If you can share a rosbag like this:
```
rosbag record /tf /tf_static /camera/color/image_raw/compressed /camera/depth/image_rect_raw /camera/color/camera_info
```
it would be helpful.

AliH17, 2025-04-26:
@matlabbe Sorry for the late reply, but I changed it on rViz.

matlabbe, 2025-04-27:
Well, if you changed `map` to `odom` frame in the global RVIZ options and you were able to see something, it means the `map`->`odom` frame was not published (rtabmap not running?) or a some tf timestamps issue. If you reproduce it, record and share the tf tree:
```
ros2 run tf2_tools view_frames
```