introlab/rtabmap_ros #1056: how to use map_assembler from rtabmap util with rtabmap_launch

aravi-2001, 2023-10-25:
I've installed everything from apt-get.except rtabmap_ros. im using rtabmap from rtabmap_launch. how can i connect it to rtabmap_util's map_assembler, because it's not running, im getting error rtabmap_launch not found ?

this is the rtabmap util launch:
<node pkg="rtabmap_util" type="map_assembler" name="map_assembler">
    <remap from="mapData" to="/rtabmap/mapData"/>
   <param name="Grid/RangeMax" type="double" value="100"/>
   <param name="Grid/DepthDecimation" type="double" value="0.1"/>

</node>

this is how i launch slam: roslaunch rtabmap_launch rtabmap.launch depth_topic:=/slam/depth_image rgb_topic:=/slam/image camera_info_topic:=/slam/camera_info rtabmapviz:=true mapping:=true frame_id:=fcu visual_odometry:=false odom_topic:=/mavros/local_position/odom

matlabbe, 2023-10-25:
Which version are you using?

Tried on my computer:
```
roslaunch rtabmap_launch rtabmap.launch depth_topic:=/slam/depth_image rgb_topic:=/slam/image camera_info_topic:=/slam/camera_info rtabmapviz:=true mapping:=true frame_id:=fcu visual_odometry:=false odom_topic:=/mavros/local_position/odom
... logging to /home/mathieu/.ros/log/236d8d9e-7380-11ee-95bf-d93a9a424f56/roslaunch-RBLT-MLABBE-01-784211.log
Checking log directory for disk usage. This may take a while.
Press Ctrl-C to interrupt
WARNING: disk usage in log directory [/home/mathieu/.ros/log] is over 1GB.
It's recommended that you use the 'rosclean' command.

started roslaunch server http://RBLT-MLABBE-01:41361/

SUMMARY
========

CLEAR PARAMETERS
 * /rtabmap/rtabmap/
 * /rtabmap/rtabmap_viz/

PARAMETERS
 * /rosdistro: noetic
 * /rosversion: 1.15.15
 * /rtabmap/rtabmap/Mem/IncrementalMemory: true
 * /rtabmap/rtabmap/Mem/InitWMWithAllNodes: false
 * /rtabmap/rtabmap/approx_sync: True
 * /rtabmap/rtabmap/config_path: 
 * /rtabmap/rtabmap/database_path: ~/.ros/rtabmap.db
 * /rtabmap/rtabmap/frame_id: fcu
 * /rtabmap/rtabmap/gen_depth: False
 * /rtabmap/rtabmap/gen_depth_decimation: 1
 * /rtabmap/rtabmap/gen_depth_fill_holes_error: 0.1
 * /rtabmap/rtabmap/gen_depth_fill_holes_size: 0
 * /rtabmap/rtabmap/gen_depth_fill_iterations: 1
 * /rtabmap/rtabmap/gen_scan: False
 * /rtabmap/rtabmap/ground_truth_base_frame_id: 
 * /rtabmap/rtabmap/ground_truth_frame_id: 
 * /rtabmap/rtabmap/initial_pose: 
 * /rtabmap/rtabmap/landmark_angular_variance: 9999.0
 * /rtabmap/rtabmap/landmark_linear_variance: 0.0001
 * /rtabmap/rtabmap/map_frame_id: map
 * /rtabmap/rtabmap/odom_frame_id: 
 * /rtabmap/rtabmap/odom_frame_id_init: 
 * /rtabmap/rtabmap/odom_sensor_sync: False
 * /rtabmap/rtabmap/odom_tf_angular_variance: 0.001
 * /rtabmap/rtabmap/odom_tf_linear_variance: 0.001
 * /rtabmap/rtabmap/publish_tf: True
 * /rtabmap/rtabmap/queue_size: 10
 * /rtabmap/rtabmap/scan_cloud_max_points: 0
 * /rtabmap/rtabmap/subscribe_depth: True
 * /rtabmap/rtabmap/subscribe_rgb: True
 * /rtabmap/rtabmap/subscribe_rgbd: False
 * /rtabmap/rtabmap/subscribe_scan: False
 * /rtabmap/rtabmap/subscribe_scan_cloud: False
 * /rtabmap/rtabmap/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap/subscribe_stereo: False
 * /rtabmap/rtabmap/subscribe_user_data: False
 * /rtabmap/rtabmap/wait_for_transform_duration: 0.2
 * /rtabmap/rtabmap_viz/approx_sync: True
 * /rtabmap/rtabmap_viz/frame_id: fcu
 * /rtabmap/rtabmap_viz/odom_frame_id: 
 * /rtabmap/rtabmap_viz/queue_size: 10
 * /rtabmap/rtabmap_viz/subscribe_depth: True
 * /rtabmap/rtabmap_viz/subscribe_rgb: True
 * /rtabmap/rtabmap_viz/subscribe_rgbd: False
 * /rtabmap/rtabmap_viz/subscribe_scan: False
 * /rtabmap/rtabmap_viz/subscribe_scan_cloud: False
 * /rtabmap/rtabmap_viz/subscribe_scan_descriptor: False
 * /rtabmap/rtabmap_viz/subscribe_stereo: False
 * /rtabmap/rtabmap_viz/wait_for_transform_duration: 0.2
[... 93 lines cut ...]
   /slam/depth_image \
   /slam/camera_info
[ INFO] [1698270470.336224393]: rtabmap_viz started.
```

and `map_assembler`:
```
rosrun rtabmap_util map_assembler mapData:=/rtabmap/mapData
[ INFO] [1698270546.421023560]: /map_assembler: regenerate_local_grids          = false
[ INFO] [1698270546.422581327]: /map_assembler(maps): map_filter_radius          = 0.000000
[ INFO] [1698270546.422596489]: /map_assembler(maps): map_filter_angle           = 30.000000
[ INFO] [1698270546.422603216]: /map_assembler(maps): map_cleanup                = true
[ INFO] [1698270546.422610159]: /map_assembler(maps): map_always_update          = false
[ INFO] [1698270546.422620116]: /map_assembler(maps): map_empty_ray_tracing      = true
[ INFO] [1698270546.422630714]: /map_assembler(maps): cloud_output_voxelized     = true
[ INFO] [1698270546.422637411]: /map_assembler(maps): cloud_subtract_filtering   = false
[ INFO] [1698270546.422645151]: /map_assembler(maps): cloud_subtract_filtering_min_neighbors = 2
[ INFO] [1698270546.422767979]: /map_assembler(maps): octomap_tree_depth         = 16
[ INFO] [1698270546.428067750]: Rtabmap namespace is "rtabmap", deduced from topic "/rtabmap/mapData"
[ INFO] [1698270546.483586279]: Called "rtabmap/get_map_data" service, initializing cache...
[ INFO] [1698270546.484419874]: map_assembler: Updating = 0.000592s, Publishing data = 0.000138s (subscribers=false)
[ INFO] [1698270546.484487547]: Called "rtabmap/get_map_data" service, initializing cache... done! The map will be assembled on next subscriber connection.
```

Both looks fine.

aravi-2001, 2023-10-31:
Hello, how can i pass the "Grid/RangeMax" and "Grid/DepthDecimation" parameters as shown above, since i keep getting this error

rosrun rtabmap_util map_assembler mapData:=/rtabmap/mapData Grid/RangeMax:="100"
terminate called after throwing an instance of 'ros::InvalidNameException'
  what():  Character [1] is not valid as the first character in Graph Resource Name [100].  Valid characters are a-z, A-Z, / and in some cases ~.
Aborted (core dumped)