introlab/rtabmap_ros #1058: ros2: Cannot set --delete_db_on_start with ComposableNode

matlabbe, 2023-10-29:
It seems that `extra_arguments` cannot be used like `arguments` for normal node: 
https://docs.ros.org/en/humble/How-To-Guides/Launching-composable-nodes.html

Interpause, 2025-01-25:
was a solution found for this? like would passing it to the ComposableNodeContainer work?

anath93, 2025-01-30:
@matlabbe Would creating a parameter work for this ?

matlabbe, 2025-02-04:
The commit above should support this, use `delete_db_on_start` parameter directly.

For the log level argument, a workwaround is to use log_to_rosout_level parameter instead. Here combined example:

```
$ ros2 run rtabmap_slam  rtabmap --ros-args -p delete_db_on_start:=true -p log_to_rosout_level:=0

[INFO] [1738645271.350845248] [rtabmap]: rtabmap(maps): latch                      = true
[INFO] [1738645271.350939447] [rtabmap]: rtabmap(maps): map_filter_radius          = 0.000000
[INFO] [1738645271.350960715] [rtabmap]: rtabmap(maps): map_filter_angle           = 30.000000
[INFO] [1738645271.350971809] [rtabmap]: rtabmap(maps): map_cleanup                = true
[INFO] [1738645271.350981017] [rtabmap]: rtabmap(maps): map_always_update          = false
[INFO] [1738645271.350989896] [rtabmap]: rtabmap(maps): map_empty_ray_tracing      = true
[INFO] [1738645271.350998673] [rtabmap]: rtabmap(maps): cloud_output_voxelized     = true
[INFO] [1738645271.351007409] [rtabmap]: rtabmap(maps): cloud_subtract_filtering   = false
[INFO] [1738645271.351016245] [rtabmap]: rtabmap(maps): cloud_subtract_filtering_min_neighbors = 2
[INFO] [1738645271.351045123] [rtabmap]: rtabmap(maps): octomap_tree_depth         = 16
[INFO] [1738645271.360715169] [rtabmap]: rtabmap: frame_id      = base_link
[INFO] [1738645271.360743413] [rtabmap]: rtabmap: map_frame_id  = map
[INFO] [1738645271.360753179] [rtabmap]: rtabmap: log_to_rosout_level  = 0
[INFO] [1738645271.360762479] [rtabmap]: rtabmap: initial_pose  = 
[INFO] [1738645271.360771212] [rtabmap]: rtabmap: use_action_for_goal  = false
[INFO] [1738645271.360779815] [rtabmap]: rtabmap: tf_delay      = 0.050000
[INFO] [1738645271.360790465] [rtabmap]: rtabmap: tf_tolerance  = 0.100000
[INFO] [1738645271.360802358] [rtabmap]: rtabmap: odom_sensor_sync   = false
[INFO] [1738645271.360810727] [rtabmap]: rtabmap: pub_loc_pose_only_when_localizing = false
[INFO] [1738645271.360819697] [rtabmap]: rtabmap: wait_for_transform = 0.200000
[INFO] [1738645271.360829833] [rtabmap]: rtabmap: gen_scan  = false
[INFO] [1738645271.360838159] [rtabmap]: rtabmap: gen_depth  = false
[INFO] [1738645271.376120347] [rtabmap]: RTAB-Map detection rate = 1.000000 Hz
[INFO] [1738645271.376263663] [rtabmap]: rtabmap: Deleted database "/home/mathieu/.ros/rtabmap.db" (--delete_db_on_start or -d are set).
[INFO] [1738645271.376295471] [rtabmap]: rtabmap: Using database from "/home/mathieu/.ros/rtabmap.db" (0 MB).
[DEBUG] (2025-02-03 21:01:11.376) GlobalMap.cpp:87::clear() Clearing
[DEBUG] (2025-02-03 21:01:11.376) GlobalMap.cpp:54::GlobalMap() cellSize_           =0.050000
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:55::GlobalMap() updateError_        =0.010000
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:87::clear() Clearing
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:87::clear() Clearing
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:54::GlobalMap() cellSize_           =0.050000
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:55::GlobalMap() updateError_        =0.010000
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:307::OctoMap() occupancyThr_=0.500000
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:308::OctoMap() probHit_=0.700000
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:309::OctoMap() probMiss_=0.400000
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:310::OctoMap() probClampingMin_=0.119200
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:311::OctoMap() probClampingMax_=0.971000
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:325::OctoMap() rangeMax_           =5.000000
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:326::OctoMap() rayTracing_         =false
[DEBUG] (2025-02-03 21:01:11.377) OctoMap.cpp:327::OctoMap() emptyFloodFillDepth_=0
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:87::clear() Clearing
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:54::GlobalMap() cellSize_           =0.050000
[DEBUG] (2025-02-03 21:01:11.377) GlobalMap.cpp:55::GlobalMap() updateError_        =0.010000
[DEBUG] (2025-02-03 21:01:11.377) Rtabmap.cpp:322::init() path=/home/mathieu/.ros/rtabmap.db
[ INFO] (2025-02-03 21:01:11.377) Rtabmap.cpp:327::init() Using database "/home/mathieu/.ros/rtabmap.db".
[INFO] [1738645271.377676742] [rtabmap]: [ INFO] (2025-02-03 21:01:11.377) Rtabmap.cpp:327::init() Using database "/home/mathieu/.ros/rtabmap.db".
[DEBUG] (2025-02-03 21:01:11.377) Rtabmap.cpp:4799::setWorkingDirectory() Comparing new working directory path "/home/mathieu/.ros" with ""
[DEBUG] (2025-02-03 21:01:11.377) Rtabmap.cpp:294::setupLogFiles() Log disabled!
[DEBUG] (2025-02-03 21:01:11.378) VWDictionary.cpp:470::update() incremental=1
[DEBUG] (2025-02-03 21:01:11.378) FlannIndex.cpp:52::release() 
[DEBUG] (2025-02-03 21:01:11.378) FlannIndex.cpp:80::release() 
[DEBUG] (2025-02-03 21:01:11.378) VWDictionary.cpp:680::update() Dictionary updated! (size=0 added=0 removed=0)
[DEBUG] (2025-02-03 21:01:11.378) VWDictionary.cpp:689::update() 
[DEBUG] (2025-02-03 21:01:11.378) Registration.cpp:49::create() type=0
[DEBUG] (2025-02-03 21:01:11.379) GlobalDescriptorExtractor.cpp:51::create() Creating global descriptor of type 0
[DEBUG] (2025-02-03 21:01:11.379) RegistrationIcp.cpp:162::parseParameters() libpointmatcher enabled! config=""
[DEBUG] (2025-02-03 21:01:11.379) Memory.cpp:562::parseParameters() 
[DEBUG] (2025-02-03 21:01:11.380) RegistrationIcp.cpp:162::parseParameters() libpointmatcher enabled! config=""
[... cut ...]
[INFO] [1738645271.595193288] [rtabmap]: rtabmap: qos_user_data   = 0
[INFO] [1738645271.595199930] [rtabmap]: rtabmap: approx_sync     = true
[INFO] [1738645271.595224069] [rtabmap]: Setup depth callback
[INFO] [1738645271.600661997] [rtabmap]: 
rtabmap subscribed to (approx sync):
   /odom \
   /rgb/image \
   /depth/image \
   /rgb/camera_info
[ INFO] (2025-02-03 21:01:11.610) CoreNode.cpp:90::main() rtabmap 0.21.9 started...
[INFO] [1738645271.610191712] [rtabmap]: [ INFO] (2025-02-03 21:01:11.610) CoreNode.cpp:90::main() rtabmap 0.21.9 started...
```

closed 2025-02-04 by matlabbe