introlab/rtabmap_ros #1059: Odometry conflict and Odom topic used

camasmie, 2023-11-01:
Hello,

I might be missing something basic. When I run visual odometry only, it works Ok, even though it is a bit unstable.

So I implemented wheel encoder odometry, which also works Ok with rtabmap, but also with inconsistent results.

However, by simply running both nodes together, the visual odometry gets rotated about 'z' axis (yaw) by minus 90 degrees.

[image]

The red arrow is 'wheel_odometry'. The green arrow is 'visual_odometry" from rtabmap. The robot is facing along the red arrow (up in the picture).

I am having rtabmap broadcast the transform between map and odom frames.

I am also implementing EKF with robot_localization and am still not able to make rtabmap to consume the 'odometry/filtered_twist' topic from the robot_localization package. So far I have been able to run it with:

      <param name="odom_frame_id"            type="string" value="odom"/>

But if I change the following parameter, it doesn't work:

      <param name="odom_frame_id"            type="string" value=""/>
      <arg name="odom_topic"               default="/odometry/filtered_twist"/> >

I am still puzzled why there seem to be a conflict between the two odometries, even by running them completely separate.

I am not broadcasting any transform from the wheel_odometry publisher node. I am letting rtabmab do it.

[image]

The EKF is consuming both odometries but rtabmap can only use the odom frame tf:

[image]

Having you seen this happening before?

Thank you in advance!

camasmie, 2023-11-01:
I have also noticed this message:

[ WARN] (2023-11-01 12:11:50.036) Odometry.cpp:317::process() Updated initial pose from xyz=0.000000,0.000000,0.000000 rpy=0.000000,-0.000000,0.000000 to xyz=0.000000,0.000000,0.000000 rpy=-0.002358,0.016916,-1.570856 with IMU orientation

matlabbe, 2023-11-01:
Set `publish_tf` to `false` for `rgbd_odometry` node, your `robot_localization` and `rgbd_odometry` nodes may be both publishing the same tf name.

Your IMU may have a 90 deg rotation on start, so rtabmap odom is started at 90 deg. Note that if you are using `robot_localization`, just set odometry from `rgbd_odometry` as differential/relative in `robot_localization` config, so `robot_localization` won't care about the initial rotation.

For rtabmap node, if you set:
```xml
<param name="odom_frame_id"  type="string" value=""/>
<remap from="odom"  to="/odometry/filtered_twist"/>
```
it should use the odometry topic.

camasmie, 2023-11-01:
@matlabbe 

Brilliant! This works great now. 

[image]

It also made a few rtabmap settings and behavior clearer to me.

I will double check my urdf, transforms, madgwick filter to see if I can spot why the IMU frame seems to be rotated. But for now, the 'differential' setting in robot_localization is working.

Thank you for your dedication and knowledge!