introlab/rtabmap_ros #1063: ERROR: Input cloud doesn't have "t", "time", "stamps" or "timestamp" field!

astef3, 2023-11-07:


astef3, 2023-11-08:
ROS distro: Noetic
Ubuntu 20.04

Hi, 
I tried to use the test_velodyne.launch file with Velodyne data from rosbag. But, when I launched the test_velodyne.launch file (unchanged) and played the velodyne rosbag, I get constant errors: 

**Input cloud doesn't have "t", "time", "stamps" or "timestamp" field!
Failed to deskew input cloud, aborting odometry update!**

Can you please help me with resolving this issue? Because I understand that the second error (deskew) is because of the first one, but I cannot understand how its possible that the input cloud is without timestamp, when Velodyne PointCloud2 messages  always provide time as well. 

Thank you in advance for your time and help.

matlabbe, 2023-11-11:
Can you verify that your bag has really a timestamp channel? Otherwise, you can disable deskewing in that [example](https://github.com/introlab/rtabmap_ros/blob/master/rtabmap_examples/launch/test_velodyne.launch).