introlab/rtabmap_ros #1065: Couldn't find executable named point_cloud_aggregator below /opt/ros/noetic/share/rtabmap_util

moooeeeep, 2023-11-14:
I've apt installed `ros-noetic-rtabmap-ros` from the ROS debian packages, and apparantly the [`point_cloud_aggregator`](http://wiki.ros.org/rtabmap_util#rtabmap_util.2Fpoint_cloud_aggregator) is not part of the installation.

Error message:

    [rosrun] Couldn't find executable named point_cloud_aggregator below /opt/ros/noetic/share/rtabmap_util

Tab completion:

    $ rosrun rtabmap_util 
    data_player               map_assembler             netvlad_tf_ros.py         odom_msg_to_tf            point_cloud_assembler     point_to_tf.py            rgbd_split                yaml_to_camera_info.py    
    gazebo_ground_truth.py    map_optimizer             objects_to_tags.py        patrol.py                 pointcloud_to_depthimage  rgbd_relay                transform_to_tf.py        

Is this intended?

Do I need to install from source?

matlabbe, 2023-11-14:
The node is indeed not installed, which is a bug. However, if you want to use it, you should be able to use the nodelet instead:
```
rosrun nodelet nodelet standalone rtabmap_util/point_cloud_aggregator
```

moooeeeep, 2023-11-15:
Thanks, will try that in the meantime!

commit 38010068 (2023-11-18): Fixed #1065
closed 2023-11-18 by matlabbe