introlab/rtabmap_ros #1066: Add the ability to reset map & keep map -> odom transform.

Krzo99, 2023-11-14:
I think it would be usefull to be able to reset the whole map, but keep map -> odom transform, so new map would be drawn in Rviz2 where the last one ended. 
This could also be done the same way it's done with odom. A service, you can call to set map -> odom transform (reset_odom_to_pose).

Let me know what you think.
Thanks.

Krzo99, 2023-11-20:
For multi-purpose mapping. The idea is, to reser the map, when mapping new areas, to speed up & lower the merged map data. But I like the idea of having a world->map transform, that you adjust for each robot, when you reset the map.

matlabbe, 2023-11-21:
I think world->map approach would make more sense. By the way, you can use `/rtabmap/load_database` service to create a new database online for convenience if you were not using it.

closed 2023-11-21 by Krzo99